openwrt/packages #11610: uwsgi: can't handle large firmware to upgrade

dengqf6, 2020-03-20:
Maintainer: @Ansuel 
Environment: OpenWrt master

Description:
When I upload a large firmware to upgrade, the UI stuck at 100% and there was an error in system log
```
daemon.err nginx[5127]: 2020/03/21 00:53:40 [error] 5154#0: *117 client intended to send too large body: 18382143 bytes, client: 10.0.0.2, server: _lan, request: "POST /cgi-bin/cgi-upload?1584723219331 HTTP/1.1", host: "10.0.0.1", referrer: "https://10.0.0.1/cgi-bin/luci/admin/system/flash"
```

Ansuel, 2020-03-20:
Post nginx conf and the size of the firmware you are trying to upload

dengqf6, 2020-03-20:
```
# Please consider creating files in /etc/nginx/conf.d/ instead of editing this.
# For details see https://openwrt.org/docs/guide-user/services/webserver/nginx

user root;

events {}

http {
	access_log off;
	log_format openwrt
		'$request_method $scheme://$host$request_uri => $status'
		' (${body_bytes_sent}B in ${request_time}s) <- $http_referer';

	include mime.types;
	default_type application/octet-stream;
	sendfile on;

	client_max_body_size 17M;
	large_client_header_buffers 2 1k;

	gzip on;
	gzip_vary on;
	gzip_proxied any;

	root /www;

	include conf.d/*.conf;
}
```

File size is in the log:
`client intended to send too large body: 18382143 bytes`

Ansuel, 2020-03-20:
will fix this right away.... 17m is too low

PR #11611 (2020-03-20): nginx: bump version and fix bug 
This fix #11610 by increasing max client body size
in nginx conf.
Bump nginx to latest version.

@hnyman
commit 0910d61c (2020-03-20): nginx: bump version and fix bug

This fix #11610 by increasing max client body size
in nginx conf.
Bump nginx to latest version.
closed 2020-03-20 by hnyman