openwrt/packages #11621: [19.07] macremapper: can not be built with enabled ASLR PIE

BKPepe, 2020-03-21:
Maintainer: @csonsino 
Environment: powerpc 8540 && aarch64 cortex-a53 , OpenWrt 19.07

Description:

```
Applying ./patches/01_fix_nf_hooks.patch using plaintext: 
patching file kernelmod/main.c
make[3]: Entering directory '/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod'
make -C /mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/linux-4.14.172 M=/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod modules
make[4]: Entering directory '/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/linux-4.14.172'
  CC [M]  /mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod/mrm_rcdb.o
  CC [M]  /mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod/mrm_runconf.o
  CC [M]  /mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod/mrm_ctlfile.o
  CC [M]  /mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod/main.o
  CC [M]  /mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod/filter_config_accelerator.o
  LD [M]  /mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod/macremapper.o
powerpc-openwrt-linux-musl-ld: unrecognized option '-specs=/home/beast/beast/workspace/turris-os-packages-kittens-turris1x/build/include/hardened-ld-pie.specs'
powerpc-openwrt-linux-musl-ld: use the --help option for usage information
scripts/Makefile.build:572: recipe for target '/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod/macremapper.o' failed
make[5]: *** [/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod/macremapper.o] Error 1
Makefile:1542: recipe for target '_module_/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod' failed
make[4]: *** [_module_/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod] Error 2
make[4]: Leaving directory '/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/linux-4.14.172'
Makefile:71: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/kernelmod'
Makefile:42: recipe for target '/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/.built' failed
make[2]: *** [/mypath/target-powerpc_8540_musl/linux-mpc85xx_p2020/macremapper-1.1.0/.built] Error 2
time: package/feeds/packages/macremapper/compile#3.40#0.48#4.61
```

neheb, 2020-03-21:
I think it's master too.

csonsino, 2020-03-23:
What config or combination of config settings do I need to reproduce this?  I set `CONFIG_PKG_ASLR_PIE_ALL=y` and master seemed to build fine for me.  Haven't tried 19.07 yet.

neheb, 2020-04-04:
@csonsino it's for powerpc target, not mips_24kc.