openwrt/packages #11632: [19.07,18.06] ttyd: can not be installed it due conflict libwebsockets-full/openssl

BKPepe, 2020-03-23:
Maintainer: @tsl0922
Environment: OpenWrt 19.07, OpenWrt 18.06,  mvebu - cortex-a9
It should happen also with OpenWrt master, but didn't try it, yet.

I see that dependency from `libwebsockets-openssl` was changed to `libwebsockets-full` in this PR https://github.com/openwrt/packages/pull/5672. Is a full variant of libwebsockets required for ttyd? Shouldn't be openssl/mbedtls variant enough?

This is a problem when I have installed mosquitto-ssl as it installs [libwebsockets-openssl](https://github.com/openwrt/packages/blob/openwrt-19.07/net/mosquitto/Makefile#L38) and it leads to conflict. It's ttyd or mosquitto.

Ping maintainer of mosquitto and libwebsockets: @karlp 

Description:

Installation of ttyd:
```
root@turris:~# opkg install ttyd
Installing ttyd (1.5.2-1.15) to root...
Downloading ttyd_1.5.2-1_arm_cortex-a9_vfpv3.ipk
Installing libwebsockets-full (3.1.0-1.0) to root...
Downloading libwebsockets-full_3.1.0-1_arm_cortex-a9_vfpv3.ipk
Collected errors:
 * check_data_file_clashes: Package libwebsockets-full wants to install file /usr/lib/libwebsockets.so
        But that file is already provided by package  * libwebsockets-openssl
 * check_data_file_clashes: Package libwebsockets-full wants to install file /usr/lib/libwebsockets.so.14
        But that file is already provided by package  * libwebsockets-openssl
 * opkg_install_cmd: Cannot install package ttyd.
```

Removal of libwebsockets-openssl:
```
root@turris:~# opkg remove libwebsockets-openssl
No packages removed.
Collected errors:
 * print_dependents_warning: Package libwebsockets-openssl is depended upon by packages:
 * print_dependents_warning:    mosquitto-ssl
 * print_dependents_warning: These might cease to work if package libwebsockets-openssl is removed.

 * print_dependents_warning: Force removal of this package with --force-depends.
 * print_dependents_warning: Force removal of this package and its dependents
 * print_dependents_warning: with --force-removal-of-dependent-packages.
```

tsl0922, 2020-03-23:
> Is a full variant of libwebsockets required for ttyd?

Yes, ttyd requires libwebsockets with libuv support since 1.6.0.

BKPepe, 2020-03-24:
Thanks! So, this seems to should be solved in mosquitto. Because I can not have installed ttyd and mosquitto-ssl at the same time.

karlp, 2020-03-26:
I think the best option is to just drop the libwebsockets-full version altogether, I'd been looking at that a month or so ago, I think it's only a tiny size difference between them anyway (full already includes openssl)  The problem really is the absolutely _relentless_ feature creep of libwebsockets, which now includes a dns resolver, a dhcp client, it's _own_ mqtt implementation, an ntp client, string formatting "humanize" functionality, which will eventually make any sort of "full" implementation a growing disaster.

n8v8R, 2020-06-15:
Tested with

>"distribution":"OpenWrt","version":"SNAPSHOT","revision":"r13552-cd09f26660","target":"mvebu/cortexa9"

* libwebsockets-full 3.2.1-1 been installed first
* ttyd 1.6.0-2 been installed after

no issues

BKPepe, 2020-06-15:
@n8v8R I think there is going some misunderstanding. You can have install ttyd (and libwebsockets-full), but you can not have installed mosquitto at the same time and there's another possibility that you can have installed mosquitto (with libwebsockets-openssl as it is depedency), but you can not install ttyd.

n8v8R, 2020-06-15:
Well, I took the the hint about the issue from the thread topic 

>ttyd: can not be installed it due conflict libwebsockets-full/openssl

Mosquito seems to be a different issue then

> Collected errors:
>  * check_data_file_clashes: Package libwebsockets-openssl wants to install file /usr/lib/libwebsockets.so
> 	But that file is already provided by package  * libwebsockets-full
>  * check_data_file_clashes: Package libwebsockets-openssl wants to install file /usr/lib/libwebsockets.so.15
> 	But that file is already provided by package  * libwebsockets-full
>  * opkg_install_cmd: Cannot install package mosquitto-ssl.

BKPepe, 2020-06-15:
Also, It is good to read the content of the post instead of just the title. 

Quotation of the original post
> This is a problem when I have installed mosquitto-ssl as it installs libwebsockets-openssl and it leads to conflict. It's ttyd or mosquitto.

Maybe next time, I will highlight it to make it more clear for you.

n8v8R, 2020-06-15:
It all sounds then like basically an issue with `opkg` not handling multiple/cross package dependencies well.

feckert, 2020-08-06:
I'm not sure if I should open a new issue, so I write it here.

I have a problem with ttyd. On a x86_64 (little endian) ttyd works as expected.
But if I use lantiq_xrx200 (big endian its a MIPS) nothing works.
The peer (router) terminates the connection.
@tsl0922 @karlp Can someone please verify that?

I did a `curl --trace-ascii dump-lantiq.txt http://192.168.0.53:7681` and got the following output in dump-lantiq.txt
```
$ cat dump-drno.txt                                                                                                                        [14:57:27]
== Info: Expire in 0 ms for 6 (transfer 0x5601db4bbf50)
== Info:   Trying 192.168.0.53...
== Info: TCP_NODELAY set
== Info: Expire in 200 ms for 4 (transfer 0x5601db4bbf50)
== Info: Connected to 192.168.0.53 (192.168.0.53) port 7681 (#0)
=> Send header, 81 bytes (0x51)
0000: GET / HTTP/1.1
0010: Host: 192.168.0.53:7681
0029: User-Agent: curl/7.64.0
0042: Accept: */*
004f:
== Info: Recv failure: Connection reset by peer
== Info: Closing connection 0
```

karlp, 2020-08-06:
I certainly can't verify it, but it wouldn't surprise me in the least if lws wsa broken in that case.  it definitely sounds like a completel separate issue though.  (and upstream won't care, because we're not on latest, and we're not on latest, because there's still lots of applications broken by latest upstream.... I love LWS)

I'm also not sure what you were hoping to see with curl?  You need to do a websockets upgrade? What does curl give you on a working setup? the same thing I imagine?

feckert, 2020-08-11:
> I certainly can't verify it, but it wouldn't surprise me in the least if lws wsa broken in that case. it definitely sounds like a completel separate issue though. (and upstream won't care, because [...]

Currently verison [4.0](https://libwebsockets.org/git/libwebsockets/tree/?h=v4.0-stable) is the latest

The way I see it we have the following packages in the package feed that use libwebsocket.

- mosquitto
- forked-daapd
- ttyd

That's not even that many?

> 
> I'm also not sure what you were hoping to see with curl? You need to do a websockets upgrade? What does curl give you on a working setup? the same thing I imagine?

If I do a curl call to a working ttyd on x86_64 I get the page. That is OK.

karlp, 2020-08-11:
I very much think the ttyd endian issue you describe should be tracked separately.  I do still plain on eliminating the full/openssl split when I update lws to 4.x, which will get rid of the issues in _this_ ticket, but I need my apps to work with lws 4.x first.

vlna, 2021-09-22:
This issue persist in OpenWrt 21.02.0, r16279-5cc0535800
[image]

wulfy23, 2021-09-23:
+ master
+ docker

https://forum.openwrt.org/t/rpi4-community-build/69998/993?u=wulfy23

gclawes, 2022-10-24:
This is still an issue:
[image]

OS is TurrisOS 6.0.1, based on OpenWRT 21.02.5
```
root@turris:~# cat /etc/os-release
NAME="TurrisOS"
VERSION="6.0.1"
ID="turrisos"
ID_LIKE="lede openwrt"
PRETTY_NAME="TurrisOS 6.0.1"
VERSION_ID="6.0.1"
HOME_URL="https://www.turris.cz/"
BUG_URL="https://gitlab.nic.cz/groups/turris/-/issues/"
SUPPORT_URL="https://www.turris.cz/support/"
BUILD_ID="r16691+128-060aa0021f"
OPENWRT_BOARD="mvebu/cortexa9"
OPENWRT_ARCH="arm_cortex-a9_vfpv3-d16"
OPENWRT_TAINTS="busybox"
OPENWRT_DEVICE_MANUFACTURER="CZ.NIC"
OPENWRT_DEVICE_MANUFACTURER_URL="https://www.turris.cz/"
OPENWRT_DEVICE_PRODUCT="Turris Omnia"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="TurrisOS 6.0.1 060aa0021f688d455767b553cfe95e29b6559b7a"
```
https://forum.turris.cz/t/turris-os-6-0-is-released/17830

https://forum.turris.cz/t/turris-os-6-0-1-is-now-released/17973

closed 2022-10-25 by BKPepe