openwrt/packages #11640: Mwan3 in the master branch causes the network to be disconnected！！！

zxlhhyccc, 2020-03-23:
Mwan3 online detection continuously detects the default IP. If the detection times exceed, the network will be cut off. Mwan3 has no such problem under the openwrt-19.07 branch。。。

feckert, 2020-03-24:
We have done the following change recently for IPv6.
https://github.com/openwrt/packages/commit/6721587e8b491f1a539cdcc507b0f4811ff2b448#diff-87268a842bab5effbe181f9f18b2d348

zxlhhyccc, 2020-03-24:
> We have done the following change recently for IPv6.
> [6721587#diff-87268a842bab5effbe181f9f18b2d348](https://github.com/openwrt/packages/commit/6721587e8b491f1a539cdcc507b0f4811ff2b448#diff-87268a842bab5effbe181f9f18b2d348)

Is it the problem caused by updating IPV6?

feckert, 2020-03-24:
It depends on whether you use IPv4 or IPv6.
I could not find any problems with IPv4 so far.
From my perspective IPv6 looks good.
What is your use case do you use IPv4 or IPv6?

zxlhhyccc, 2020-03-24:
> 这取决于您使用的是IPv4还是IPv6。
> 到目前为止，我找不到IPv4的任何问题。
> 从我的角度来看，IPv6看起来不错。
> 您使用IPv4或IPv6的用例是什么？

> It depends on whether you use IPv4 or IPv6.
> I could not find any problems with IPv4 so far.
> From my perspective IPv6 looks good.
> What is your use case do you use IPv4 or IPv6?

No use cases are configured, just the default configuration... But the network will not be intermittent, if you do not compile mwan3, the routing is all right！

zxlhhyccc, 2020-03-25:
> It depends on whether you use IPv4 or IPv6.
> I could not find any problems with IPv4 so far.
> From my perspective IPv6 looks good.
> What is your use case do you use IPv4 or IPv6?

Test mawan3 again, intermittent disconnection still exists:

1. After mwan3 is compiled into the route, intermittent network interruption, even if it is closed or disabled is still a problem;

2. Remove mwan3 does not compile to the route, the whole route runs normally;

3. Modify the parameters of mwan3 configuration, or reconfigure, the problem remains;

Therefore, it can be concluded that there may be an incompatibility between mwan3 and the master branch, resulting in a spacer outage.

I use ipv4..

feckert, 2020-03-25:
Sorry, I don't know what you mean! I installed the last OpenWrt master with my config from yesterday and my setup works.
* Did you change the metric of the interfaces in the network configuration?
* Does the interfaces works without mwan3 started?
* Is this a OpenWrt firmware or a custom firmware?
* I need the config mwan3 and network config in /etc/config and your installed version number

Unfortunately I cannot do anything with error description.

zxlhhyccc, 2020-03-25:
mwan3的配置：
config globals 'globals'
	option mmx_mask '0x3F00'
	option rtmon_interval '5'
	list rt_table_lookup '1'
	list rt_table_lookup '2'
	list rt_table_lookup '220'

config interface 'wan'
	list track_ip '114.114.114.114'
	list track_ip '119.29.29.29'
	list track_ip '223.5.5.5'
	list track_ip '9.9.9.9'
	list track_ip '209.244.0.3'
	option family 'ipv4'
	option reliability '2'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'
	option enabled '1'
	option initial_state 'online'
	option track_method 'ping'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option failure_interval '5'
	option recovery_interval '5'

config interface 'wan6'
	option enabled '0'
	list track_ip '2001:4860:4860::8844'
	list track_ip '2001:4860:4860::8888'
	list track_ip '2620:0:ccd::2'
	list track_ip '2620:0:ccc::2'
	option family 'ipv6'
	option reliability '2'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'

config interface 'wanb'
	option enabled '0'
	list track_ip '8.8.4.4'
	list track_ip '8.8.8.8'
	list track_ip '208.67.222.222'
	list track_ip '208.67.220.220'
	option family 'ipv4'
	option reliability '1'
	option count '1'
	option timeout '2'
	option failure_latency '1000'
	option recovery_latency '500'
	option failure_loss '20'
	option recovery_loss '5'
	option interval '5'
	option down '3'
	option up '8'

config interface 'wanb6'
	option enabled '0'
	list track_ip '2001:4860:4860::8844'
	list track_ip '2001:4860:4860::8888'
	list track_ip '2620:0:ccd::2'
	list track_ip '2620:0:ccc::2'
	option family 'ipv6'
	option reliability '1'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'

config member 'wan_m1_w3'
	option interface 'wan'
	option metric '1'
	option weight '3'

config member 'wan_m2_w3'
	option interface 'wan'
	option metric '2'
	option weight '3'

config member 'wanb_m1_w2'
	option interface 'wanb'
	option metric '1'
	option weight '2'

config member 'wanb_m2_w2'
	option interface 'wanb'
	option metric '2'
	option weight '2'

config member 'wan6_m1_w3'
	option interface 'wan6'
	option metric '1'
	option weight '3'

config member 'wan6_m2_w3'
	option interface 'wan6'
	option metric '2'
	option weight '3'

config member 'wanb6_m1_w2'
	option interface 'wanb6'
	option metric '1'
	option weight '2'

config member 'wanb6_m2_w2'
	option interface 'wanb6'
	option metric '2'
	option weight '2'

config policy 'wan_only'
	list use_member 'wan_m1_w3'
	list use_member 'wan6_m1_w3'

config policy 'wanb_only'
	list use_member 'wanb_m1_w2'
	list use_member 'wanb6_m1_w2'

config policy 'balanced'
	list use_member 'wan_m1_w3'
	list use_member 'wanb_m1_w2'
	list use_member 'wan6_m1_w3'
	list use_member 'wanb6_m1_w2'

config policy 'wan_wanb'
	list use_member 'wan_m1_w3'
	list use_member 'wanb_m2_w2'
	list use_member 'wan6_m1_w3'
	list use_member 'wanb6_m2_w2'

config policy 'wanb_wan'
	list use_member 'wan_m2_w3'
	list use_member 'wanb_m1_w2'
	list use_member 'wan6_m2_w3'
	list use_member 'wanb6_m1_w2'

config rule 'https'
	option sticky '1'
	option dest_port '443'
	option proto 'tcp'
	option use_policy 'balanced'

config rule 'default_rule'
	option dest_ip '0.0.0.0/0'
	option use_policy 'balanced'

zxlhhyccc, 2020-03-25:
> Sorry, I don't know what you mean! I installed the last OpenWrt master with my config from yesterday and my setup works.
>
> * Did you change the metric of the interfaces in the network configuration? [...]
installed version number：
[image]

zxlhhyccc, 2020-03-25:
> 对不起，我不明白你的意思！我从昨天开始使用配置安装了最后一个OpenWrt主服务器，并且安装正常。
> 
> * 您是否更改了网络配置中接口的度量标准？
> * 如果没有启动mwan3，接口是否可以正常工作？
> * 这是OpenWrt固件还是自定义固件？
> * 我需要/ etc / config中的配置mwan3和网络配置以及您安装的版本号
> 
> 不幸的是，我无法对错误描述做任何事情。

> Sorry, I don't know what you mean! I installed the last OpenWrt master with my config from yesterday and my setup works.
>
> * Did you change the metric of the interfaces in the network configuration? [...]

1. Use the dynamic DDNS of 1907 branch to configure the routing table under the global mwan3 operation is normal, but use the dynamic DDNS of master branch to configure the same network outage;

2. Even if mwan3 is turned off or uninstalled, the problem remains;

zxlhhyccc, 2020-03-25:
19.07 branch mwan3 and DNS -ddns-scripts are replaced on the master main branch and are running normally. Therefore, there is definitely a bug in mwan3 of the current master branch, please check and fix it...

ptpt52, 2020-03-29:
> The mwan3 and DNS -scripts of the 19.07 branch are replaced on the master main branch and are running normally. Therefore, mwan3 of the master branch is definitely buggy, or may not be very compatible with DNS -scripts, please check and fix...

你找我沟通一下好了。

feckert, 2020-03-30:
@ptpt52 could you please have a look? I do not know what the problem is? 
This are the changes I have done since 19.07 release in master.

https://github.com/openwrt/packages/commit/581d7df844f47e5284245db47b36e21138d2653e#diff-34590d11545c84b961d1992ce3aefdeb
https://github.com/openwrt/packages/commit/a689e168aaa9c9850e35c5ff3f2f0745d04d5006#diff-34590d11545c84b961d1992ce3aefdeb
https://github.com/openwrt/packages/commit/14dd084662ae9f92091c34a177ef0e9a3c2913c8#diff-34590d11545c84b961d1992ce3aefdeb
https://github.com/openwrt/packages/commit/6d638c3e23f66850cc8bd272b9fe5b9f11bec784#diff-34590d11545c84b961d1992ce3aefdeb
https://github.com/openwrt/packages/commit/6721587e8b491f1a539cdcc507b0f4811ff2b448#diff-34590d11545c84b961d1992ce3aefdeb

I think this is not a software issue. I think there is a configuration problem.

ptpt52, 2020-03-31:
> @ptpt52 could you please have a look? I do not know what the problem is?
> This are the changes I have done since 19.07 release in master.
> 
> [581d7df#diff-34590d11545c84b961d1992ce3aefdeb](https://github.com/openwrt/packages/commit/581d7df844f47e5284245db47b36e21138d2653e#diff-34590d11545c84b961d1992ce3aefdeb)
> [a689e16#diff-34590d11545c84b961d1992ce3aefdeb](https://github.com/openwrt/packages/commit/a689e168aaa9c9850e35c5ff3f2f0745d04d5006#diff-34590d11545c84b961d1992ce3aefdeb)
> [14dd084#diff-34590d11545c84b961d1992ce3aefdeb](https://github.com/openwrt/packages/commit/14dd084662ae9f92091c34a177ef0e9a3c2913c8#diff-34590d11545c84b961d1992ce3aefdeb)
> [6d638c3#diff-34590d11545c84b961d1992ce3aefdeb](https://github.com/openwrt/packages/commit/6d638c3e23f66850cc8bd272b9fe5b9f11bec784#diff-34590d11545c84b961d1992ce3aefdeb)
> [6721587#diff-34590d11545c84b961d1992ce3aefdeb](https://github.com/openwrt/packages/commit/6721587e8b491f1a539cdcc507b0f4811ff2b448#diff-34590d11545c84b961d1992ce3aefdeb)
> 
> I think this is not a software issue. I think there is a configuration problem.

LGTM

ptpt52, 2020-03-31:
> The mwan3 and DNS -scripts of the 19.07 branch are replaced on the master main branch and are running normally. Therefore, mwan3 of the master branch is definitely buggy, or may not be very compatible with DNS -scripts, please check and fix...

what is your issue, how does it happen

feckert, 2020-05-29:
Not furhter comment by @zxlhhyccc.
So I assume that's was fixed.
Thanks @ptpt52 for the check.

closed 2020-05-29 by feckert