openwrt/packages #11665: net/openvswitch: build fails without ipv6

pandasauce, 2020-03-25:
Maintainer: @yousong 
Environment: ipq806x / Netgear R7800 / v19.07.2

Description:

Building v19.07.2 with `Enable IPv6 support in packages` unticked fails and results in the following errors related to IPv6 in `net/openvswitch`. Normal build succeeds. Looks like this package may have some hardcoded dependencies on IPv6 or the config option is not being checked for in some spots?

```
make[3]: Entering directory '/sources/feeds/packages/net/openvswitch'
rm -f /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/.built
touch /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/.built_check
cat /dev/null > /sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/openvswitch.symvers; for subdir in .; do cat /sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/*.symvers 2>/dev/null > /sources/b
uild_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/$subdir/Module.symvers; done
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch
-2.11.1:openvswitch-2.11.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99  -I/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/sources/staging_di
r/target-arm_cortex-a15+neon-vfpv4_musl_eabi/include -I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/usr/include -I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/include/fortify
-I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/sou
rces/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1:openvswitch-2.11.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99  -I/sourc
es/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/include -I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/usr/include -I/s
ources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/include/fortify -I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/include " LDFLAGS="-L/sources/staging_dir/target-arm_cortex-a15+neon-
vfpv4_musl_eabi/usr/lib -L/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/usr/lib -L/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4
_gcc-7.5.0_musl_eabi/lib -znow -zrelro " PYTHONPATH="/sources/staging_dir/hostpkg/lib/python3.7:/sources/staging_dir/hostpkg/lib/python3.7/site-packages" make  -C /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_
generic/openvswitch-2.11.1/. AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=har
d -iremap/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1:openvswitch-2.11.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gn
u99" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi
-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" ;
make[4]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1'
make  all-recursive
make[5]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1'
Making all in datapath
make[6]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath'
Making all in linux
make[7]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux'
make -C /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-4.14.171 ARCH=arm M=/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux mod
ules
make[8]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-4.14.171'
  CC [M]  /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.o
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1676:15: error: variable 'ip6gre_protocol' has initializer but incomplete type
 static struct inet6_protocol ip6gre_protocol __read_mostly = {
               ^~~~~~~~~~~~~~
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1677:3: error: 'struct inet6_protocol' has no member named 'handler'
  .handler     = gre_rcv,
   ^~~~~~~
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1677:17: warning: excess elements in struct initializer
  .handler     = gre_rcv,
                 ^~~~~~~
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1677:17: note: (near initialization for 'ip6gre_protocol')
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1678:3: error: 'struct inet6_protocol' has no member named 'err_handler'
  .err_handler = ip6gre_err,
   ^~~~~~~~~~~
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1678:17: warning: excess elements in struct initializer
  .err_handler = ip6gre_err,
                 ^~~~~~~~~~
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1678:17: note: (near initialization for 'ip6gre_protocol')
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1679:3: error: 'struct inet6_protocol' has no member named 'flags'
  .flags       = INET6_PROTO_NOPOLICY|INET6_PROTO_FINAL,
   ^~~~~
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1679:17: error: 'INET6_PROTO_NOPOLICY' undeclared here (not in a function); did you mean 'NET_IPV4_CONF_NOPOLIC
Y'?
  .flags       = INET6_PROTO_NOPOLICY|INET6_PROTO_FINAL,
                 ^~~~~~~~~~~~~~~~~~~~
                 NET_IPV4_CONF_NOPOLICY
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1679:38: error: 'INET6_PROTO_FINAL' undeclared here (not in a function); did you mean 'INET6_PROTO_NOPOLICY'?
  .flags       = INET6_PROTO_NOPOLICY|INET6_PROTO_FINAL,
                                      ^~~~~~~~~~~~~~~~~
                                      INET6_PROTO_NOPOLICY
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1679:17: warning: excess elements in struct initializer
  .flags       = INET6_PROTO_NOPOLICY|INET6_PROTO_FINAL,
                 ^~~~~~~~~~~~~~~~~~~~
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1679:17: note: (near initialization for 'ip6gre_protocol')
[... 22 lines cut ...]
[... cut ...]
package/Makefile:111: recipe for target 'package/feeds/packages/openvswitch/compile' failed
make[2]: *** [package/feeds/packages/openvswitch/compile] Error 2
make[2]: Leaving directory '/sources'
package/Makefile:107: recipe for target '/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/sources'
/sources/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

yousong, 2020-03-25:
The failure occurred when building datapath kmod in the ovs source tree, the one in the upstream linux kernel tree should be fine.

For now, I think we need to make the intree variant depend on `IPV6`, please see if the following patch could skip building of the intree kmods

```
diff --git a/net/openvswitch/Makefile b/net/openvswitch/Makefile
index bb3c3dd43..b26cbf400 100644
--- a/net/openvswitch/Makefile
+++ b/net/openvswitch/Makefile
@@ -55,7 +55,7 @@ define OvsKmodPackageTemplate
      CATEGORY:=Kernel modules
      SUBMENU:=Network Support
      TITLE:=$(ovs_kmod_$(1)_title)
-     DEPENDS:=$(ovs_kmod_$(1)_depends) $(if $(call ovs_kmod_is_intree,$(1)),@DEVEL $(if $(ovs_kmod_intree_not_supported),@BROKEN))
+     DEPENDS:=$(ovs_kmod_$(1)_depends) $(if $(call ovs_kmod_is_intree,$(1)),@IPV6 @DEVEL $(if $(ovs_kmod_intree_not_supported),@BROKEN))
      PROVIDES:=$(call ovs_kmod_package_provides,$(1))
      KCONFIG:=$(ovs_kmod_$(1)_kconfig)
      FILES:=$(ovs_kmod_$(1)_files)

```

pandasauce, 2020-03-25:
That didn't seem to help. Do I need to run a clean build for it to kick in?

```
make[3]: Entering directory '/sources/feeds/packages/net/openvswitch'
rm -f /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/.built
touch /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/.built_check
cat /dev/null > /sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/openvswitch.symvers; for subdir in .; do cat /sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/*.symvers 2>/dev/null > /sources/b
uild_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/$subdir/Module.symvers; done
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch
-2.11.1:openvswitch-2.11.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99  -I/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/sources/staging_di
r/target-arm_cortex-a15+neon-vfpv4_musl_eabi/include -I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/usr/include -I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/include/fortify
-I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/sou
rces/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1:openvswitch-2.11.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99  -I/sourc
es/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/include -I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/usr/include -I/s
ources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/include/fortify -I/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/include " LDFLAGS="-L/sources/staging_dir/target-arm_cortex-a15+neon-
vfpv4_musl_eabi/usr/lib -L/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.5.0_musl_eabi/usr/lib -L/sources/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4
_gcc-7.5.0_musl_eabi/lib -znow -zrelro " PYTHONPATH="/sources/staging_dir/hostpkg/lib/python3.7:/sources/staging_dir/hostpkg/lib/python3.7/site-packages" make  -C /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_
generic/openvswitch-2.11.1/. AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=har
d -iremap/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1:openvswitch-2.11.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gn
u99" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi
-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" ;
make[4]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1'
make  all-recursive
make[5]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1'
Making all in datapath
make[6]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath'
Making all in linux
make[7]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux'
make -C /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-4.14.171 ARCH=arm M=/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux mod
ules
make[8]: Entering directory '/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-4.14.171'
  CC [M]  /sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.o
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1676:15: error: variable 'ip6gre_protocol' has initializer but incomplete type
 static struct inet6_protocol ip6gre_protocol __read_mostly = {
               ^~~~~~~~~~~~~~
/sources/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/openvswitch-2.11.1/datapath/linux/ip6_gre.c:1677:3: error: 'struct inet6_protocol' has no member named 'handler'
[... cut ...]
package/Makefile:111: recipe for target 'package/feeds/packages/openvswitch/compile' failed
make[2]: *** [package/feeds/packages/openvswitch/compile] Error 2
make[2]: Leaving directory '/sources'
package/Makefile:107: recipe for target '/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/sources/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/sources'
/sources/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

yousong, 2020-03-25:
Please share output of the following command

```
grep -iE '_ipv6|openvswitch' .config
```

After a new run of menuconfig or defconfig, kmod-openvswitch-intree should only be selectable when CONFIG_IPV6 is enabled.

yousong, 2020-03-26:
Hi @pandasauce , I just also backported the `@IPV6` fix to openwrt-19.07.  Please give it a try to see if it will de-select kmod-openvswitch-intree and friends when `CONIFG_IPV6` is not set

pandasauce, 2020-03-26:
Running `defconfig` again did it. Just flashed the resulting build. Thank you!

closed 2020-04-18 by yousong