openwrt/packages #11669: backport to stable branches instructions

lucize, 2020-03-25:
can somebody please make a wiki entry on how to update the stable branches, I did it once but I can't find the instructions anymore
thank you !

hnyman, 2020-03-26:
What do you mean?
Backport fixes from master to a stable branch?

* checkout the stable branch, git pull to update it
* create a topic branch for the PR
* git cherry-pick the fix commit(s) from master  (or manually adjust the fiels as necessary)
* git push to origin, set tracking branch
* create PR

```
git checkout openwrt-19.07
git checkout -b fixbranch
git cherry-pick -x <commit_hash>
git push -u origin fixbranch
<create PR>
```

lucize, 2020-03-26:
@hnyman Yes, that's what I was searching for and couldn't find this in time to add frr and libyang to  stable so we can drop the obsoleted quagga package

closed 2025-11-14 by GeorgeSapkin