openwrt/packages #11675: acme: Commits missing in 19.07 backport

scde, 2020-03-26:
Maintainer: @tohojo 
Environment: OpenWrt 19.07.2 r10947-65030d81f3

Description:
Hi, I am running 19.07.2 which should have the `2.8.5-1` version of `acme` but its missing two commits from the master branch which were made before the `2.8.5-1` release. You can compare [the 19.07 branch](https://github.com/openwrt/packages/commits/2c0a307eeb4b2c2f6aa34b54dc0bcac6037c185d/net/acme) and [the master branch](https://github.com/openwrt/packages/commits/ab6d05d366de9df7265523222c3614520de1bb7c/net/acme) and see commits c6b4d7f and 983cc99 are missing from the backport.

Thank you in advance, have a nice day and stay healthy =)

tohojo, 2020-03-26:
Fixed in 9ba7b1799c2326d97157059f0888f33fc16f5721

closed 2020-03-26 by tohojo

scde, 2020-03-26:
@tohojo Great, thanks.

commit 9ba7b179 (2020-03-26): acme: Backport two bug fixes from master

This backports the following bug-fix commits from the master branch:

c6b4d7f367de ("acme: Include empty 'dns' config option by default")
983cc995a3b2 ("acme: Correctly handle domain state dir for ECC certificates")

Fixes #11675.