openwrt/packages #11676: gnutls: 3.6.12 build fails (gmp issue)

BlueMax, 2020-03-26:
Maintainer: @nmav 
Environment:
Arch Linux 5.5.11
OpenWrt master 556b8581a15c855b2de0efbea6b625ab16cc9daf
./scripts/feeds update -a
./scripts/feeds install -d y libpam libgnutls libopenldap libidn2 libssh2 liblzma libnetsnmp jansson
(these packages are mandatory. Otherwise `make` declares `missing` deps)
kernel 5.4 testing 
gcc 9.3.0, binutils 2.34.1 (same on default 8.3.0, 2.31.1)

Description:
`gnutls` compile breaks with `gmp.h: No such file`. 
`staging_dir/host/include/gmp.h` and `build_dir/host/gmp-6.2.0` exist (and look ok so far). 
My last working snapshot is `OpenWrt SNAPSHOT, r12425-d83150e3fb` (kernel 5.4.22).

Any ideas?
```
$ make -j1 package/gnutls/compile V=s
make[1]: Entering directory '/media/LinuxPart2/OpenWRT/openwrt-compile'
make[2]: Entering directory '/media/LinuxPart2/OpenWRT/openwrt-compile/package/libs/toolchain'
echo "libc" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libatomic" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "librt" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/media/LinuxPart2/OpenWRT/openwrt-compile/package/libs/toolchain'
time: package/libs/toolchain/compile#0.14#0.04#0.18
make[2]: Entering directory '/media/LinuxPart2/OpenWRT/openwrt-compile/package/libs/nettle'
echo "libnettle" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/nettle.default.install
make[2]: Leaving directory '/media/LinuxPart2/OpenWRT/openwrt-compile/package/libs/nettle'
time: package/libs/nettle/compile#0.13#0.05#0.16
make[2]: Entering directory '/media/LinuxPart2/OpenWRT/openwrt-compile/package/firmware/linux-firmware'
echo "bnx2-firmware" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
echo "r8169-firmware" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[2]: Leaving directory '/media/LinuxPart2/OpenWRT/openwrt-compile/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#0.19#0.05#0.21
make[2]: Entering directory '/media/LinuxPart2/OpenWRT/openwrt-compile/package/firmware/prism54-firmware'
make[2]: Nothing to be done for 'compile'.
make[2]: Leaving directory '/media/LinuxPart2/OpenWRT/openwrt-compile/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.11#0.03#0.12
make[2]: Entering directory '/media/LinuxPart2/OpenWRT/openwrt-compile/package/kernel/linux'
echo "kernel" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-aead" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-authenc" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-cmac" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-ecb" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-ecdh" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-hash" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-hw-padlock" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-kpp" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-manager" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-null" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
WARNING: kmod-crypto-pcompress is not available in the kernel config - generating empty package
echo "kmod-crypto-pcompress" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-crypto-user" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-hwmon-core" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-i2c-core" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-i2c-algo-bit" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-hid" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-input-core" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-input-evdev" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-lib-crc-ccitt" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-lib-crc16" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-libphy" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
WARNING: kmod-mii is not available in the kernel config - generating empty package
echo "kmod-mii" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-phy-realtek" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-r8169" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
echo "kmod-e1000" >> /media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
[... 138 lines cut ...]
make[2]: *** [Makefile:251: /media/LinuxPart2/OpenWRT/openwrt-compile/build_dir/target-x86_64_musl/gnutls-3.6.12/.built] Error 2
make[2]: Leaving directory '/media/LinuxPart2/OpenWRT/openwrt-compile/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#0.52#0.10#0.60
make[1]: *** [package/Makefile:113: package/feeds/packages/gnutls/compile] Error 2
make[1]: Leaving directory '/media/LinuxPart2/OpenWRT/openwrt-compile'
make: *** [/media/LinuxPart2/OpenWRT/openwrt-compile/include/toplevel.mk:227: package/gnutls/compile] Fehler 2
```
When adding 'libgmp' package many 'undefined symbols' come up instead.
Log (excerpt):
```
make  all-am
make[8]: Entering directory '/media/LinuxPart2/OpenWRT/openwrt-compile/build_dir/target-x86_64_musl/gnutls-3.6.12/lib/nettle'
/bin/sh ../../libtool  --tag=CC   --mode=link x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/media/LinuxPart2/OpenWRT/openwrt-compile/build_dir/target-x86_64_musl/gnutls-3.6.12=gnutls-3.6.12 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -L/media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/usr/lib -L/media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/target-x86_64_musl/lib -L/media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/toolchain-x86_64_gcc-9.3.0_musl/usr/lib -L/media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/toolchain-x86_64_gcc-9.3.0_musl/lib -znow -zrelro  -o curve448/eccdata curve448/eccdata-eccdata.o  ../../gl/libgnu.la 
libtool: link: x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/media/LinuxPart2/OpenWRT/openwrt-compile/build_dir/target-x86_64_musl/gnutls-3.6.12=gnutls-3.6.12 -W [... cut ...]
/media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/toolchain-x86_64_gcc-9.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/9.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: curve448/eccdata-eccdata.o: in function `ecc_curve_init_str':
eccdata.c:(.text+0x75): undefined reference to `__gmpz_init_set_str'
/media/LinuxPart2/OpenWRT/openwrt-compile/staging_dir/toolchain-x86_64_gcc-9.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/9.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: eccdata.c:(.text+0x87): undefined reference to `__gmpz_init_set_str'
[...]
```

nmav, 2020-03-27:
Can you compile nettle with gmp on the openwrt repo? I wonder whether it is gmp-devel in the build system you are missing (gnutls bundles some nettle code now and it inherits this dependency).

BlueMax, 2020-03-27:
Yes, `nettle` compiles properly. I've noticed that the `nettle` package has a configuration option to choose between `gmp` (PACKAGE_libnettle) and `mini-gmp` (LIBNETTLE_MINI). `gmp` depends on `libgmp` (PACKAGE_libgmp) whereas `mini-gmp` explicitly excludes it. The description is: `use mini-gmp instead of gmp; the library will be much smaller at a 10x performance penalty. Note that this option may have side effects to programs that link to both nettle and gmp`. So i've switched to full gmp, cleared the buildsystem (make dirclean) and recompiled again. Now it mocks about missing `tpmtool`.
I will see if i can handle injecting gmp-devel into the build system.
```
cp -fpR /media/LinuxPart2/OpenWRT/openwrt-compile/build_dir/target-x86_64_musl/gnutls-3.6.12/ipkg-install/usr/bin/tpmtool /media/LinuxPart2/OpenWRT/openwrt-compile/build_dir/target-x86_64_musl/gnutls-3.6.12/ipkg-x86_64/gnutls-utils/usr/bin/
cp: cannot stat '/media/LinuxPart2/OpenWRT/openwrt-compile/build_dir/target-x86_64_musl/gnutls-3.6.12/ipkg-install/usr/bin/tpmtool': No such file or directory
make[3]: *** [Makefile:257: /media/LinuxPart2/OpenWRT/openwrt-compile/bin/packages/x86_64/packages/gnutls-utils_3.6.12-1_x86_64.ipk] Error 1
```

BlueMax, 2020-03-30:
This is not a bug.
The reason for the compile errors is `./scripts/feeds install -d y` especially the `-d y`. Installing packages like this not only enables them but also enables all suboptions. For GnuTLS that's critical as it includes 'tpm' and 'use external libtasn1' that might lead to the errors above.
It compiles properly if you keep packages on default suboptions. Sorry for any inconvenience.

closed 2020-03-30 by BlueMax