openwrt/packages #11685: vpn-policy-routing

dannykorpan, 2020-03-28:
Maintainer:@stangri
Environment:
arch ARMv7 Processor rev 5 (v7l)
model AVM FRITZ!Box 4040
firmware OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.057.55219-13dd17f

Description:
In README of the packages there is info about using DNSMASQ ipset:  Another benefit of using dnsmasq's ipset is that it also automatically adds third-level domains to the ipset: if domain.com is added to the policy, this policy will affect all *.domain.com subdomains.
But if I'm adding e.g. derstandard.at, www.derstandard.at is not routing trough the vpn, only derstandard.at.
Is there anything I'm doing wrong or I'm not understand?
In Remote addresses / domains there is the Domain: derstandard.at

Thanks for your help.

Kind Regards

stangri, 2020-03-29:
Please check the rest of the readme for the information to be submitted with any request for help.

stangri, 2021-04-20:
@aparcar @jow- @neheb  as I don't have merge rights for packages, I can't close this.

closed 2021-04-20 by neheb