openwrt/packages #11689: dawn fails to compile with newer musl

neheb, 2020-03-29:
Maintainer: @PolynomialDivision 

```
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/dawn-2020-03-28-b967c1e0/src/include/dawn_iwinfo.h:16:21: error: unknown type name '__uint8_t'; did you mean 'u_int8_t'?
   16 | int get_rssi_iwinfo(__uint8_t *client_addr);
      |                     ^~~~~~~~~
      |                     u_int8_t
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/dawn-2020-03-28-b967c1e0/src/include/dawn_iwinfo.h:37:26: error: unknown type name '__uint8_t'; did you mean 'u_int8_t'?
   37 | int get_bandwidth_iwinfo(__uint8_t *client_addr, float *rx_rate, float *tx_rate);
      |                          ^~~~~~~~~
      |                          u_int8_t
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/dawn-2020-03-28-b967c1e0/src/include/dawn_iwinfo.h:46:26: error: unknown type name '__uint8_t'; did you mean 'u_int8_t'?
   46 | int compare_essid_iwinfo(__uint8_t *bssid_addr, __uint8_t *bssid_addr_to_compare);
      |                          ^~~~~~~~~
      |                          u_int8_t
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/dawn-2020-03-28-b967c1e0/src/include/dawn_iwinfo.h:46:49: error: unknown type name '__uint8_t'; did you mean 'u_int8_t'?
   46 | int compare_essid_iwinfo(__uint8_t *bssid_addr, __uint8_t *bssid_addr_to_compare);
      |                                                 ^~~~~~~~~
      |                                                 u_int8_t

```

In my branch, I've removed several glibc headers from musl. As a result, this happens.

The __int types are glibc extensions.

PolynomialDivision, 2020-03-29:
Thanks. I removed the extensions. Could u try to compile or point to your branch so I can compile myself?

neheb, 2020-03-29:
Just do

```
curl https://github.com/neheb/openwrt/commit/0e2281ab90433138ac0210602fc57b01ba6bf161.patch | git am
```

and recompile the toolchain.

Locally I edited the .h and .c files to remove the __ prefix. Compiled fine.

PolynomialDivision, 2020-03-29:
```cpp
/tmp/Build/nick/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/include/sys/cdefs.h:1:2: error: #warning usage of non-standard #include <sys/cdefs.h> is deprecated [-Werror=cpp]
 #warning usage of non-standard #include <sys/cdefs.h> is deprecated
  ^~~~~~~
cc1: all warnings being treated as errors
make[8]: *** [gload.lo] Error 1
```
I can not compile the toolchain?

neheb, 2020-03-29:
Are you including that header?

neheb, 2020-03-29:
A quick google search shows the error being in gcc. I'm curious why -Werror is being passed.

PolynomialDivision, 2020-03-29:
> A quick google search shows the error being in gcc. I'm curious why -Werror is being passed.

I use werror in my [cmakelist.txt](https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/CMakeLists.txt#L6)
Is that problematic? Right now, I make a clean compile of the toolchain again.

PolynomialDivision, 2020-03-29:
@neheb :
I make a clean compile of the toolchain without any feeds... I'm still getting the same error.

If you look at `toolchain-mips_24kc_gcc-8.4.0_musl/include/sys/glibc-types.h:4`:
```cpp
In file included from /tmp/Build/nick/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/include/sys/glibc-types.h:4,
                 from /tmp/Build/nick/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/include/features.h:46,
                 from /tmp/Build/nick/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/include/limits.h:4,
                 from /tmp/Build/nick/openwrt/build_dir/toolchain-mips_24kc_gcc-8.4.0_musl/gcc-8.4.0-final/gcc/include-fixed/limits.h:194,
                 from /tmp/Build/nick/openwrt/build_dir/toolchain-mips_24kc_gcc-8.4.0_musl/gcc-8.4.0-final/gcc/include-fixed/syslimits.h:7,
                 from /tmp/Build/nick/openwrt/build_dir/toolchain-mips_24kc_gcc-8.4.0_musl/gcc-8.4.0-final/gcc/include-fixed/limits.h:34,
                 from /tmp/Build/nick/openwrt/build_dir/toolchain-mips_24kc_gcc-8.4.0_musl/gcc-8.4.0/libatomic/libatomic_i.h:35,
                 from /tmp/Build/nick/openwrt/build_dir/toolchain-mips_24kc_gcc-8.4.0_musl/gcc-8.4.0/libatomic/gload.c:25:
/tmp/Build/nick/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/include/sys/cdefs.h:1:2: error: #warning usage of non-standard #include <sys/cdefs.h> is deprecated [-Werror=cpp]
 #warning usage of non-standard #include <sys/cdefs.h> is deprecated
```

```cpp
#ifndef __MUSL_GLIBC_TYPES_H
#define __MUSL_GLIBC_TYPES_H

#include <sys/cdefs.h>

/* Convenience types.  */
typedef unsigned char __u_char;
typedef unsigned short int __u_short;
typedef unsigned int __u_int;
typedef unsigned long int __u_long;

/* Fixed-size types, underlying types depend on word size and compiler.  */
typedef signed char __int8_t;
typedef unsigned char __uint8_t;
typedef signed short int __int16_t;
typedef unsigned short int __uint16_t;
typedef signed int __int32_t;
typedef unsigned int __uint32_t;
#if __WORDSIZE == 64
typedef signed long int __int64_t;
typedef unsigned long int __uint64_t;
#else
__extension__ typedef signed long long int __int64_t;
__extension__ typedef unsigned long long int __uint64_t;
#endif

#define __off64_t off_t
#define __loff_t off_t
typedef char *__caddr_t;
#define __locale_t locale_t

#define __gid_t gid_t
#define __uid_t uid_t

#endif
```

neheb, 2020-03-29:
No the issue is with GCC. Then again, I compile GCC9 locally...

I will do some tests when I can.

PolynomialDivision, 2020-03-29:
> No the issue is with GCC. Then again, I compile GCC9 locally...
> 
> I will do some tests when I can.

I have:
`gcc version 9.3.0 (Arch Linux 9.3.0-1) `

closed 2020-03-29 by hnyman

neheb, 2020-03-29:
Hrm.....Maybe it has to do with the host GCC after all. In which case I'll have to revert my change...

neheb, 2020-03-29:
> > No the issue is with GCC. Then again, I compile GCC9 locally...
> > I will do some tests when I can.
> 
> On Build-Server:
> 
> ```
> gcc version 4.8.5 20150623 (Red Hat 4.8.5-39) (GCC) 
> ```
> 
> `CentOS`

what version of centos is that?

neheb, 2020-03-29:
Anyway, latest version compiles file locally. This is on Fedora 31.

neheb, 2020-03-30:
Downloaded a CentOS 7 ISO. I cannot reproduce that failure.

PolynomialDivision, 2020-03-30:
> Downloaded a CentOS 7 ISO. I cannot reproduce that failure.

`CentOS Linux release 7.7.1908 (Core)`

What did u do?

I will checkout your repro and commit and will restart building. :)

PolynomialDivision, 2020-03-30:
Okay what I did
```
git clone https://github.com/neheb/openwrt
git checkout mangix
make menuconfig
make -j128
[Fails]
make -j1 V=s
```cpp
Toolchain seems now fine because it fails at some package
```
/tmp/Build/nick/testing/openwrt/build_dir/target-mips_74kc+dsp2_musl/fstools-2020-01-21-deb745f8/libfstools/overlay.c: In function 'foreachdir':
/tmp/Build/nick/testing/openwrt/build_dir/target-mips_74kc+dsp2_musl/fstools-2020-01-21-deb745f8/libfstools/overlay.c:92:49: error: 'GLOB_ONLYDIR' undeclared (first use in this function); did you mean 'GLOB_TILDE'?
  if (!glob(globdir, GLOB_NOESCAPE | GLOB_MARK | GLOB_ONLYDIR, NULL, &gl))
                                                 ^~~~~~~~~~~~
                                                 GLOB_TILDE
/tmp/Build/nick/testing/openwrt/build_dir/target-mips_74kc+dsp2_musl/fstools-2020-01-21-deb745f8/libfstools/overlay.c:92:49: note: each undeclared identifier is reported only once for each function it appears in
```

neheb, 2020-03-30:
For the purpose of this PR, you'll want do do

```
make package/dawn/compile
```

That fstools failure is a different story. I've submitted a patch for it.

PR #11702 (2020-03-29): dawn: remove glibc headers extensions
Fixes #11689.
commit 849e6df2 (2020-03-29): dawn: remove glibc headers extensions

Fixes #11689