openwrt/packages #11701: wsdd2: init script problem

dongjuanyong, 2020-03-29:
Maintainer: @Andy2244 
Environment: x86_64, OpenWrt snapshot
Description:
Wsdd2 can't start. Here is the log:
```
daemon.info procd: Instance wsdd2::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
```
When I looked into the init script, I found the command line to run was:
```
/usr/bin/wsdd2 -w -i br-lan -N OpenWrt -G WORKGROUP -b vendor:default-string-default-string,model:Intel(R)_Pentium(R)_CPU_N3710_@_1.60GHz_:_4_Core_4_Thread
```
The colon in the model string caused the problem, since the original string in /tmp/sysinfo/model was:
```
Intel(R) Pentium(R) CPU N3710 @ 1.60GHz : 4 Core 4 Thread
```
Thus I think in the init script, we must handle colons in these strings the same way as spaces.

neheb, 2020-03-30:
Sounds like a quoting bug. Try this:

```
diff --git a/net/wsdd2/files/wsdd2.init b/net/wsdd2/files/wsdd2.init
index 24595e0f3..96d79f8eb 100644
--- a/net/wsdd2/files/wsdd2.init
+++ b/net/wsdd2/files/wsdd2.init
@@ -33,19 +33,19 @@ start_service() {
 	local nb_name
 	nb_name="$(grep -i 'netbios name' $SMB_CONF | awk -F'=' '{print $2}' | tr -d ' \n')"
 	if [ -n "$nb_name" ]; then
-		NB_PARM="-N $nb_name"
+		NB_PARM="$nb_name"
 	else
 		local hostname
 		hostname="$(cat /proc/sys/kernel/hostname)"
-		NB_PARM="-N $hostname"
+		NB_PARM="$hostname"
 	fi
 
 	local wg_name
 	wg_name="$(grep -i 'workgroup' $SMB_CONF | awk -F'=' '{print $2}' | tr -d ' \n')"
 	if [ -n "$wg_name" ]; then
-		WG_PARM="-G $wg_name"
+		WG_PARM="$wg_name"
 	else
-		WG_PARM="-G WORKGROUP"
+		WG_PARM="WORKGROUP"
 	fi
 
 	# resolve lan interface (BUG: No multi-interface binds atm)
@@ -67,17 +67,21 @@ start_service() {
 	fi
 	if [ -n "$board_vendor" ] && [ -n "$board_model" ]; then
 		if [ -n "$board_sku" ]; then
-			BI_PARM="-b vendor:$board_vendor,model:$board_model,sku:$board_sku"
+			BI_PARM="vendor:$board_vendor,model:$board_model,sku:$board_sku"
 		else
-			BI_PARM="-b vendor:$board_vendor,model:$board_model"
+			BI_PARM="vendor:$board_vendor,model:$board_model"
 		fi
 	fi
 
 	procd_open_instance
-    procd_set_param command /usr/bin/wsdd2 -w $BIND_IF_PARM $NB_PARM $WG_PARM $BI_PARM
-    procd_set_param respawn
-    procd_set_param file $SMB_CONF
-    procd_close_instance
+	procd_set_param command /usr/bin/wsdd2
+	procd_append_param command "$BIND_IF_PARM"
+	procd_append_param command -N "$NB_PARM"
+	procd_append_param command -G "$WG_PARM"
+	procd_append_param command -b "$BI_PARM"
+	procd_set_param respawn
+	procd_set_param file $SMB_CONF
+	procd_close_instance
 }
 
 service_triggers() {
```

dongjuanyong, 2020-03-30:
No luck:
```
daemon.info procd: Instance wsdd2::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
```
However, this simply works:
```
--- a/wsdd2.init        2020-03-30 09:24:24.568590700 +0800
+++ b/wsdd2.init        2020-03-30 09:25:09.258704700 +0800
@@ -63,7 +63,7 @@
                board_sku="$(cat /tmp/sysinfo/board_name | awk -F',' '{print $2}' | tr ' ' '_' | tr -d ' \n')"
        fi
        if [ -e /tmp/sysinfo/model ]; then
-               board_model="$(cat /tmp/sysinfo/model | tr ' ' '_' | tr -d ' \n')"
+               board_model="$(cat /tmp/sysinfo/model | awk -F':' '{print $1}' | tr ' ' '_' | tr -d ' \n')"
        fi
        if [ -n "$board_vendor" ] && [ -n "$board_model" ]; then
                if [ -n "$board_sku" ]; then
```

Andy2244, 2020-03-30:
Yeah sorry missed the ":" special char, just setup a PR for it. I also could not get the quoting to work correctly, thats why i ended up with the char replace suff.

PS: Still kinda busy atm, so have no time atm for openWRT. Will see how things develop in the next week.

dongjuanyong, 2020-03-31:
Tested and #11712 works for me. Thank you.

closed 2020-03-31 by dongjuanyong