openwrt/packages #11703: mwan3: rule not working for IPv6

brianjmurrell, 2020-03-29:
Maintainer: @feckert 
Environment: WNDR4300, 19.07.2

Description:
Rules with IPv6 addresses don't seem to be causing appropriate entries to the IPv6 mangle table.

```
config rule 'NNTP_IPv6'
	option dest_ip '2001:1234:5678:119::20/127'
	option dest_port '119'
	option proto 'tcp'
	option sticky '0'
	option use_policy 'wan1_only'
```

Doesn't cause the necessary:

```
Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    5   300 mwan3_policy_wan1_only  tcp  --  *      *       ::/0            2001:1234:5678:119::20/127      multiport sports 0:65535 multiport dports 119 mark match 0x0/0x3f00 /* Teksavvy_NNTP_IPv6 */
```

rule to be added.

feckert, 2020-03-30:
Which rule is added and how does it look?
Or what is the difference to your rule that you have added per hand?

brianjmurrell, 2020-03-30:
> Which rule is added and how does it look?

That the problem.  No rule is added at all.

> Or what is the difference to your rule that you have added per hand?

No difference.  There is no rule added at all.

feckert, 2020-03-30:
Are there any other IPv6 rules added by mwan3?
Could you send me your output of `iptables-save`?
Please remove secret values before sending.

brianjmurrell, 2020-03-30:
> Are there any other IPv6 rules added by mwan3?

Yes:

```
    0     0 mwan3_policy_wan1_only  tcp      *      *       ::/0                 ::/0                 multiport sports 0:65535 multiport dports 25 mark match 0x0/0x3f00 /* SMTP */
```

which comes from:

```
config rule 'SMTP'
	option dest_port '25'
	option proto 'tcp'
	option sticky '0'
	option use_policy 'wan1_only'
```

> Could you send me your output of `iptables-save`?

You probably want `ip6tables-save` yes?  Here's the `mangle` table output from that:

```
*mangle
:PREROUTING ACCEPT [3838598:2460407743]
:INPUT ACCEPT [75298:6809352]
:FORWARD ACCEPT [3607423:2436699790]
:OUTPUT ACCEPT [105268:18257538]
:POSTROUTING ACCEPT [3707951:2454601277]
:mwan3_connected - [0:0]
:mwan3_hook - [0:0]
:mwan3_iface_in_henet - [0:0]
:mwan3_iface_in_wan1_6 - [0:0]
:mwan3_ifaces_in - [0:0]
:mwan3_policy_balanced - [0:0]
:mwan3_policy_henet_only - [0:0]
:mwan3_policy_wan0_henet_wan1 - [0:0]
:mwan3_policy_wan0_only - [0:0]
:mwan3_policy_wan0_wan1 - [0:0]
:mwan3_policy_wan1_henet_wan0 - [0:0]
:mwan3_policy_wan1_only - [0:0]
:mwan3_rules - [0:0]
-A PREROUTING -j mwan3_hook
-A OUTPUT -j mwan3_hook
-A mwan3_connected -m set --match-set mwan3_connected dst -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 133 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 134 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 136 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 137 -j RETURN
-A mwan3_hook -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_ifaces_in
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_connected
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_rules
-A mwan3_hook -j CONNMARK --save-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected
-A mwan3_iface_in_henet -i 6in4-henet -m set --match-set mwan3_connected_v6 src -m mark --mark 0x0/0x3f00 -m comment --comment default -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_iface_in_henet -i 6in4-henet -m mark --mark 0x0/0x3f00 -m comment --comment henet -j MARK --set-xmark 0x300/0x3f00
-A mwan3_iface_in_wan1_6 -i pppoe-wan1 -m set --match-set mwan3_connected_v6 src -m mark --mark 0x0/0x3f00 -m comment --comment default -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_iface_in_wan1_6 -i pppoe-wan1 -m mark --mark 0x0/0x3f00 -m comment --comment wan1_6 -j MARK --set-xmark 0x500/0x3f00
-A mwan3_ifaces_in -m mark --mark 0x0/0x3f00 -j mwan3_iface_in_henet
-A mwan3_ifaces_in -m mark --mark 0x0/0x3f00 -j mwan3_iface_in_wan1_6
-A mwan3_policy_balanced -m mark --mark 0x0/0x3f00 -m statistic --mode random --probability 0.39999999991 -m comment --comment "wan1_6 2 5" -j MARK --set-xmark 0x500/0x3f00
-A mwan3_policy_balanced -m mark --mark 0x0/0x3f00 -m comment --comment "henet 3 3" -j MARK --set-xmark 0x300/0x3f00
-A mwan3_policy_henet_only -m mark --mark 0x0/0x3f00 -m comment --comment "henet 3 3" -j MARK --set-xmark 0x300/0x3f00
-A mwan3_policy_wan0_henet_wan1 -m mark --mark 0x0/0x3f00 -m comment --comment "henet 3 3" -j MARK --set-xmark 0x300/0x3f00
-A mwan3_policy_wan0_only -m mark --mark 0x0/0x3f00 -m comment --comment unreachable -j MARK --set-xmark 0x3e00/0x3f00
-A mwan3_policy_wan1_henet_wan0 -m mark --mark 0x0/0x3f00 -m comment --comment "wan1_6 2 2" -j MARK --set-xmark 0x500/0x3f00
-A mwan3_policy_wan1_only -m mark --mark 0x0/0x3f00 -m comment --comment "wan1_6 2 2" -j MARK --set-xmark 0x500/0x3f00
-A mwan3_rules -p tcp -m multiport --sports 0:65535 -m multiport --dports 25 -m mark --mark 0x0/0x3f00 -m comment --comment SMTP -j mwan3_policy_wan1_only
-A mwan3_rules -m mark --mark 0x0/0x3f00 -m comment --comment default_rule -j mwan3_policy_wan0_wan1
COMMIT
```

brianjmurrell, 2020-03-31:
Was my last update helpful?

feckert, 2020-04-01:
Sorry I could not reproduce your issue!

```
Generated by ip6tables-save v1.8.4 on Wed Apr  1 11:32:50 2020
*mangle
:PREROUTING ACCEPT [2:304]
:INPUT ACCEPT [2:304]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
:mwan3_connected - [0:0]
:mwan3_hook - [0:0]
:mwan3_ifaces_in - [0:0]
:mwan3_policy_wan_only - [0:0]
:mwan3_policy_wan_wwan - [0:0]
:mwan3_policy_wwan_only - [0:0]
:mwan3_rules - [0:0]
:qos_Default - [0:0]
:qos_Default_ct - [0:0]
-A PREROUTING -j mwan3_hook
-A FORWARD -o eth0 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
-A OUTPUT -j mwan3_hook
-A mwan3_connected -m set --match-set mwan3_connected dst -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 133 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 134 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 136 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 137 -j RETURN
-A mwan3_hook -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_ifaces_in
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_connected
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_rules
-A mwan3_hook -j CONNMARK --save-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected
-A mwan3_policy_wan_only -m mark --mark 0x0/0x3f00 -m comment --comment unreachable -j MARK --set-xmark 0x3e00/0x3f00
-A mwan3_policy_wan_wwan -m mark --mark 0x0/0x3f00 -m comment --comment unreachable -j MARK --set-xmark 0x3e00/0x3f00
-A mwan3_policy_wwan_only -m mark --mark 0x0/0x3f00 -m comment --comment unreachable -j MARK --set-xmark 0x3e00/0x3f00
-A mwan3_rules -d 2001:1234:5678:119::20/127 -p tcp -m multiport --sports 0:65535 -m multiport --dports 119 -m mark --mark 0x0/0x3f00 -m comment --comment NNTP_IPv6 -j mwan3_policy_wan_wwan
-A mwan3_rules -m mark --mark 0x0/0x3f00 -m comment --comment default_rule -j mwan3_policy_wan_wwan
-A qos_Default -j CONNMARK --restore-mark --nfmask 0xf --ctmask 0xf
-A qos_Default -m mark --mark 0x0/0xf -j qos_Default_ct
-A qos_Default -p udp -m mark --mark 0x0/0xf0 -m length --length 0:500 -j MARK --set-xmark 0x22/0xff
-A qos_Default -p icmp -j MARK --set-xmark 0x11/0xff
-A qos_Default -p tcp -m mark --mark 0x0/0xf0 -m tcp --sport 1024:65535 --dport 1024:65535 -j MARK --set-xmark 0x44/0xff
-A qos_Default -p udp -m mark --mark 0x0/0xf0 -m udp --sport 1024:65535 --dport 1024:65535 -j MARK --set-xmark 0x44/0xff
-A qos_Default -j CONNMARK --save-mark --nfmask 0xff --ctmask 0xff
-A qos_Default_ct -p udp -m mark --mark 0x0/0xf -m comment --comment SIP -m udp -m multiport --ports 5060 -j MARK --set-xmark 0x22/0xff
-A qos_Default_ct -p tcp -m mark --mark 0x0/0xf -m comment --comment SIP -m tcp -m multiport --ports 5060 -j MARK --set-xmark 0x22/0xff
-A qos_Default_ct -p udp -m mark --mark 0x0/0xf -m comment --comment "RTP (Voice)" -m udp -m multiport --ports 10000:20000 -j MARK --set-xmark 0x22/0xff
-A qos_Default_ct -j CONNMARK --save-mark --nfmask 0xff --ctmask 0xff
COMMIT
# Completed on Wed Apr  1 11:32:50 2020
```

brianjmurrell, 2020-04-03:
Problem turned out to be **Rule [redacted]_NNTP_IPv6 exceeds max of 15 chars. Not setting rule**

Buried in the syslog though.  Would be better if this were exposed to the `mwan3` command when, say `mwan3 start` is run.

closed 2020-04-03 by brianjmurrell

feckert, 2020-04-03:
Thanks for the feedback. I would set it to my tasks, to write a message additional to the stderr in `mwan3 start` command