openwrt/packages #11704: Snort:

DaveRush82, 2020-03-29:
Maintainer: @flyn-org 
Environment: raspberry pi 4B. aarch64_cortex-a72. Snapshot installed since not supported in release yet.  Currently running the March 28 build: r12752-258f070d1a. 

Description: Snort fails to start, with error regarding "endrpcent: symbol not found":

```
Loading dynamic engine /usr/lib/snort_dynamicengine/libsf_engine.so... done
Loading all dynamic preprocessor libs from /usr/lib/snort_dynamicpreprocessor...
  Loading dynamic preprocessor library /usr/lib/snort_dynamicpreprocessor/libsf_appid_preproc.so... ERROR: Failed to load /usr/lib/snort_dynamicpreprocessor/libsf_appid_preproc.so: Error relocating /usr/lib/snort_dynamicpreprocessor/libsf_appid_preproc.so: endrpcent: symbol not found
Fatal Error, Quitting..
```

ldd against that .so returns the following:
```
root@OpenWrt:/etc/snort# ldd /usr/lib/snort_dynamicpreprocessor/libsf_appid_preproc.so
	ldd (0x7fac6d2000)
	libluajit-5.1.so.2 => /usr/lib/libluajit-5.1.so.2 (0x7fac08c000)
	libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x7fac055000)
	libdnet.so.1 => /usr/lib/libdnet.so.1 (0x7fac036000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x7fabfef000)
	libpcap.so.1 => /usr/lib/libpcap.so.1 (0x7fabfa5000)
	libuuid.so.1 => /usr/lib/libuuid.so.1 (0x7fabf8e000)
	libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1 (0x7fabd29000)
	libdaq.so.4 => /usr/lib/libdaq.so.4 (0x7fabd14000)
	libz.so.1 => /usr/lib/libz.so.1 (0x7fabcef000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fabccc000)
	libc.so => ldd (0x7fac6d2000)
Error relocating /usr/lib/snort_dynamicpreprocessor/libsf_appid_preproc.so: endrpcent: symbol not found
Error relocating /usr/lib/snort_dynamicpreprocessor/libsf_appid_preproc.so: getrpcent: symbol not found
root@OpenWrt:/etc/snort#
```

I see libtirpc as a dependency, should I have opened this against that package instead?

DaveRush82, 2020-03-29:
Forgot a few details:
Snort is at version 2.9.15-1.
libtirpc is at 1.2.5-2.

flyn-org, 2021-08-06:
@DaveRush82, does this problem still exist? I have updated the snort package since your report.

tduque, 2021-08-17:
I still have this very same issue.

flyn-org, 2021-08-17:
@tduque, what snort package version and what OpenWrt version?

tduque, 2021-08-19:
I'm using:
target: x86_64 
C-Linker: Musl
Openwrt v21.02.0-rc3
snort: 2.9.17-2

Fever-Wits, 2021-09-17:
Hello,
I have the same problem

ldd /usr/lib/snort_dynamicpreprocessor/libsf_appid_preproc.so
        ldd (0x7f8ca15000)
        libluajit-5.1.so.2 => /usr/lib/libluajit-5.1.so.2 (0x7f8c3ce000)
        libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x7f8c395000)
        libdnet.so.1 => /usr/lib/libdnet.so.1 (0x7f8c375000)
        libpcre.so.1 => /usr/lib/libpcre.so.1 (0x7f8c32e000)
        libpcap.so.1 => /usr/lib/libpcap.so.1 (0x7f8c2e4000)
        libuuid.so.1 => /usr/lib/libuuid.so.1 (0x7f8c2cd000)
        libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1 (0x7f8c062000)
        libdaq.so.4 => /usr/lib/libdaq.so.4 (0x7f8c04d000)
        libz.so.1 => /usr/lib/libz.so.1 (0x7f8c028000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f8c005000)
        libc.so => ldd (0x7f8ca15000)
Error relocating /usr/lib/snort_dynamicpreprocessor/libsf_appid_preproc.so: endrpcent: symbol not found
Error relocating /usr/lib/snort_dynamicpreprocessor/libsf_appid_preproc.so: getrpcent: symbol not found

snort - 2.9.17-2
libtirpc - 1.3.2-1
DISTRIB_DESCRIPTION='OpenWrt 21.02.0 r16279-5cc0535800'
DISTRIB_TARGET='bcm27xx/bcm2711'
DISTRIB_ARCH='aarch64_cortex-a72'

Regards,

lordneo, 2021-12-02:
Still pops up the same error.

flyn-org, 2021-12-14:
I just built a new version (https://github.com/openwrt/packages/pull/17329), and now I see this:
```
Error relocating /usr/bin/snort: gettid: symbol not found
```
I would be happy to migrate responsibility to a new maintainer. I use Snort 3 now. The `gettid` function should be in Musl 1.2.2 according to https://musl.libc.org/releases.html.

keithkostol, 2022-11-29:
I ran into this as well and was able to fix the linking error.  Snort will now start and run for me on openwrt 22.03 with snort version 2.9.19.  I updated `003-include-tirpc.patch` to patch snort's `configure.in` script to add tirpc as a linked library.  Here's the updated patch file:

```
--- a/src/dynamic-preprocessors/appid/service_plugins/service_rpc.c
+++ b/src/dynamic-preprocessors/appid/service_plugins/service_rpc.c
@@ -32,6 +32,7 @@
 #include "flow.h"
 #include "service_api.h"
 
+#include <tirpc/rpc/rpcent.h>
 #if defined(FREEBSD) || defined(OPENBSD)
 #include "rpc/rpc.h"
 #endif
--- a/configure.in
+++ b/configure.in
@@ -890,6 +890,11 @@ if test "x$enable_dlclose" = "xno"; then
 fi
 
 ##################################################
+# openwrt needs libtirpc to be linked
+##################################################
+LIBS="${LIBS} -ltirpc"
+
+##################################################
 # Fedora 28+ does not have inbuilt SunRPC support#
 # in glibc and is separately availble in tirpc   #
 # package. Make sure we've got the library and   #
```

I'm not sure if this is a robust enough solution for openwrt or not.  The configure.in script has a bit of code surrounding the other libraries that I am unfamiliar with, though this patch would only apply to openwrt builds anyway.