openwrt/packages #11731: golang: Enable build on aarch64(arm64) host

cmheia, 2020-04-01:
Maintainer: @jefferyto
Environment: Host is RK3328(aarch64) box, SDK is self build from OpenWrt v19.07.2 source

Description:
Before building any app written in go, a bootstrap blob is need to generate a host go compiler.
Currently this is `go1.4-bootstrap-20171003.tar.gz`.
As we all known, arm64 is not support in go 1.4.
But I'm perfer to build ipks in my small arm box, and I managed to work it out with a few problem to resolve.

Here is what I do:
 1. Build `go-linux-arm64-bootstrap-1.14.1.tbz` in an amd64 PC
 2. Extract `go-linux-arm64-bootstrap-1.14.1.tbz` as `/data/go-linux-arm64-bootstrap-1.14.1/` in arm box
 3. Add an option `GOLANG_BOOTSTRAP_BASE_DIR` to `lang/golang/golang/Makefile`
 4. Add `CONFIG_GOLANG_BOOTSTRAP_BASE_DIR="/data/go-linux-arm64-bootstrap-1.14.1"` to `.config` in SDK in arm box

After doing that, building golang app no longer failed with `go-bootstrap cannot be installed on linux/arm64.  Stop.`.

**BUT** there are two problem not resolved:

 - If remove the line after `define Host/Compile` ie **`# Build GO`**, build will stop with `extraneous 'endif'.  Stop.`
 - If `CONFIG_GOLANG_BOOTSTRAP_BASE_DIR` is set to empty, building golang app not stop at `go-bootstrap cannot be installed on linux/arm64`.

Any help @jefferyto ?

---

Here is what I do to `lang/golang/golang/Makefile`:
```
diff --git a/lang/golang/golang/Makefile b/lang/golang/golang/Makefile
index dd50c2d32..b5d06b3ea 100644
--- a/lang/golang/golang/Makefile
+++ b/lang/golang/golang/Makefile
@@ -30,6 +30,17 @@ PKG_BUILD_DIR:=$(BUILD_DIR)/go-$(PKG_VERSION)
 PKG_BUILD_PARALLEL:=1
 PKG_USE_MIPS16:=0
 
+define Package/golang/config
+	menu "Configuration"
+
+	config GOLANG_BOOTSTRAP_BASE_DIR
+		string
+		prompt "golng bootstrap base directory"
+		default ""
+
+	endmenu
+endef
+
 PKG_GO_WORK_DIR:=$(PKG_BUILD_DIR)/.go_work
 PKG_GO_HOST_CACHE_DIR:=$(PKG_GO_WORK_DIR)/host_cache
 PKG_GO_TARGET_CACHE_DIR:=$(PKG_GO_WORK_DIR)/target_cache
@@ -62,9 +73,13 @@ BOOTSTRAP_SOURCE:=go1.4-bootstrap-20171003.tar.gz
 BOOTSTRAP_SOURCE_URL:=$(GO_SOURCE_URLS)
 BOOTSTRAP_HASH:=f4ff5b5eb3a3cae1c993723f3eab519c5bae18866b5e5f96fe1102f0cb5c3e52
 
+ifeq ($(CONFIG_GOLANG_BOOTSTRAP_BASE_DIR),)
 BOOTSTRAP_BUILD_DIR:=$(HOST_BUILD_DIR)/.go_bootstrap
 BOOTSTRAP_WORK_DIR:=$(BOOTSTRAP_BUILD_DIR)/.go_work
 BOOTSTRAP_CACHE_DIR:=$(BOOTSTRAP_WORK_DIR)/cache
+else
+BOOTSTRAP_BUILD_DIR:=$(CONFIG_GOLANG_BOOTSTRAP_BASE_DIR)
+endif
 
 BOOTSTRAP_GO_VALID_OS_ARCH:= \
   darwin_386     darwin_amd64 \
@@ -153,31 +168,41 @@ define Download/golang-bootstrap
   URL:=$(BOOTSTRAP_SOURCE_URL)
   HASH:=$(BOOTSTRAP_HASH)
 endef
-$(eval $(call Download,golang-bootstrap))
 
+ifeq ($(CONFIG_GOLANG_BOOTSTRAP_BASE_DIR),)
+$(eval $(call Download,golang-bootstrap))
 $(eval $(call GoCompiler/AddProfile,Bootstrap,$(BOOTSTRAP_BUILD_DIR),,bootstrap,$(GO_HOST_OS_ARCH)))
+endif
+
 $(eval $(call GoCompiler/AddProfile,Host,$(HOST_BUILD_DIR),$(HOST_GO_PREFIX),$(HOST_GO_VERSION_ID),$(GO_HOST_OS_ARCH)))
 $(eval $(call GoCompiler/AddProfile,Package,$(PKG_BUILD_DIR),$(GO_TARGET_PREFIX),$(GO_TARGET_VERSION_ID),$(GO_OS_ARCH)))
 
 define Host/Prepare
 	$(call Host/Prepare/Default)
+  ifeq ($(CONFIG_GOLANG_BOOTSTRAP_BASE_DIR),)
 	mkdir -p $(BOOTSTRAP_BUILD_DIR)
 	$(BOOTSTRAP_UNPACK)
+  endif
 endef
 
 define Host/Compile
+  # Build GO
+  ifeq ($(CONFIG_GOLANG_BOOTSTRAP_BASE_DIR),)
 	$(call GoCompiler/Bootstrap/CheckHost,$(BOOTSTRAP_GO_VALID_OS_ARCH))
+  endif
 	$(call GoCompiler/Host/CheckHost,$(HOST_GO_VALID_OS_ARCH))
 
 	mkdir -p \
 		$(BOOTSTRAP_CACHE_DIR) \
 		$(HOST_GO_CACHE_DIR)
 
+  ifeq ($(CONFIG_GOLANG_BOOTSTRAP_BASE_DIR),)
 	$(call GoCompiler/Bootstrap/Make, \
 		GOCACHE=$(BOOTSTRAP_CACHE_DIR) \
 		CC=$(HOSTCC_NOCACHE) \
 		CXX=$(HOSTCXX_NOCACHE) \
 	)
+  endif
 
 	$(call GoCompiler/Host/Make, \
 		GOROOT_BOOTSTRAP=$(BOOTSTRAP_BUILD_DIR) \
```

jefferyto, 2020-04-01:
I'm not able to help you with your custom modifications at the moment (you may want to ask on the [forum](https://forum.openwrt.org/)), but being able to use an external bootstrap Go is an interesting option and I may add a similar config option in the future.

jefferyto, 2020-05-18:
Support for an external bootstrap Go has been added to master (49a2ff493b46029e5acc1a902a1e850307bc20bc) - can you see if this works for you?

I don't think it will be backported to 19.07 but it shouldn't be too hard to manually apply the patch (or cherry pick the commit) in your own repo.

cmheia, 2020-05-18:
> Support for an external bootstrap Go has been added to master ([49a2ff4](https://github.com/openwrt/packages/commit/49a2ff493b46029e5acc1a902a1e850307bc20bc)) - can you see if this works for you?
> 
> I don't think it will be backported to 19.07 but it shouldn't be too hard to manually apply the patch (or cherry pick the commit) in your own repo.

---

Works for me, thanks.
Build test on 19.07.2:

---
```diff
diff --git a/lang/golang/golang/Makefile b/lang/golang/golang/Makefile
index dd50c2d32..bf6e3f635 100644
--- a/lang/golang/golang/Makefile
+++ b/lang/golang/golang/Makefile
@@ -125,6 +125,23 @@ This package provides an assembler, compiler, linker, and compiled
 libraries for the Go programming language.
 endef
 
+define Package/golang/config
+menu "Configuration"
+
+config GOLANG_EXTERNAL_BOOTSTRAP_ROOT
+	string "External bootstrap Go root directory"
+	help
+	  Path to a working Go tree (>= Go 1.4), with bin, pkg, and src
+	  subdirectories and the Go compiler at bin/go.
+
+	  If specified, the existing Go installation will be used to
+	  compile host (buildroot) Go.
+
+	  Leave blank to compile the default bootstrap Go.
+
+endmenu
+endef
+
 define Package/golang-doc
 $(call Package/golang/Default)
   TITLE+= (documentation)
@@ -148,22 +165,29 @@ This package provides the Go programming language source files needed
 for cross-compilation.
 endef
 
+EXTERNAL_BOOTSTRAP_DIR:=$(call qstrip,$(CONFIG_GOLANG_EXTERNAL_BOOTSTRAP_ROOT))
+USE_DEFAULT_BOOTSTRAP:=$(if $(EXTERNAL_BOOTSTRAP_DIR),,1)
+
 define Download/golang-bootstrap
   FILE:=$(BOOTSTRAP_SOURCE)
   URL:=$(BOOTSTRAP_SOURCE_URL)
   HASH:=$(BOOTSTRAP_HASH)
 endef
-$(eval $(call Download,golang-bootstrap))
 
-$(eval $(call GoCompiler/AddProfile,Bootstrap,$(BOOTSTRAP_BUILD_DIR),,bootstrap,$(GO_HOST_OS_ARCH)))
 $(eval $(call GoCompiler/AddProfile,Host,$(HOST_BUILD_DIR),$(HOST_GO_PREFIX),$(HOST_GO_VERSION_ID),$(GO_HOST_OS_ARCH)))
 $(eval $(call GoCompiler/AddProfile,Package,$(PKG_BUILD_DIR),$(GO_TARGET_PREFIX),$(GO_TARGET_VERSION_ID),$(GO_OS_ARCH)))
 
-define Host/Prepare
+ifeq ($(USE_DEFAULT_BOOTSTRAP),1)
+  $(eval $(call GoCompiler/AddProfile,Bootstrap,$(BOOTSTRAP_BUILD_DIR),,bootstrap,$(GO_HOST_OS_ARCH)))
+
+  $(eval $(call Download,golang-bootstrap))
+
+  define Host/Prepare
 	$(call Host/Prepare/Default)
 	mkdir -p $(BOOTSTRAP_BUILD_DIR)
 	$(BOOTSTRAP_UNPACK)
-endef
+  endef
+endif
 
 define Host/Compile
 	$(call GoCompiler/Bootstrap/CheckHost,$(BOOTSTRAP_GO_VALID_OS_ARCH))
@@ -180,7 +204,7 @@ define Host/Compile
 	)
 
 	$(call GoCompiler/Host/Make, \
-		GOROOT_BOOTSTRAP=$(BOOTSTRAP_BUILD_DIR) \
+		GOROOT_BOOTSTRAP=$(if $(USE_DEFAULT_BOOTSTRAP),$(BOOTSTRAP_BUILD_DIR),$(EXTERNAL_BOOTSTRAP_DIR)) \
 		GOCACHE=$(HOST_GO_CACHE_DIR) \
 		CC=$(HOSTCC_NOCACHE) \
 		CXX=$(HOSTCXX_NOCACHE) \
```

closed 2020-05-28 by neheb