openwrt/packages #11741: wifidog-ng-nossl: package compile error net/netfilter/nf_nat_l3proto.h: No such file or directory

peterwillcn, 2020-04-02:
wifi-dog
```
2020-04-02T06:42:09.3332020Z /home/runner/work/openwrt/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/main.c:17:10: fatal error: net/netfilter/nf_nat_l3proto.h: No such file or directory
2020-04-02T06:42:09.3332897Z make[5]: *** [/home/runner/work/openwrt/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/main.o] Error 1
2020-04-02T06:42:09.3333777Z make[4]: *** [/home/runner/work/openwrt/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl] Error 2
2020-04-02T06:42:09.3334456Z make[3]: *** [/home/runner/work/openwrt/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/.built] Error 2
2020-04-02T06:42:09.3335026Z make[2]: *** [package/feeds/packages/wifidog-ng/compile] Error 2
2020-04-02T06:42:09.3335386Z WARNING: Firmware file ucode16_sslpn.fw not found in the mapping lists
2020-04-02T06:42:09.3335687Z WARNING: Firmware file lp1bsinitvals20.fw not found in the mapping lists
2020-04-02T06:42:09.3335982Z WARNING: Firmware file a0g1initvals9.fw not found in the mapping lists
2020-04-02T06:42:09.3336274Z WARNING: Firmware file lcn2initvals24.fw not found in the mapping lists
2020-04-02T06:42:09.3336564Z WARNING: Firmware file sslpn2bsinitvals19.fw not found in the mapping lists
2020-04-02T06:42:09.3336841Z WARNING: Firmware file ucode16_lp.fw not found in the mapping lists
2020-04-02T06:42:09.3337141Z WARNING: Firmware file lcn0initvals26.fw not found in the mapping lists
2020-04-02T06:42:09.3337430Z WARNING: Firmware file ucode21_sslpn.fw not found in the mapping lists
2020-04-02T06:42:09.3337715Z WARNING: Firmware file ucode24_mimo.fw not found in the mapping lists
2020-04-02T06:42:09.3338001Z WARNING: Firmware file ucode20_sslpn_nobt.fw not found in the mapping lists
2020-04-02T06:42:09.3338271Z WARNING: Firmware file n0initvals22.fw not found in the mapping lists
2020-04-02T06:42:09.3338562Z WARNING: Firmware file sslpn4bsinitvals22.fw not found in the mapping lists
2020-04-02T06:42:09.3338860Z WARNING: Firmware file b0g0bsinitvals9.fw not found in the mapping lists
2020-04-02T06:42:09.3339326Z WARNING: Firmware file n1bsinitvals20.fw not found in the mapping lists
2020-04-02T06:42:09.3339720Z WARNING: Firmware file n0bsinitvals22.fw not found in the mapping lists
2020-04-02T06:42:09.3340078Z WARNING: Firmware file n18bsinitvals32.fw not found in the mapping lists
2020-04-02T06:42:09.3340370Z WARNING: Firmware file ucode16_sslpn_nobt.fw not found in the mapping lists
2020-04-02T06:42:09.3340676Z WARNING: Firmware file ucode19_sslpn.fw not found in the mapping lists
2020-04-02T06:42:09.3340973Z WARNING: Firmware file ucode21_sslpn_nobt.fw not found in the mapping lists
2020-04-02T06:42:09.3341269Z WARNING: Firmware file lcn2bsinitvals26.fw not found in the mapping lists
2020-04-02T06:42:09.3341568Z WARNING: Firmware file ucode9.fw not found in the mapping lists
2020-04-02T06:42:09.3341861Z WARNING: Firmware file lcn0bsinitvals26.fw not found in the mapping lists
2020-04-02T06:42:09.3342247Z WARNING: Firmware file a0g1bsinitvals13.fw not found in the mapping lists
2020-04-02T06:42:09.3342618Z WARNING: Firmware file n0bsinitvals17.fw not found in the mapping lists
2020-04-02T06:42:09.3342921Z WARNING: Firmware file lcn1initvals25.fw not found in the mapping lists
2020-04-02T06:42:09.3343219Z WARNING: Firmware file n0absinitvals11.fw not found in the mapping lists
2020-04-02T06:42:09.3343529Z WARNING: Firmware file n0initvals17.fw not found in the mapping lists
2020-04-02T06:42:09.3343820Z WARNING: Firmware file lcn0bsinitvals25.fw not found in the mapping lists
2020-04-02T06:42:09.3344100Z WARNING: Firmware file sslpn3bsinitvals21.fw not found in the mapping lists
2020-04-02T06:42:09.3344400Z WARNING: Firmware file ucode27_sslpn.fw not found in the mapping lists
2020-04-02T06:42:09.3344729Z WARNING: Firmware file ucode22_sslpn.fw not found in the mapping lists
2020-04-02T06:42:09.3345026Z WARNING: Firmware file lcn1bsinitvals25.fw not found in the mapping lists
2020-04-02T06:42:09.3345322Z WARNING: Firmware file lp1initvals20.fw not found in the mapping lists
2020-04-02T06:42:09.3345621Z WARNING: Firmware file lcn2initvals26.fw not found in the mapping lists
2020-04-02T06:42:09.3345901Z WARNING: Firmware file b0g0initvals9.fw not found in the mapping lists
2020-04-02T06:42:09.3346197Z WARNING: Firmware file lp1bsinitvals22.fw not found in the mapping lists
2020-04-02T06:42:09.3346646Z WARNING: Firmware file lcn2bsinitvals25.fw not found in the mapping lists
2020-04-02T06:42:09.3346944Z WARNING: Firmware file sslpn1bsinitvals20.fw not found in the mapping lists
2020-04-02T06:42:09.3350002Z WARNING: Firmware file n1initvals20.fw not found in the mapping lists
2020-04-02T06:42:09.3350392Z WARNING: Firmware file lp2initvals19.fw not found in the mapping lists
2020-04-02T06:42:09.3351045Z WARNING: Firmware file lp0initvals16.fw not found in the mapping lists
2020-04-02T06:42:09.3351350Z WARNING: Firmware file lcn1bsinitvals26.fw not found in the mapping lists
2020-04-02T06:42:09.3351649Z WARNING: Firmware file sslpn0bsinitvals16.fw not found in the mapping lists
2020-04-02T06:42:09.3351959Z WARNING: Firmware file a0g1bsinitvals9.fw not found in the mapping lists
2020-04-02T06:42:09.3352263Z WARNING: Firmware file lp0bsinitvals16.fw not found in the mapping lists
2020-04-02T06:42:09.3352541Z WARNING: Firmware file ucode19_sslpn_nobt.fw not found in the mapping lists
2020-04-02T06:42:09.3352848Z WARNING: Firmware file n18initvals32.fw not found in the mapping lists
2020-04-02T06:42:09.3353142Z WARNING: Firmware file a0g0bsinitvals9.fw not found in the mapping lists
2020-04-02T06:42:09.3353437Z WARNING: Firmware file lcn0initvals25.fw not found in the mapping lists
2020-04-02T06:42:09.3353725Z WARNING: Firmware file lcn1initvals26.fw not found in the mapping lists
2020-04-02T06:42:09.3354013Z WARNING: Firmware file ucode22_mimo.fw not found in the mapping lists
2020-04-02T06:42:09.3355378Z WARNING: Firmware file n2initvals19.fw not found in the mapping lists
2020-04-02T06:42:09.3355730Z WARNING: Firmware file a0g1initvals13.fw not found in the mapping lists
2020-04-02T06:42:09.3355943Z WARNING: Firmware file ucode17_mimo.fw not found in the mapping lists
2020-04-02T06:42:09.3356143Z WARNING: Firmware file lcn2bsinitvals24.fw not found in the mapping lists
2020-04-02T06:42:09.3356341Z WARNING: Firmware file lp2bsinitvals19.fw not found in the mapping lists
2020-04-02T06:42:09.3356538Z WARNING: Firmware file lcn1initvals24.fw not found in the mapping lists
2020-04-02T06:42:09.3356736Z WARNING: Firmware file lp1initvals22.fw not found in the mapping lists
2020-04-02T06:42:09.3356927Z WARNING: Firmware file sslpn0initvals16.fw not found in the mapping lists
2020-04-02T06:42:09.3357123Z WARNING: Firmware file sslpn2initvals19.fw not found in the mapping lists
2020-04-02T06:42:09.3357299Z WARNING: Firmware file sslpn4initvals22.fw not found in the mapping lists
2020-04-02T06:42:09.3357495Z WARNING: Firmware file lcn2initvals25.fw not found in the mapping lists
2020-04-02T06:42:09.3357688Z WARNING: Firmware file lcn1bsinitvals24.fw not found in the mapping lists
[... 60 lines cut ...]
2020-04-02T06:42:11.6847869Z WARNING: Firmware file n2initvals19.fw not found in the mapping lists
2020-04-02T06:42:11.6847982Z WARNING: Firmware file a0g1initvals13.fw not found in the mapping lists
2020-04-02T06:42:11.6848107Z WARNING: Firmware file ucode17_mimo.fw not found in the mapping lists
2020-04-02T06:42:11.6848225Z WARNING: Firmware file lcn2bsinitvals24.fw not found in the mapping lists
2020-04-02T06:42:23.8201753Z WARNING: Firmware file lp2bsinitvals19.fw not found in the mapping lists
2020-04-02T06:42:23.8202545Z WARNING: Firmware file lcn1initvals24.fw not found in the mapping lists
2020-04-02T06:42:23.8202905Z WARNING: Firmware file lp1initvals22.fw not found in the mapping lists
2020-04-02T06:42:23.8203201Z WARNING: Firmware file sslpn0initvals16.fw not found in the mapping lists
2020-04-02T06:42:23.8203490Z WARNING: Firmware file sslpn2initvals19.fw not found in the mapping lists
2020-04-02T06:42:23.8203773Z WARNING: Firmware file sslpn4initvals22.fw not found in the mapping lists
2020-04-02T06:42:23.8205389Z WARNING: Firmware file lcn2initvals25.fw not found in the mapping lists
2020-04-02T06:42:23.8205694Z WARNING: Firmware file lcn1bsinitvals24.fw not found in the mapping lists
2020-04-02T06:42:23.8206070Z WARNING: Firmware file ucode20_sslpn.fw not found in the mapping lists
[... cut ...]
2020-04-02T06:42:23.8207468Z WARNING: Firmware file sslpn3initvals21.fw not found in the mapping lists
2020-04-02T06:42:23.8207881Z WARNING: Firmware file n2bsinitvals19.fw not found in the mapping lists
2020-04-02T06:42:23.8208844Z make[1]: *** [/home/runner/work/openwrt/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2020-04-02T06:42:23.8209460Z Build failed - please re-run with -j1 to see the real error message
2020-04-02T06:42:23.8209675Z make: *** [world] Error 1
```

zhaojh329, 2020-04-03:
Fixed. 7bd51

peterwillcn, 2020-04-03:
> Fixed.
> [536d1ad](https://github.com/openwrt/packages/commit/536d1adc604baede01f4d7dc976f2e907927bd51)
Have you tested it? The bug remains unfixed 😄
```
2020-04-03T06:34:44.1700624Z /home/runner/work/packages/packages/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/main.c:17:10: fatal error: net/netfilter/nf_nat_l3proto.h: No such file or directory
2020-04-03T06:34:44.1701660Z make[5]: *** [/home/runner/work/packages/packages/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/main.o] Error 1
2020-04-03T06:34:44.1702610Z make[4]: *** [/home/runner/work/packages/packages/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl] Error 2
2020-04-03T06:34:44.1703413Z make[3]: *** [/home/runner/work/packages/packages/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/.built] Error 2
2020-04-03T06:34:44.1704235Z make[2]: *** [package/feeds/packages/wifidog-ng/compile] Error 2
2020-04-03T06:34:44.1705075Z make[1]: *** [/home/runner/work/packages/packages/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
```

zhaojh329, 2020-04-03:
Have you updated?

peterwillcn, 2020-04-03:
Yes, see https://github.com/peterwillcn/packages/actions/runs/69593667/workflow

```        
./scripts/feeds update -a
        #rm -rf feeds/packages/net/wifidog*
        ./scripts/feeds install -a
```

neheb, 2020-04-03:
The buildbots are not seeing this failure anymore.

peterwillcn, 2020-04-04:
@zhaojh329 Found the problem, in an old version on the version of (SRC - git packages https://git.openwrt.org/feed/packages.git ^ cf44e13), do not update lead to compile failed.
http://downloads.openwrt.org/snapshots/targets/x86/64/feeds.buildinfo
```
src-git packages https://git.openwrt.org/feed/packages.git^cf44e13
src-git luci https://git.openwrt.org/project/luci.git^89a50c2
src-git routing https://git.openwrt.org/feed/routing.git^512240f
src-git telephony https://git.openwrt.org/feed/telephony.git^06ec029
```

neheb, 2020-04-08:
Closing since this is fixed.

closed 2020-04-08 by neheb