openwrt/packages #11746: dawn will not build!

tapper82, 2020-04-02:
After adding luci-app-dawn to my build using make menuconfig build fails  when building umdns. I am building for mvebu wrt3200acm.

tapper82, 2020-04-02:
@PolynomialDivision

PolynomialDivision, 2020-04-02:
@tapper82 
I can not reproduce it? Can you give me more information? In particular, I would be interested in the error message.

tapper82, 2020-04-02:
make[4]: Entering directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
make[5]: Entering directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
make[6]: Entering directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
make[6]: Leaving directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
make[6]: Entering directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
[ 11%] Building C object CMakeFiles/umdns.dir/dns.c.o
/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/dns.c: In function 'dns_consume_question':
/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/dns.c:261:2: error: converting a packed 'struct dns_question' pointer (alignment 1) to a 'uint16_t' {aka 'short unsigned int'} pointer (alignment 2) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  261 |  uint16_t *swap = (uint16_t *) q;
      |  ^~~~~~~~
In file included from /home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/dns.c:37:
/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/dns.h:67:8: note: defined here
   67 | struct dns_question {
      |        ^~~~~~~~~~~~
cc1: all warnings being treated as errors
CMakeFiles/umdns.dir/build.make:75: recipe for target 'CMakeFiles/umdns.dir/dns.c.o' failed
make[6]: *** [CMakeFiles/umdns.dir/dns.c.o] Error 1
make[6]: Leaving directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/umdns.dir/all' failed
make[5]: *** [CMakeFiles/umdns.dir/all] Error 2
make[5]: Leaving directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
Makefile:47: recipe for target '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/.built' failed
make[3]: *** [/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/.built] Error 2
time: package/network/services/umdns/compile#0.51#0.10#0.76

tapper82, 2020-04-02:
I changed -o2 -pipe to -os -pipe

-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /home/tapper/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-9.3.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc
-- Check for working C compiler: /home/tapper/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-9.3.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_ARG1
    CMAKE_CXX_COMPILER
    CMAKE_CXX_COMPILER_ARG1
    CMAKE_CXX_FLAGS_RELEASE
    CMAKE_FIND_ROOT_PATH_MODE_LIBRARY
    CMAKE_MODULE_LINKER_FLAGS
    CMAKE_SHARED_LINKER_FLAGS
    DL_LIBRARY

-- Build files have been written to: /home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417
make[4]: Entering directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
make[5]: Entering directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
make[6]: Entering directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
Scanning dependencies of target umdns
make[6]: Leaving directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
make[6]: Entering directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
[ 11%] Building C object CMakeFiles/umdns.dir/main.c.o
[ 22%] Building C object CMakeFiles/umdns.dir/dns.c.o
/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/dns.c: In function 'dns_consume_question':
/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/dns.c:261:2: error: converting a packed 'struct dns_question' pointer (alignment 1) to a 'uint16_t' {aka 'short unsigned int'} pointer (alignment 2) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  261 |  uint16_t *swap = (uint16_t *) q;
      |  ^~~~~~~~
In file included from /home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/dns.c:37:
/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/dns.h:67:8: note: defined here
   67 | struct dns_question {
      |        ^~~~~~~~~~~~
cc1: all warnings being treated as errors
CMakeFiles/umdns.dir/build.make:75: recipe for target 'CMakeFiles/umdns.dir/dns.c.o' failed
make[6]: *** [CMakeFiles/umdns.dir/dns.c.o] Error 1
make[6]: Leaving directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/umdns.dir/all' failed
make[5]: *** [CMakeFiles/umdns.dir/all] Error 2
make[5]: Leaving directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417'
Makefile:47: recipe for target '/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/.built' failed
make[3]: *** [/home/tapper/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/umdns-2018-01-02-78974417/.built] Error 2
time: package/network/services/umdns/compile#0.79#0.16#1.78

PolynomialDivision, 2020-04-03:
And without selecting `luci-app-dawn` it compiles? So just selecting `umdns`?

tapper82, 2020-04-03:
I did not try and build just umdns alone. Sorry I should of made that clear. I just selected luci-app-dawn. I will try a build of just adding unmdns now.

tapper82, 2020-04-03:
Yeah It's defo umdns. It fails with same thing.

PolynomialDivision, 2020-04-03:
Okay, so it is no problem of my package.
Please write the [maintainer of umdns](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/network/services/umdns/Makefile;h=ffbc3fed351a021320b0961fd17429b513ecd255;hb=HEAD#l19). Since this is OpenWrt branch and not the Community maintained package feed, you should go to OpenWrt Forum, OpenWrt Mailinglist or OpenWrt IRC.

Could u please cose the issue?

tapper82, 2020-04-03:
Will do thanks for your help.

closed 2020-04-03 by tapper82

PolynomialDivision, 2020-04-04:
@tapper82 
There is a patch on the dev mailing list, that fixes your umdns problem. :)

PolynomialDivision, 2020-04-05:
@tapper82 The patch is merged! :+1: You could try again.

tapper82, 2020-04-05:
Hi is there any dox out there to explain more about using dawn?

PolynomialDivision, 2020-04-06:
> Hi is there any dox out there to explain more about using dawn?

You can use `luci-app-dawn` to configure the values and the behavior of dawn.
You can find the configuration parameters [here](https://github.com/berlin-open-wireless-lab/DAWN/blob/master/README.md#Configuration).