openwrt/packages #11749: midnight commander: mouse clicks cause terminal sequences to be emitted in command line

dl12345, 2020-04-03:
Currently, if clicking with a mouse anywhere on the mc window, terminal sequences are emitted on the command line. This bug is [reported here](https://midnight-commander.org/attachment/ticket/3954/mc-3954-mouse-with-new-terminfo.patch) and a [patch is available here](https://midnight-commander.org/attachment/ticket/3954/mc-3954-mouse-with-new-terminfo.patch)

The following patch is the patch linked above that has been applied and refreshed with quilt so that it applies cleanly. I confirm that this patch fixes the reported problem

```
commit 472acdb6dd0d30e0642b9134a6fb2c1de4c18f35
Author: Egmont Koblinger <egmont@gmail.com>
Date:   Fri Feb 14 22:14:19 2020 +0100

    Ticket #3954: Fix mouse handling with newer terminfo entries

--- a/lib/tty/key.c
+++ b/lib/tty/key.c
@@ -2123,8 +2123,12 @@ tty_get_event (struct Gpm_Event *event,
 #endif /* KEY_MOUSE */
                           || c == MCKEY_EXTENDED_MOUSE))
     {
-        /* Mouse event */
-        xmouse_get_event (event, c == MCKEY_EXTENDED_MOUSE);
+        /* Mouse event. See tickets 2956 and 3954 for extended mode detection. */
+        xmouse_get_event (event, c == MCKEY_EXTENDED_MOUSE
+#ifdef KEY_MOUSE
+                                 || (c == KEY_MOUSE && xmouse_seq == NULL && xmouse_extended_seq != NULL)
+#endif /* KEY_MOUSE */
+                                 );
         c = (event->type != 0) ? EV_MOUSE : EV_NONE;
     }
     else if (c == MCKEY_BRACKETED_PASTING_START)
--- a/lib/tty/mouse.c
+++ b/lib/tty/mouse.c
@@ -90,8 +90,14 @@ init_mouse (void)
 
     case MOUSE_XTERM_NORMAL_TRACKING:
     case MOUSE_XTERM_BUTTON_EVENT_TRACKING:
-        define_sequence (MCKEY_MOUSE, xmouse_seq, MCKEY_NOACTION);
-        define_sequence (MCKEY_EXTENDED_MOUSE, xmouse_extended_seq, MCKEY_NOACTION);
+        if (xmouse_seq != NULL)
+        {
+            define_sequence (MCKEY_MOUSE, xmouse_seq, MCKEY_NOACTION);
+        }
+        if (xmouse_extended_seq != NULL)
+        {
+            define_sequence (MCKEY_EXTENDED_MOUSE, xmouse_extended_seq, MCKEY_NOACTION);
+        }
         break;
 
     default:
--- a/lib/tty/tty.c
+++ b/lib/tty/tty.c
@@ -310,9 +310,17 @@ tty_init_xterm_support (gboolean is_xter
         }
     }
 
-    /* No termcap for SGR extended mouse (yet), hardcode it for now */
+    /* There's only one termcap entry "kmous", typically containing "\E[M" or "\E[<".
+     * We need the former in xmouse_seq, the latter in xmouse_extended_seq.
+     * See tickets 2956 and 3954 for details. */
     if (xmouse_seq != NULL)
+    {
+        if (strcmp (xmouse_seq, ESC_STR "[<") == 0)
+        {
+            xmouse_seq = NULL;
+        }
         xmouse_extended_seq = ESC_STR "[<";
+    }
 }
 
 /* --------------------------------------------------------------------------------------------- */
```

dibdot, 2020-04-03:
Are you willing to open a PR for that?

dibdot, 2020-04-03:
The comments from the author regarding the mentioned ticket doesn't sound like a proper fix (http://midnight-commander.org/ticket/3954), "It's a quick ugly hotfix, no cleanup refactoring." I think we should wait for the next mc release ... at least until the ticket has has been closed with a proper fix.

dl12345, 2020-04-03:
Fair enough. I see your point. 

This fix does not appear to be scheduled for the [upcoming 4.8.25 release](http://midnight-commander.org/wiki/NEWS-4.8.25).

There could potentially be some libc compatibility issues in an upcoming release, since I note that the above link states a minimal glib compatibility of 2.30, so what effect this will have on musl remains to be seen.

I've applied this patch to my copy of mc in the meantime, since (for me at least) this bug render mc much less usable on OpenWrt. 

At least with this ticket we have the problem identified and the cause documented in case it's reported again.

vcunat, 2020-06-26:
Fix merged into master upstream: https://github.com/MidnightCommander/mc/commit/d9da121b

dibdot, 2020-07-02:
should be fixed now in master and 19.07. Thanks @BKPepe

PR #12694 (2020-07-02): [19.07] mc: fix mouse handling
Maintainer: @dibdot 
Compile tested: Turris 1.1, powerpc_8540, OpenWrt 19.07.3
Run tested: Konsole - KDE Terminal emulator version 20.04.2 and mouse clicking in mc

Description:
Fixes #11749 by backporting patch https://github.com/MidnightCommander/mc/commit/b92c7f86b4be3b200e0f2de713a4c40b28599e61. I tried to compile mc with that patch and together with this one https://github.com/MidnightCommander/mc/commit/8877d218ddf7347cd813f60fa465545f11e8af77 but the issue was present to me there as well. When I used just the first one, it works. So, this is why I don't wa [... cut]
closed 2020-07-02 by dibdot