openwrt/packages #11750: unbound-control 1.10.0-2: sending SIGSEGV to unbound-control for invalid read access from 00000000

vollkommenheit, 2020-04-03:
Package: unbound-control
Version: 1.10.0-2
Depends: libc, libopenssl1.1, unbound-daemon
Status: install user installed
Section: net
Architecture: mips_24kc
Size: 45121
Filename: unbound-control_1.10.0-2_mips_24kc.ipk
Description: This package contains the Unbound control utility.
Installed-Time: 1585766447

Maintainer: Not sure if unbound-control is maintained by @EricLuehrsen
Environment: 
ath79-generic-netgear_wndr3800
r12776-437eb41f23

Description:

Initially posted at forum.openwrt.org
https://forum.openwrt.org/t/unbound-control-1-10-0-2-sending-sigsegv-to-unbound-control-for-invalid-read-access-from-00000000/59201

Wed Apr 1 12:01:38 2020 kern.info kernel: [141571.460081] do_page_fault(): sending SIGSEGV to unbound-control for invalid read access from 00000000
Wed Apr 1 12:01:38 2020 kern.info kernel: [141571.474976] ra = 77df0d2d in libunbound.so.8.1.7[77dbc000+b3000]

Now, I'm getting extra info in syslog, when unbound-control gets triggered by a shell script,  pointing to libc

Thu Apr  2 22:01:55 2020 kern.info kernel: [253187.975857] do_page_fault(): sending SIGSEGV to unbound-control for invalid read access from 00000000
Thu Apr  2 22:01:55 2020 kern.info kernel: [253187.985262] epc = 77eca230 in libc.so[77e56000+97000]
Thu Apr  2 22:01:55 2020 kern.info kernel: [253187.990544] ra  = 77dd2d2d in libunbound.so.8.1.7[77d9e000+b3000]
Thu Apr  2 22:01:55 2020 local0.warn dnsmasq[6451]: script process exited with status 139

EricLuehrsen, 2020-04-12:
The answer is in the error log. Unbound control apparently exits harshly when configured off.
`[1585965988] unbound-control[32619:0] warning: control-enable is 'no' in the config file. Segmentation fault`

vollkommenheit, 2020-04-12:
I have ruled it out though, as I mentioned in the initial forum.openwrt.org post, that I can use Linux Linux unbound-control 1.10.1 -> Openwrt unbound 1.10.0.2 successfully, so there's really nothing wrong with my access control.

EricLuehrsen, 2020-04-13:
I will guess its something "else" in the variant packages because I cannot replicate it but it appeared when ipset was added. This happened before when flashing OpenWrt base and installing packages from the main server would break, only some times for some users. This would not occur for locally built test packages FTP up to the router before release. Not an easy fix ...

vollkommenheit, 2020-04-13:
Ah, thanks.

EricLuehrsen, 2020-06-04:
Something just isn't working when `opkg` installs variants, but only sometimes. I think I will remove variants and have "light weight" be a custom build option instead. It might put low ROM devices out of the market for Unbound when using pre-built packages, but this is too unreliable.

vollkommenheit, 2020-06-11:
This gave me the hint to install the "heavy" variant of unbound (didn't realize there's an unbound-daemon-heavy). Now unbound-control works as expected. And, as I suspected after reading your reply in an Openwrt forum thread, the "heavy" unbound-daemon also seems to have cured my problem of "light" unbound-daemon intermittently completely stopping performing lookups after enabling DoT in unbound.

PR #12581 (2020-06-27): unbound: reduced to one package with selectable options
Maintainer: me
Compile tested: Linksys WRT3200ACM all options
Run tested: Linksys WRT3200ACM with default or lighter options
Description: Enough is enough. Dual build Unbound and helper packages is a mess. One Unbound package will be presented with fair defaults. Up select or down select options under libraries/network/libunbound to enhance Unbound or squeeze Unbound into your desired target. Also add a new option to pop in helper DNS programs on the local host. I don't have use cases for the extended options. So I compiled them, but I do [... cut]
closed 2020-06-11 by vollkommenheit