openwrt/packages #11751: mwan3: RTNETLINK answers: File exists

brianjmurrell, 2020-04-03:
Maintainer: @feckert 
Environment: ar71xx 19.07.2

Description:

When I run `mwan3 restart` I get a spew of `RTNETLINK answers: File exists`.

I added some xtrace to `/etc/hotplug.d/iface/15-mwan3` and discovered it happens during:

```
+ ip -6 route add table 5 fd31:aeb1:48df::/64 dev br-lan proto static metric 1024 pref medium
+ read line
+ ip -6 route add table 5 fd8c:5604:c29c:c562::/64 dev tun0 proto kernel metric 256 pref medium
+ read line
+ ip -6 route add table 5 fe80::/10 dev pppoe-wan1 metric 1 pref medium
+ read line
+ ip -6 route add table 5 fe80::/10 dev pppoe-wan1 proto kernel metric 256 pref medium
+ read line
+ ip -6 route add table 5 fe80::/64 dev 6in4-henet proto kernel metric 256 pref medium
+ read line
+ ip -6 route add table 5 fe80::/64 dev br-guest proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev br-lan proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev eth0 proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev eth0.2 proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev eth0.3 proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev ifb4eth0.2 proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev tun0 proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev wlan0 proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev wlan0-1 proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev wlan1 proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 fe80::/64 dev wlan1-1 proto kernel metric 256 pref medium
RTNETLINK answers: File exists
+ read line
+ ip -6 route add table 5 2001:470:1c:6f8::/64 dev 6in4-henet proto kernel metric 256 pref medium
+ read line
+ ip -6 route add table 5 2001:470:1d:6f8::/64 dev br-lan proto static metric 1024 pref medium
+ read line
+ ip -6 route add table 5 2607:f2c0:a000:13d::/64 dev pppoe-wan1 proto static metric 256 pref medium
+ read line
+ ip -6 route add table 5 2607:f2c0:f00f:cd00::/64 dev br-lan proto static metric 1024 pref medium
+ read line
```

I have two IPv6 interfaces, so perhaps there is some idempotency issues here when configuring more than on interface?

feckert, 2020-05-29:
To better understand what the problem is, I need more information about what you are actually doing. How is the configuration of mwan3 or the network?

jamesmacwhite, 2020-07-13:
@brianjmurrell @feckert I think @aaronjg PR fixed this issue with his IPv6 changes.

feckert, 2020-07-14:
I will close this issue. If the error occurs again please reopen.

closed 2020-07-14 by feckert