openwrt/packages #11764: Vote about Facebook packages removal

BKPepe, 2020-04-04:
There are several packages, which were added to this repository last year, but they are not maintained since June 2019. I don't mind if anyone from Facebook is willing to maintain it. But currently, I'm in favor to remove it. @neheb was updating them in the past, but so far maintainer @ammubhave seems to be active on Github, but those packages are getting outdated here. I doubt that anyone else is using them.

I am talking primary about these libraries, which are being develop by Facebook:
- libfolly
Version in master branch: 2019.10.07.00
Upstream version: v2020.03.30.00
- fbzmq
Version in master branch: 2019.10.07.00
Upstream version: 2020.03.30.00
- fbthrift
Version in master branch: 2019.10.07.00
Upstream version: 2020.03.30.00
- libwangle
Version in master branch: 2019.10.07.00
Upstream version: 2020.03.30.00
- libfizz
Version in master branch: 2019.10.07.00
Upstream version: 2020.03.30.00

It is not just about them, but in this repository, there are a lot packages, which are not maintained, but they can be use by someone else. And the full build should take less time.

diizzyy, 2020-04-04:
Go ahead, I've already mentioned my concerns about merging PRs left and right.

neheb, 2020-04-04:
The initial goal of those PRs was to get Facebook's OpenR routing daemon in the tree. That never got done.

There are actually a couple of more libraries. I'll spin up a PR when I can.

BKPepe, 2020-04-04:
> The initial goal of those PRs was to get Facebook's OpenR routing daemon in the tree. That never got done.

For completeness, I am posting that PR about OpenR: https://github.com/openwrt/packages/pull/8951

neheb, 2020-04-04:
https://github.com/openwrt/packages/pull/11766

closed 2020-04-26 by BKPepe