openwrt/packages #11796: libssh2 compilation error

qiuzi, 2020-04-07:
`make[3]: Entering directory '/home/sda3/openwrt/feeds/packages/libs/libssh2'
touch /home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0/.prepared_358af877eff9697be560a077e886d176_6664517399ebbbc92a37c5bb081b5c53_check
. /home/sda3/openwrt/include/shell.sh; gzip -dc /home/sda3/openwrt/dl/libssh2-1.9.0.tar.gz | tar -C /home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0
touch /home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0/.prepared_358af877eff9697be560a077e886d176_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0/.configured_*
rm -f /home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.libssh2_installed
mkdir -p /home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0
(cd /home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0; CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0=libssh2-1.9.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0=libssh2-1.9.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/lib -L/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=mipsel -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER="/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc" -DCMAKE_C_COMPILER_ARG1="" -DCMAKE_CXX_COMPILER="/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-g++" -DCMAKE_CXX_COMPILER_ARG1="" -DCMAKE_ASM_COMPILER="/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc" -DCMAKE_ASM_COMPILER_ARG1="" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/lib -L/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/lib -L/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/lib -L/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/usr;/home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl -DCMAKE_PREFIX_PATH=/home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl -DCMAKE_SKIP_RPATH=TRUE -DBUILD_SHARED_LIBS=ON -DBUILD_TESTING=OFF -DENABLE_ZLIB_COMPRESSION=ON -DCLEAR_MEMORY=ON -DCRYPTO_BACKEND=mbedTLS /home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0 )
-- The C compiler identification is GNU 8.4.0
-- Check for working C compiler: /home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc
-- Check for working C compiler: /home/sda3/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ZLIB: /home/sda3/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libz.so (found version "1.2.11") 
-- Looking for include file unistd.h
-- Looking for include file unistd.h - found
-- Looking for include file inttypes.h
-- Looking for include file inttypes.h - found
-- Looking for include file stdlib.h
-- Looking for include file stdlib.h - found
-- Looking for include file sys/select.h
-- Looking for include file sys/select.h - found
-- Looking for include file sys/uio.h
-- Looking for include file sys/uio.h - found
-- Looking for include file sys/socket.h
-- Looking for include file sys/socket.h - found
-- Looking for include file sys/ioctl.h
-- Looking for include file sys/ioctl.h - found
-- Looking for include file sys/time.h
-- Looking for include file sys/time.h - found
-- Looking for include file sys/un.h
-- Looking for include file sys/un.h - found
-- Looking for include file windows.h
-- Looking for include file windows.h - not found
-- Looking for include file ws2tcpip.h
-- Looking for include file ws2tcpip.h - not found
-- Looking for include file winsock2.h
-- Looking for include file winsock2.h - not found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of long long
-- Check size of long long - done
-- Looking for gettimeofday
-- Looking for gettimeofday - found
-- Looking for strtoll
-- Looking for strtoll - found
-- Looking for snprintf
-- Looking for snprintf - found
-- Looking for memset_s
-- Looking for memset_s - not found
-- Looking for poll
-- Looking for poll - found
-- Looking for socket
-- Looking for socket - found
-- Looking for select
-- Looking for select - found
-- Looking for inet_addr
-- Looking for inet_addr - found
-- Performing Test HAVE_O_NONBLOCK
-- Performing Test HAVE_O_NONBLOCK - Success
-- Looking for include file arpa/inet.h
-- Looking for include file arpa/inet.h - found
-- Looking for include file netinet/in.h
[... 102 lines cut ...]
/home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0/src/libssh2_priv.h:286:5: error: unknown type name '_libssh2_ec_key'
     _libssh2_ec_key *private_key;   /* SSH2 ecdh private key */
     ^~~~~~~~~~~~~~~
/home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0/src/libssh2_priv.h:626:39: error: 'SHA_DIGEST_LENGTH' undeclared here (not in a function); did you mean 'MAX_SHA_DIGEST_LEN'?
     unsigned char server_hostkey_sha1[SHA_DIGEST_LENGTH];
                                       ^~~~~~~~~~~~~~~~~
                                       MAX_SHA_DIGEST_LEN
/home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0/src/libssh2_priv.h:629:41: error: 'SHA256_DIGEST_LENGTH' undeclared here (not in a function); did you mean 'MAX_SHA_DIGEST_LEN'?
     unsigned char server_hostkey_sha256[SHA256_DIGEST_LENGTH];
                                         ^~~~~~~~~~~~~~~~~~~~
                                         MAX_SHA_DIGEST_LEN
/home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0/src/libssh2_priv.h:929:7: error: expected specifier-qualifier-list before '_libssh2_cipher_type'
       _libssh2_cipher_type(algo);
[... cut ...]
         path = getenv("SSH_AUTH_SOCK");
                ^~~~~~
                getopt
/home/sda3/openwrt/build_dir/target-mipsel_24kc_musl/libssh2-1.9.0/src/agent.c:154:14: warning: assignment to 'const char *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
         path = getenv("SSH_AUTH_SOCK");
              ^
src/CMakeFiles/libssh2.dir/build.make:79: recipe for target 'src/CMakeFiles/libssh2.dir/agent.c.o' failed
make[6]: *** [src/CMakeFiles/libssh2.dir/agent.c.o] Error 1
`

neheb, 2020-04-08:
CMake throws this error:

```
-- The following REQUIRED packages have not been found:

    mbedTLS
```

mbedTLS is not build on your end for some reason.

qiuzi, 2020-04-08:
@neheb Maybe libcurl is not associated with related components by default?

qiuzi, 2020-04-08:
make menuconfig manually selects libssh2 and successfully compiles
It is recommended to add a default association

neheb, 2020-04-09:
mbedtls is already default. See: https://github.com/openwrt/packages/blob/master/libs/libssh2/Config.in#L5

I'm guessing this is selected by libcurl. libssh2 is not built by default.

neheb, 2020-04-10:
ping @jefferyto

jefferyto, 2020-04-13:
So you were trying to build libcurl and got this error?

qiuzi, 2020-04-13:
@jefferyto yes, there will be an error associated with curl

jefferyto, 2020-04-13:
Can you post your `.config` file?

qiuzi, 2020-04-16:
@jefferyto 
`#
# File Transfer
#
# CONFIG_PACKAGE_aria2 is not set
# CONFIG_PACKAGE_atftp is not set
# CONFIG_PACKAGE_atftpd is not set
CONFIG_PACKAGE_curl=y
.........
CONFIG_PACKAGE_libcurl=y

#
# SSL support
#
CONFIG_LIBCURL_MBEDTLS=y
# CONFIG_LIBCURL_WOLFSSL is not set
# CONFIG_LIBCURL_OPENSSL is not set
# CONFIG_LIBCURL_GNUTLS is not set
# CONFIG_LIBCURL_NOSSL is not set

#
# Supported protocols
#
# CONFIG_LIBCURL_DICT is not set
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
# CONFIG_LIBCURL_GOPHER is not set
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_COOKIES=y
# CONFIG_LIBCURL_IMAP is not set
# CONFIG_LIBCURL_LDAP is not set
# CONFIG_LIBCURL_POP3 is not set
# CONFIG_LIBCURL_RTSP is not set
# CONFIG_LIBCURL_SSH2 is not set
CONFIG_LIBCURL_NO_SMB="!"
# CONFIG_LIBCURL_SMTP is not set
# CONFIG_LIBCURL_TELNET is not set
# CONFIG_LIBCURL_TFTP is not set
# CONFIG_LIBCURL_NGHTTP2 is not set

#
# Miscellaneous
#
CONFIG_LIBCURL_PROXY=y
# CONFIG_LIBCURL_CRYPTO_AUTH is not set
# CONFIG_LIBCURL_TLS_SRP is not set
# CONFIG_LIBCURL_LIBIDN2 is not set
# CONFIG_LIBCURL_THREADED_RESOLVER is not set
# CONFIG_LIBCURL_ZLIB is not set
# CONFIG_LIBCURL_UNIX_SOCKETS is not set
# CONFIG_LIBCURL_LIBCURL_OPTION is not set
# CONFIG_LIBCURL_VERBOSE is not set
# CONFIG_PACKAGE_libcxx is not set
# CONFIG_PACKAGE_libdaemon is not set
# CONFIG_PACKAGE_libdaq is not set
# CONFIG_PACKAGE_libdb47 is not set
# CONFIG_PACKAGE_libdb47xx is not set
# CONFIG_PACKAGE_libdbi is not set
# CONFIG_PACKAGE_libdbus is not set
# CONFIG_PACKAGE_libdevmapper is not set
# CONFIG_PACKAGE_libdmapsharing is not set
# CONFIG_PACKAGE_libdnet is not set
# CONFIG_PACKAGE_libdouble-conversion is not set
# CONFIG_PACKAGE_libdrm is not set
# CONFIG_PACKAGE_libdw is not set
# CONFIG_PACKAGE_libecdsautil is not set
# CONFIG_PACKAGE_libedit is not set
# CONFIG_PACKAGE_libelf is not set
# CONFIG_PACKAGE_libesmtp is not set
# CONFIG_PACKAGE_libestr is not set
[... 213 lines cut ...]
# CONFIG_PACKAGE_libslang2-modules is not set
# CONFIG_PACKAGE_libsmartcols is not set
# CONFIG_PACKAGE_libsndfile is not set
# CONFIG_PACKAGE_libsoc is not set
# CONFIG_PACKAGE_libsocks is not set
CONFIG_PACKAGE_libsodium=y

#
# Configuration
#
CONFIG_LIBSODIUM_MINIMAL=y
# CONFIG_PACKAGE_libsoup is not set
# CONFIG_PACKAGE_libsoxr is not set
# CONFIG_PACKAGE_libspeex is not set
# CONFIG_PACKAGE_libspeexdsp is not set
# CONFIG_PACKAGE_libspice-server is not set
# CONFIG_PACKAGE_libss is not set
# CONFIG_PACKAGE_libssh is not set
# CONFIG_PACKAGE_libssh2 is not set
# CONFIG_PACKAGE_libstoken is not set
# CONFIG_PACKAGE_libstrophe is not set
# CONFIG_PACKAGE_libsysrepo is not set
# CONFIG_PACKAGE_libtalloc is not set
# CONFIG_PACKAGE_libtasn1 is not set
# CONFIG_PACKAGE_libtheora is not set`

jefferyto, 2020-04-16:
Can you attach the full file? Also can you try doing a clean checkout and build again?

qiuzi, 2020-04-16:
It can be compiled after pull. Thank you very much.

qiuzi, 2020-04-28:
@jefferyto 
`CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_phicomm_k2p=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBSODIUM_MINIMAL=y
# CONFIG_LUCI_CSSTIDY is not set
# CONFIG_LUCI_JSMIN is not set
CONFIG_LUCI_LANG_en=y
CONFIG_LUCI_LANG_zh_Hans=y
CONFIG_NGINX_STREAM_CORE_MODULE=y
CONFIG_PACKAGE_boost=y
CONFIG_PACKAGE_boost-date_time=y
CONFIG_PACKAGE_boost-program_options=y
CONFIG_PACKAGE_boost-system=y
CONFIG_PACKAGE_ca-bundle=y
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_dnscrypt-proxy=y
CONFIG_PACKAGE_dnscrypt-proxy-resolvers=y
# CONFIG_PACKAGE_dnsmasq is not set
CONFIG_PACKAGE_dnsmasq-full=y
CONFIG_PACKAGE_dnsmasq_full_auth=y
CONFIG_PACKAGE_dnsmasq_full_conntrack=y
CONFIG_PACKAGE_dnsmasq_full_dhcp=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_dnsmasq_full_dnssec=y
CONFIG_PACKAGE_dnsmasq_full_ipset=y
CONFIG_PACKAGE_dnsmasq_full_noid=y
CONFIG_PACKAGE_dnsmasq_full_tftp=y
CONFIG_PACKAGE_ebtables-utils=y
CONFIG_PACKAGE_hostapd=y
CONFIG_PACKAGE_iftop=y
CONFIG_PACKAGE_ipset=y
CONFIG_PACKAGE_iptables-mod-geoip=y
CONFIG_PACKAGE_iptables-mod-tproxy=y
CONFIG_PACKAGE_kmod-ebtables-ipv4=y
CONFIG_PACKAGE_kmod-ebtables-ipv6=y
CONFIG_PACKAGE_kmod-ipt-conntrack-extra=y
CONFIG_PACKAGE_kmod-ipt-extra=y
CONFIG_PACKAGE_kmod-ipt-geoip=y
CONFIG_PACKAGE_kmod-ipt-ipset=y
CONFIG_PACKAGE_kmod-ipt-tproxy=y
CONFIG_PACKAGE_kmod-ledtrig-default-on=y
CONFIG_PACKAGE_kmod-ledtrig-heartbeat=y
CONFIG_PACKAGE_kmod-ledtrig-netdev=y
CONFIG_PACKAGE_kmod-ledtrig-timer=y
CONFIG_PACKAGE_kmod-nf-conntrack-netlink=y
CONFIG_PACKAGE_kmod-nfnetlink=y
CONFIG_PACKAGE_kmod-nft-bridge=y
CONFIG_PACKAGE_kmod-nft-core=y
CONFIG_PACKAGE_kmod-nft-netdev=y
CONFIG_PACKAGE_kmod-sched-connmark=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-tcp-bbr=y
CONFIG_PACKAGE_libcares=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libev=y
CONFIG_PACKAGE_libevent2=y
CONFIG_PACKAGE_libgmp=y
CONFIG_PACKAGE_libipset=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_libmbedtls=y
CONFIG_PACKAGE_libmnl=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnetfilter-conntrack=y
CONFIG_PACKAGE_libnettle=y
CONFIG_PACKAGE_libnfnetlink=y
CONFIG_PACKAGE_libnftnl=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpcre=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libsodium=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-acl=y
CONFIG_PACKAGE_luci-app-adrules=y
# CONFIG_PACKAGE_luci-app-diskman_INCLUDE_btrfs_progs is not set
# CONFIG_PACKAGE_luci-app-diskman_INCLUDE_lsblk is not set
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-nft-qos=y
CONFIG_PACKAGE_luci-app-rp-pppoe-server=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-i18n-acl-en=y
CONFIG_PACKAGE_luci-i18n-adrules-zh-cn=y
CONFIG_PACKAGE_luci-i18n-base-en=y
CONFIG_PACKAGE_luci-i18n-base-zh-cn=y
CONFIG_PACKAGE_luci-i18n-firewall-en=y
CONFIG_PACKAGE_luci-i18n-firewall-zh-cn=y
CONFIG_PACKAGE_luci-i18n-nft-qos-en=y
CONFIG_PACKAGE_luci-i18n-nft-qos-zh-cn=y
CONFIG_PACKAGE_luci-i18n-rp-pppoe-server-en=y
CONFIG_PACKAGE_luci-i18n-rp-pppoe-server-zh-cn=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-material=y
CONFIG_PACKAGE_netcat=y
CONFIG_PACKAGE_nft-qos=y
CONFIG_PACKAGE_nftables-nojson=y
CONFIG_PACKAGE_nginx=y
CONFIG_PACKAGE_nping=y
CONFIG_PACKAGE_privoxy=y
CONFIG_PACKAGE_rp-pppoe-common=y
CONFIG_PACKAGE_rp-pppoe-server=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wpad-basic=m
CONFIG_PACKAGE_zlib=y
CONFIG_PRIVOXY_disable-acl-support=y
CONFIG_PRIVOXY_disable-editor=y
CONFIG_PRIVOXY_disable-force=y
CONFIG_PRIVOXY_disable-image-blocking=y
CONFIG_PRIVOXY_disable-toggle=y
CONFIG_PRIVOXY_disable-trust-files=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
`

jefferyto, 2020-04-28:
Please
* Do a clean checkout / build to reproduce the error
* Post the **full** error message
* Attach the **full** `.config` file
* Ping the actual maintainer of libssh2 (not me)

closed 2020-04-29 by qiuzi