openwrt/packages #11808: [19.07] botan can not be compiled with ccache

BKPepe, 2020-04-09:
Maintainer: @hbl0307106015
Environment: powerpc8540, mvebu (cortexa53, cortexa9), OpenWrt 19.07.
Description:

1. Download SDK (for example: `openwrt-sdk-19.07.2-mpc85xx-p2020_gcc-7.5.0_musl.Linux-x86_64.tar.xz`)
2. Extract it
3. Update and install feeds
4. In `make menuconfig`, in `Advanced configuration options (for developers)` select `Use ccache`
5. Compile botan

It fails with this output:
```
   INFO: Configuring to build Botan 2.10.0 (revision git:3a459487df5de4cb411efdda0010e617d6903284)
   INFO: ./configure.py invoked with options "--cpu=powerpc --cc-bin=/openwrt-sdk-19.07.2-mpc85xx-p2020_gcc-7.5.0_musl.Linux-x86_64/staging_dir/toolchain-powerpc_8540_gcc-7.5.0_musl/bin/ccache_cxx --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --optimize-for-size --disable-neon"
   INFO: Autodetected platform information: OS="Linux" machine="x86_64" proc="x86_64"
   INFO: Guessing target OS is linux (use --os to set)
  ERROR: Could not figure out what compiler type '/openwrt-sdk-19.07.2-mpc85xx-p2020_gcc-7.5.0_musl.Linux-x86_64/staging_dir/toolchain-powerpc_8540_gcc-7.5.0_musl/bin/ccache_cxx' is, use --cc to set
make[2]: *** [Makefile:96: /openwrt-sdk-19.07.2-mpc85xx-p2020_gcc-7.5.0_musl.Linux-x86_64/build_dir/target-powerpc_8540_musl/Botan-2.10.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
```

neheb, 2020-04-09:
I removed this in master. The goal was to use it for kea. But OpenSSL ended up being used instead.

It was also quite large last I remember...

hbl0307106015, 2020-05-06:
Maybe you can backport from master, use openssl instead of botan.

neheb, 2020-05-28:
Erm kea uses OpenSSL with both. The botan package is still present though.