openwrt/packages #11814: wget: Installing package breaks opkg due to default IPv6 preference

iskunk, 2020-04-10:
Maintainer: Peter Wagner (cannot find GitHub username)
Environment: OpenWrt 19.07.2

Description: Installing the `wget` package causes `opkg` to hang on network operations (or at least take an unreasonably long time) because it preferentially connects to `downloads.openwrt.org` via IPv6, which my ISP does not support. This is a different behavior from uclient-fetch.

I addressed this issue by placing an `/etc/wgetrc` file ([reference](https://www.gnu.org/software/wget/manual/html_node/Wgetrc-Commands.html)) with the setting `inet4_only = on`, although `prefer_family = IPv4` is probably more suitable in the general case.

ysc3839, 2020-04-10:
I think it's a wget issue. HTTP clients should be able to fallback to IPv4 if IPv6 is not available.

BRabbit777, 2020-04-29:
@iskunk I had the same behavior without the wget package installed. Just the uclient-fetch wget that opkg depends on.

I'm going to try your work around.

iskunk, 2020-04-30:
I don't think uclient-fetch uses `/etc/wgetrc`... did that do anything?

I have not been able to reproduce this issue on a different device. Could you check if the hang-behavior persists? It could potentially be a problem with an upstream resolver that clears up after some time.

BRabbit777, 2020-04-30:
the /etc/wgetrc file didn't make a difference. and it is still an issue this morning for me.

Note that I can very easily toggle the issue by turning off ipv6 on my loopback interface. adding `option ipv6 '0'` to the loopback interface in `/etc/config/network` will reliably cause `opkg update` to break.

In addition I can also run the command `wget http://downloads.openwrt.org/releases/19.07.2/packages/arm_cortext-a9_vfpv3/telephony/Packages.gz` and get the error message, however if I add the `-4`  switch to the command to force ipv4 then it runs with no issue. Thanks!

keithbentrup, 2021-07-18:
I landed on this issue from google while experiencing the same issue (opkg will not fallback to ipv4 via wget when ipv6 is unavailable). The only way that I could resolve the issue was by wrapping wget with a shell script:
1. `mv /usr/bin/wget /usr/bin/wget.bak`
2. create a new `/usr/bin/wget` with contents:
```sh
#!/bin/sh
/usr/bin/wget.bak -4 "$@"
```
3. `chmod +x /usr/bin/wget`

Then I could run all the normal `opkg` commands.  

I hope this hack helps someone else until it's resolved.

satmandu, 2022-01-25:
Here is a little script that can be used if you're expecting ipv6 to suddenly break (say if you're setting up tunnelbroker.)

```
if ! grep "#!/bin/sh" /usr/bin/wget; then
  mv /usr/bin/wget /usr/bin/wget.bak
  echo -e "#!/bin/sh\n/usr/bin/wget.bak -4 \"\$@\"" > /usr/bin/wget && chmod +x /usr/bin/wget
fi
```