openwrt/packages #11826: mwan3: Does not respect "compile with(out) IPV6 support"

anonymous-one, 2020-04-11:
Looks like mwan3 doesn't respect if OpenWRT has been compiled with the enabled / disable IPV6 build setting.

I have a custom OpenWRT build with IPV6 disabled (via Build Options in make menuconfig), and this is what I get when getting mwan3 status:

```
/usr/sbin/mwan3: line 1214: ip6tables: not found
```

A simple check if iptables6 exists would suffice.

feckert, 2020-04-21:
That's right. It wasn't supposed to happen like this until now.
If you have the time and inclination, you can submit a pullrequest that will fix the problem. I have it on my list but due to time constraints I will not be able to implement it any time soon.

feckert, 2020-07-30:
This was solved by @aaronjg (thanks for the work).
Please try the latest version 2.8.12

- [master](https://github.com/openwrt/packages/commit/a0d66d4eebefe6e89c582df2b4dc972ea7c5c7f5#diff-34590d11545c84b961d1992ce3aefdeb) 
- [openwrt-19.07](https://github.com/openwrt/packages/commit/dc25b38f1317ad5a849a6ef12b46f4edfe53a7e4#diff-34590d11545c84b961d1992ce3aefdeb
)

closed 2020-07-30 by feckert