openwrt/packages #11828: irqbalance compilation errors with glibc

dl12345, 2020-04-11:
This is probably glibc specific, since it has not been reported previously. I cannot get irqbalance to compile successfully with glibc without the following additions to the Makefile

```
diff --git a/utils/irqbalance/Makefile b/utils/irqbalance/Makefile
index 88172af..4f20dff 100644
--- a/utils/irqbalance/Makefile
+++ b/utils/irqbalance/Makefile
@@ -32,6 +32,7 @@ define Package/irqbalance
   PKG_BUILD_DEPENDS:=glib2
   SECTION:=utils
   CATEGORY:=Utilities
+  DEPENDS:=+libintl-full +libpthread
   TITLE:=IRQ usage balancing for multi-core systems
   URL:=https://github.com/Irqbalance/irqbalance
 endef
@@ -42,6 +43,8 @@ define Package/irqbalance/description
  increase performance.
 endef
 
+TARGET_LDFLAGS+=-lpthread -lintl
+
 CONFIGURE_ARGS+= \
        --disable-numa \
        --with-libcap_ng=no \
```

neheb, 2020-04-12:
ping @hnyman

hnyman, 2020-04-27:
closed in preference of #11969

Including always  libintl-full would be wrong.

closed 2020-04-27 by hnyman