openwrt/packages #11840: tini not compiling with glibc

neheb, 2020-04-14:
```
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/gcc/mipsel-openwrt-linux-gnu/9.3.0/../../../../mipsel-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/libc.a(iofclose.o): in function `_IO_new_fclose':
iofclose.c:(.text+0x2e4): undefined reference to `_Unwind_Resume'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/gcc/mipsel-openwrt-linux-gnu/9.3.0/../../../../mipsel-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/libc.a(iofclose.o):(.data.DW.ref.__gcc_personality_v0[DW.ref.__gcc_personality_v0]+0x0): undefined reference to `__gcc_personality_v0'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/gcc/mipsel-openwrt-linux-gnu/9.3.0/../../../../mipsel-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/libc.a(iofflush.o): in function `_IO_fflush':
iofflush.c:(.text+0x1b8): undefined reference to `_Unwind_Resume'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/gcc/mipsel-openwrt-linux-gnu/9.3.0/../../../../mipsel-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/libc.a(iofwrite.o): in function `_IO_fwrite':
iofwrite.c:(.text+0x2c8): undefined reference to `_Unwind_Resume'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/gcc/mipsel-openwrt-linux-gnu/9.3.0/../../../../mipsel-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/libc.a(wfileops.o): in function `_IO_wfile_underflow':
wfileops.c:(.text+0x950): undefined reference to `_Unwind_Resume'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/gcc/mipsel-openwrt-linux-gnu/9.3.0/../../../../mipsel-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/libc.a(fputc.o): in function `fputc':
fputc.c:(.text+0x20c): undefined reference to `_Unwind_Resume'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/gcc/mipsel-openwrt-linux-gnu/9.3.0/../../../../mipsel-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/libc.a(fileops.o): in function `_IO_new_file_underflow':
fileops.c:(.text+0x720): undefined reference to `_Unwind_Resume'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/gcc/mipsel-openwrt-linux-gnu/9.3.0/../../../../mipsel-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-9.3.0_glibc/lib/libc.a(iofputs.o):iofputs.c:(.text+0x1f8): more undefined references to `_Unwind_Resume' follow
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Maintainer @G-M0N3Y-2503

G-M0N3Y-2503, 2020-04-17:
I also reproduced this just now on x86_64.
What's the reason for supporting glibc as well as musl? 
it seems to me it's an edge case option somewhat hidden away.

neheb, 2020-04-17:
I mostly agree. For some reason, it's git versions that are tracked here.

G-M0N3Y-2503, 2020-04-17:
Hmmmm, well I might come back to this one. From what I recall the dynamic tini has little success in containers. So I'm not sure only building the dynamic tini for glibc would suffice.

closed 2021-01-08 by neheb