openwrt/packages #11845: docker-ce: unable to access host if useing macvlan

feckert, 2020-04-14:
Maintainer: @G-M0N3Y-2503
Environment: x86_64, APU3, OpenWrt master

Description:
I know this is a common problem with macvlan. But for this possibility there is a workaround that works for debian and ubuntu. Only I can't get it to work with OpenWrt.

https://blog.oddbit.com/post/2018-03-12-using-docker-macvlan-networks/

Now I ask myself if this is a problem of OpenWrt and docker or is this now wanted. This can also be related to the kernel which is quite new in openwrt. My research on the internet showed in some comments to this topic that all packets destined for the host where the macvlan is running are filtered.

That what i want I think is a common procedure when I port VMs to docker env.
So is there a option that i can set? Or does I miss something!

Summary what I want and does not work:
I want to connect to the internet via macvlan from my container on the same host.

Summary what I have and does work:
The connection to other devices in this network segment just works fine.

feckert, 2020-04-17:
@G-M0N3Y-2503 I have fixed my problem! If some one also needs this setup and also has the same problems.
This are the step I did to get this working on openwrt.

1. Add the following lines to /etc/config/network to setup the interface and routing stuff on openwrt.
To get the need interfaces and routing configured by netifd
```
config device 'container1'
		option name 'container1'
		option type 'macvlan'
		option ifname 'eth1'
		option mode 'bridge'

config interface 'docker1'
		option proto 'static'
		option ipaddr '192.168.0.249'
		option netmask '255.255.255.252'
		option ifname 'container1'

config route 'route_docker1'
		option interface 'docker1'
		option target '192.168.0.248'
		option netmask '255.255.255.252'
```
2. Restart the network with `/etc/init.d/network restart` or reboot the system.

3. After that you have also add the interface **docker1** to the firewall lan zone.

4. With this command you have to setup the macvlan for docker. The important thing to note here is the gateway IP. This is not the IP in the lan of the router the gateway IP is of the overlay net (subnet).
**docker network create -d macvlan -o parent=eth1 --subnet 192.168.0.0/24 --gateway 192.168.0.249 --ip-range 192.168.0.248/30 container1**

5. And then your could start your container with
**docker run -it --rm --net container1 --ip 192.168.0.250 alpine**

PS: I have also set the following entries in /etc/sysctl.d because docker complains with `docker info` about this. But I am not sure if this es needed.

```
net.bridge.bridge-nf-call-ip6tables=1
net.bridge.bridge-nf-call-iptables=1
```

closed 2020-04-17 by feckert

G-M0N3Y-2503, 2020-04-17:
Thanks for the update 👍
Would you say your solution is a workaround for OpenWrt? 
Or is the same configuration needed on other distros?
I wonder if the network create is supposed to create those interfaces and rules.

feckert, 2020-04-22:
> Would you say your solution is a workaround for OpenWrt?
> Or is the same configuration needed on other distros?

Apart from the sysctl stuff (that some one could please verify @G-M0N3Y-2503 ) this is the same solution and configuration as on other distros. So I think we should add a sysctl file that enables this stuff on docker installation. Then we are on the safe side.

> I wonder if the network create is supposed to create those interfaces and rules.

As You can see in my uci config the needed interfaces are created.