openwrt/packages #11847: nginx/nginx-utils: Enable access from the link-local address

tiagogaspar8, 2020-04-14:
Maintainer: Thomas Heil <heil@terminal-consulting.de> Ansuel Smith <ansuelsmth@gmail.com> (@heil @Ansuel )
Environment: All architectures on the latest OpenWRT trunk

Description:

Lately, nginx has been gradually converted to uci configuration mode, and OpenWRT's web interface is currently automatically configured at startup. One of those configurations is the IP addresses that nginx listens to, it adds the IPv4 addres, the IPv6 global address, but not the IPv6 link local.
I'd like to ask that the link-local address be added to the listening ports, since it is usefull when someone is running OpenWrt in IPv6 only mode and for example if the DHCPv6-PD doesn't work, there isn't any way of accessing the web interface in a "failsafe" mode.
I have been searching trough code and I see that the addresses are gathered from ubus, and from some tests I have done the ubus doesn't provide the IPv6 link-local addresses so I have no idea how to get them but maybe we can find out a way.

Thanks for the help.

peter-stadler, 2020-04-14:
I will look into it, right now I see only grepping `ip addr show br-lan` …

peter-stadler, 2020-04-16:
We could also re-use the code from `ip addr show br-lan` or look how `dnsmasq` is doing it (a call to `netstat -nlp` shows that it is listening on link-local addresses).

But, there are two more problems, since a link-local IPv6 address needs a [scope id](https://en.wikipedia.org/wiki/IPv6_address#Scoped_literal_IPv6_addresses):
1. Nginx does not support listening on link-local addresses directly, although there is a [workaround](https://trac.nginx.org/nginx/ticket/623) using `/etc/hosts`. I used that for testing, but for production we would likely use another solution …
2. Firefox and Chromium does not allow link-local addresses (for me neither in the location bar nor as a link in a html file). So, enabling them in Nginx would be useful only for other browsers  (I read about Konqueror as well as lynx and tried curl, which support scoped addresses).

Right now, I think it is more suitable to use [ssh for link-local addresses](https://www.itdojo.com/ssh-using-link-local-ipv6-addresses/). For accessing the web interface you can open a tunnel by something [like](https://superuser.com/questions/125043/web-browsing-over-ssh):
`ssh -N -L 1280:localhost:80 -L 12443:localhost:443 root@ROUTERIP%PCINTERFACE`
Then you can access the web interface from the browser from [http](http://localhost:1280/) or [https](https://localhost:12443/).

What do you think?

tiagogaspar8, 2020-04-30:
> We could also re-use the code from `ip addr show br-lan`

I believe that this is a great idea.

> Nginx does not support listening on link-local addresses directly

Can't we just leave the original configuration, which lets nginx bind to all the addresses and let the firewall manage which zones are allowed to connect?

> Firefox and Chromium does not allow link-local addresses

You can, you just can't enter `https://fe80::f958:b9ff:fefe:91f5` you have to enter `https://[fe80::f958:b9ff:fefe:91f5]`

> For accessing the web interface you can open a tunnel

I'll use myself as an example, I use windows, on windows I don't believe you can use something like that. And since the world is moving to IPv6 (slowly but moving) we should be readdy to face it in every way.

peter-stadler, 2020-05-01:
> > We could also re-use the code from `ip addr show br-lan`
>
> I believe that this is a great idea.

I saw also [another possibility](https://github.com/openwrt/packages/pull/11456#issuecomment-616556109) and would wait until we know what is the best way before implementing it.

> > Nginx does not support listening on link-local addresses directly
>
> Can't we just leave the original configuration, which lets nginx bind to all the addresses and let the firewall manage which zones are allowed to connect?

This is a third possibility, but the firewall can only decide it for the whole Nginx process AFAIK. I think it would be good to have a default server that is available on the LAN only (including all `/etc/nginx/conf.d/*.locations`) and allow the user to setup more servers (in `/etc/nginx/conf.d/FQDN.conf` or via UCI), which can be available on the WAN, too.

Then, other packages (e.g. LuCI, Ariang or Etesync) can install their `.locations` files, which become available on LAN initially. Users can include some of these files in their own servers exposing selected apps to the WAN, too.

> > Firefox and Chromium does not allow link-local addresses
> 
> You can, you just can't enter `https://fe80::f958:b9ff:fefe:91f5` you have to enter `https://[fe80::f958:b9ff:fefe:91f5]`

Yes, this works for other IPv6 addresses but not for link local ones. They usually need a scope id (something like `https://[fe80::f958:b9ff:fefe:91f5%enp3s0]`) such that they can be routed. 

I tried to enter both versions in Firefox and Chromium, without success:
* They could not open e.g. `https://[fe80::f958:b9ff:fefe:91f5]` 
* with the scope id `%enp3s0` they opened the Searchpage instead.

It was working at the same time with `curl https://[fe80::f958:b9ff:fefe:91f5%enp3s0]`, so I concluded it was a problem of the browsers. (I found reports that a link in a HTML page would work contrary to entering the address in the bar, but it did not work for me neither. And there were other reports that lynx and Konqueror would support scoped IPv6 addresses, I did not try that).

I do not have a Windows machine right now, so Firefox/Chromium may behave differently there. Which browser did you test?

> > For accessing the web interface you can open a tunnel
> 
> I'll use myself as an example, I use windows, on windows I don't believe you can use something like that. And since the world is moving to IPv6 (slowly but moving) we should be readdy to face it in every way.

Could you test if on Windows work the [tutorial](http://realprogrammers.com/how_to/set_up_an_ssh_tunnel_with_putty.html) for [putty, or plink](https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) with the following command?
`plink.exe -v -L 1280:localhost:80 -pw passphrase root@ROUTERIP%PCINTERFACE`
You could get the [interface id](https://www.itdojo.com/ssh-using-link-local-ipv6-addresses/) (PCINTERFACE) by`ipconfig` or by `netsh interface ipv6 show address` …

tiagogaspar8, 2020-07-07:
Hi!

I'm sorry for the looong delay in making a reply, but things have been hard.
I'm a bit thorn between enabling access from the link-local address (possible or not) cause if we're gonna compare the IPv4's 192.168.1.1 to the IPv6 version we should be enabling access not from the link-local, but from the fd00:: address, and from the scripts I saw, this is already enabled, and with that I mean:

1. OpenWrt randomly generates a ULA prefix at first boot
2. OpenWrt assingns it to each interface
3. Nginx reloads and adds that address to the "listening" addresses

So this is a non issue, cause when OpenWrt has no connection to the outside, devices still recieve an IP addres (from the fd00:: auto generated prefix range), get a dns server from router advertisements, and can access the web interface from the openwrt.lan address even, right?

On a side note, is there any way to enable http2 with the integrated web interface? In the old "no nginx-utils" days I would add the "http2" option to the listening ports and it just worked.

And that brings me to a suggestion, wouldn't it be easier to make nginx listen on both:: and 0.0.0.0, and accesses to the interface be regulated by the firewall?

This brings me to an unrelated topic about the firewall, where there should be a folder for services (for example nginx) to add their rules if they're installed instead of adding them to the "firewall" file and to have them baked in the default firewall configuration.

Also just wanna thank you for sticking with this "issue" and say sorry for the "random" rambling about issues.

peter-stadler, 2020-07-09:
Thank you :-) Right now I am busy, I will write a longer response in the next week.

jow-, 2020-07-09:
Services can publish firewall rule requests via ubus and fw3 will dynamically add rules on demand. You can take a look at the igmpproxy init script for a comprehensive example.

peter-stadler, 2020-07-10:
Thank you, too :-) Although this is really nice, I think a firewall cannot give us what we want: Some server parts of Nginx should be available only locally (by default e.g. LuCI), other parts on the whole Internet (published). But, I will look into the details the next week …

peter-stadler, 2020-07-14:
Hi!

> I'm a bit thorn between enabling access from the link-local address (possible or not) cause if we're gonna compare the IPv4's 192.168.1.1 to the IPv6 version we should be enabling access not from [...]

Yes, the fd00::xx address should work already. Now I think the best would to listen on 0.0.0.0 as well as :: and use allow/deny to restrict a server locally. I will propose that for the new version of nginx-util (I made the changes already on the PR #11456 and will do some more testing).

> So this is a non issue, cause when OpenWrt has no connection to the outside, devices still recieve an IP addres (from the fd00:: auto generated prefix range), get a dns server from router advertisements, and can access the web interface from the openwrt.lan address even, right?

Yes, this should work already (and also if we use the allow/deny approach).

> On a side note, is there any way to enable http2 with the integrated web interface? In the old "no nginx-utils" days I would add the "http2" option to the listening ports and it just worked.

Right now this is not possible for local connections (server parts including `lan_ssl.listen` for that). It will be possible with the new version of nginx-util (independently if we use the allow/deny or a modified listen approach).

I think you know it, but maybe for others reading this: Right now only the package `nginx-all-modules` is compiled with `http2` support (neither `nginx` nor `nginx-ssl` have it), but maybe we will change this soon (and have `nginx-ssl` with `http2` support integrated).

> And that brings me to a suggestion, wouldn't it be easier to make nginx listen on both:: and 0.0.0.0, and accesses to the interface be regulated by the firewall?

(I suppose the firewall is not flexible enough: One Nginx process can handle different server parts that should behave differently imho).

> Also just wanna thank you for sticking with this "issue" and say sorry for the "random" rambling about issues.

Thank you again for your input :-)

tiagogaspar8, 2020-09-15:
Heyyy,

So, it's been some time, sorry for the delay, hope everyone's doing well.

> Services can publish firewall rule requests via ubus and fw3 will dynamically add rules on demand. You can take a look at the igmpproxy init script for a comprehensive example.
I see, thanks for the reminder.

> (I suppose the firewall is not flexible enough: One Nginx process can handle different server parts that should behave differently imho).
I get what you mean, and that's a great point of view.

> I think you know it, but maybe for others reading this: Right now only the package nginx-all-modules is compiled with http2 support (neither nginx nor nginx-ssl have it), but maybe we will change this soon (and have nginx-ssl with http2 support integrated).
I've been watching the great changes you made to the software, and I want to congratulate you. Just one complaint (which might be from my side but you'd know better)
For some reason, when my router boots up, nginx doesn't bind to any port, which is weird. I always have to go to the shell and restart it, after that everything is fine.

Thanks again everyone!

peter-stadler, 2020-09-16:
Thank you, I am very well (but a lot is changing now, so I do not have much time). I hope you are well, too.

> For some reason, when my router boots up, nginx doesn't bind to any port, which is weird. I always have to go to the shell and restart it, after that everything is fine.
Yes, I did experience it sometimes. I think it is a timing issue, when nginx looks up the lan addresses before they are assigned to the interface. I thought about a hack for awaiting that. But if we use the deny/allow approach in the new version, it would not be needed; I finally published a draft PR for the nginx-util: #11456 (to take effect we have to change nginx, too).

tiagogaspar8, 2020-09-19:
I see what you mean, Then I hope this gets merged soon cause this issue is very annoying, and, as a second update, it seems the issue is still present.
I'll be looking and following that pull request.

Thanks!

closed 2020-09-19 by tiagogaspar8