openwrt/packages #11850: Dump1090 daemon won't load (Startup-> Init scripts)

robman501a, 2020-04-15:
I can get dump1090 to run manually but the daemon won't load. I've checked the system log and found this.
`Tue Apr 14 23:53:09 2020 daemon.info procd: Instance dump1090::instance1 s in a crash loop 6 crashes, 0 seconds since last crash`
I'm trying to use the Luci-app-dump1090 page to configure it. How can I tell what is causing dump1090 to crash?
This is what I am using..

> Model	Netgear WNDR3700
> Architecture	Atheros AR7161 rev 2
> Firmware Version	OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.057.55219-13dd17f
> Kernel Version	4.14.171

neheb, 2020-04-17:
Got the same thing with 3.8.0. With 3.8.1, I do not. Unfortunately, it errors saying I have no rtlsdr hardware. Meaning I can't test this properly.

ping @ysc3839

robman501a, 2020-04-17:
I think it's a common bug because I've found the workaround from an older post.
https://forum.archive.openwrt.org/viewtopic.php?id=62440&

neheb, 2020-05-08:
How about the latest version? There is also https://github.com/openwrt/packages/pull/12104

vale-max, 2020-06-01:
ok, fought a bit with it and found the issue.
The dump1090 version in openwrt package has no embedded http server, it relies on the uhttpd instance configured.
Unfortunately the config file included in the package has:
option html_dir '/usr/share/dump1090'
This causes dump1090 to be launched with option "--html-dir" that causes it to fail.
It is enough to remove option html_dir to have it working.

Package luci-app-dump1090 should be fixed too:
html_dir should not valorize the dump1090 config item but the uhttpd instance home directory

write_json should default to the data directory of the html directory, which is a symlink pointing to dir /var/run/dump1090

This would save quite some headache setting all up.

andriej, 2022-04-05:
To anyone who may end up here: just make it empty: ( '' ) in /etc/config/dump1090 at html-dir option

Still valid tho