openwrt/packages #11852: gccgo openwrt

server46, 2020-04-15:
how do I add gccgo to the master branch I really need it

yousong, 2020-04-15:
Try build it by enabling toolchain option `INSTALL_GCCGO`

https://github.com/openwrt/openwrt/blob/ad19751edc21ae713bd95df6b93be64bd1e0c612/toolchain/gcc/Config.in#L74-L80

server46, 2020-04-15:
yes I already know that, but that alone is not enough. I see the gccgo patch for chaos calmer is very different. whereas the eglibc master branch no longer exists

server46, 2020-04-15:
`   diff -ur orig/package/libs/toolchain/Makefile new/package/libs/toolchain/Makefile
--- orig/package/libs/toolchain/Makefile	2015-11-10 08:09:30.086903500 +0000
+++ new/package/libs/toolchain/Makefile	2015-11-15 14:02:44.125873000 +0000
@@ -254,6 +254,33 @@
 	endmenu
 endef
 
+define Package/libgo
+$(call Package/gcc/Default)
+  TITLE:=Go support library
+  DEPENDS+=@INSTALL_GCCGO
+  DEPENDS+=@USE_EGLIBC
+endef
+
+define Package/libgo/config
+       menu "Configuration"
+               depends EXTERNAL_TOOLCHAIN && PACKAGE_libgo
+
+       config LIBGO_ROOT_DIR
+               string
+               prompt "libgo shared library base directory"
+               depends EXTERNAL_TOOLCHAIN && PACKAGE_libgo
+               default TOOLCHAIN_ROOT  if !NATIVE_TOOLCHAIN
+               default "/"  if NATIVE_TOOLCHAIN
+
+       config LIBGO_FILE_SPEC
+               string
+               prompt "libgo shared library files (use wildcards)"
+               depends EXTERNAL_TOOLCHAIN && PACKAGE_libgo
+               default "./usr/lib/libgo.so.*"
+
+       endmenu
+endef
+
 define Package/ldd
 $(call Package/libc/Default)
   DEPENDS:=@!USE_MUSL
@@ -410,11 +437,24 @@
 	$(CP) $(TOOLCHAIN_DIR)/lib/libgfortran.so.* $(1)/usr/lib/
   endef
 
+  define Package/libgo/install
+    $(INSTALL_DIR) $(1)/usr/lib
+    $(if $(CONFIG_TARGET_avr32)$(CONFIG_TARGET_coldfire),,$(CP) $(TOOLCHAIN_DIR)/lib/libgo.so.* $(1)/usr/lib/)
+  endef
+
   define Package/libssp/install
 	$(INSTALL_DIR) $(1)/lib
 	$(CP) $(TOOLCHAIN_DIR)/lib/libssp.so.* $(1)/lib/
   endef
 
+  define Package/libgo/install
+    for file in $(call qstrip,$(CONFIG_LIBGO_FILE_SPEC)); do \
+        dir=`dirname $$$$file` ; \
+        $(INSTALL_DIR) $(1)/$$$$dir ; \
+        $(CP) $(call qstrip,$(CONFIG_LIBGO_ROOT_DIR))/$$$$file $(1)/$$$$dir/ ; \
+    done
+  endef
+
   define Package/libstdcpp/install
 	$(INSTALL_DIR) $(1)/usr/lib
 	$(CP) $(TOOLCHAIN_DIR)/lib/libstdc++.so.* $(1)/usr/lib/
@@ -642,5 +682,6 @@
 $(eval $(call BuildPackage,libthread-db))
 $(eval $(call BuildPackage,librt))
 $(eval $(call BuildPackage,libgfortran))
+$(eval $(call BuildPackage,libgo))
 $(eval $(call BuildPackage,ldd))
 $(eval $(call BuildPackage,ldconfig))
diff -ur orig/toolchain/gcc/Config.in new/toolchain/gcc/Config.in
--- orig/toolchain/gcc/Config.in	2015-11-10 08:10:42.251585900 +0000
+++ new/toolchain/gcc/Config.in	2015-11-15 14:02:44.121873000 +0000
@@ -69,3 +69,10 @@
 	default n
 	help
 	    Build/install GNU fortran compiler ?
+
+config INSTALL_GCCGO
+	bool
+	prompt "Build/install gccgo compiler?" if TOOLCHAINOPTS && !(GCC_VERSION_4_6 || GCC_VERSION_4_6_LINARO)
+	default n
+	help
+	    Build/install GNU gccgo compiler ?
\ No newline at end of file
diff -ur orig/toolchain/gcc/common.mk new/toolchain/gcc/common.mk
--- orig/toolchain/gcc/common.mk	2015-11-10 08:10:42.254522700 +0000
+++ new/toolchain/gcc/common.mk	2015-11-15 14:02:44.129873000 +0000
@@ -88,7 +88,7 @@
 HOST_STAMP_INSTALLED:=$(STAGING_DIR_HOST)/stamp/.gcc_$(GCC_VARIANT)_installed
 
 SEP:=,
-TARGET_LANGUAGES:="c,c++$(if $(CONFIG_INSTALL_LIBGCJ),$(SEP)java)$(if $(CONFIG_INSTALL_GFORTRAN),$(SEP)fortran)"
+TARGET_LANGUAGES:="c,c++$(if $(CONFIG_INSTALL_LIBGCJ),$(SEP)java)$(if $(CONFIG_INSTALL_GFORTRAN),$(SEP)fortran)$(if $(CONFIG_INSTALL_GCCGO),$(SEP)go)"
 
 export libgcc_cv_fixed_point=no
 ifdef CONFIG_USE_UCLIBC
 `

server46, 2020-04-15:
I tried the gccgo patch here [](https://gist.github.com/conoro/887362ebdbbada2cfbc6)

`   go1: warning: '-Werror=' argument '-Werror=format-security' is not valid for Go `

jefferyto, 2020-04-20:
The last time I looked at gccgo (I tried the same patches you have), I gave up and instead went with the Go compiler from golang.org because libgo depended on ucontext (setcontext/getcontext/makecontext), but musl didn't implement it (and didn't appear to have any plans to add support).

This is probably why `INSTALL_GCCGO` depends on `USE_GLIBC`. I don't know what is the current status of libgo/musl with regard to ucontext.

A few links I kept on this:
* https://dev.archive.openwrt.org/ticket/18611
* https://gcc.gnu.org/bugzilla/show_bug.cgi?id=68931
* https://groups.google.com/forum/#!topic/golang-nuts/fAElwJu-QUM
* https://www.openwall.com/lists/musl/2016/02/04/3
* https://wiki.musl-libc.org/open-issues.html#ucontext.h

May I ask, why do you need gccgo? Is there something the existing Go package build system doesn't do that you need?

jefferyto, 2020-04-20:
BTW I believe you can bypass the `-Werror=format-security` error by deselecting the `PKG_CHECK_FORMAT_SECURITY` config option (or adding `PKG_CHECK_FORMAT_SECURITY:=0` to the package Makefile).

predators46, 2020-04-20:
> The last time I looked at gccgo (I tried the same patches you have), I gave up and instead went with the Go compiler from golang.org because libgo depended on ucontext [...]

I just want to use gccgo

predators46, 2020-04-20:
> BTW I believe you can bypass the `-Werror=format-security` error by deselecting the `PKG_CHECK_FORMAT_SECURITY` config option (or adding `PKG_CHECK_FORMAT_SECURITY:=0` to the package Makefile).

	config PKG_CHECK_FORMAT_SECURITY

jefferyto, 2020-04-21:
You can deselect it using `make menuconfig`.

Since gcc is in the main OpenWrt repo and not here, I suggest you file any further issues at https://bugs.openwrt.org/. If you need any further help, please make a post at https://forum.openwrt.org/.

closed 2020-05-02 by BKPepe