openwrt/packages #11863: htop missing backtrace capability in musl

nrobert13, 2020-04-16:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: champetier.etienne@gmail.com
Environment: (put here arch, model, OpenWrt version)
mvebu 
Linksys WRT1900ACS 
v19.07.2 with musl

Description:
Building htop for  fails with the following error. This is due to the fact that musl doesn't implement the backtrace capability opposed to glibc.
```
arm-openwrt-linux-muslgnueabi-gcc -pedantic -Wall -Wextra -std=c99 -D_XOPEN_SOURCE_EXTENDED -DSYSCONFDIR=\"/etc\" -I"./linux" -rdynamic -D_GNU_SOURCE -D_DEFAULT_SOURCE -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/nrobert/opt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/htop-2.2.0:htop-2.2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -L/home/nrobert/opt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/home/nrobert/opt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/home/nrobert/opt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/home/nrobert/opt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -znow -zrelro  -o htop   AvailableMetersPanel.o CategoriesPanel.o CheckItem.o ClockMeter.o ColorsPanel.o ColumnsPanel.o CPUMeter.o CRT.o MainPanel.o DisplayOptionsPanel.o FunctionBar.o Hashtable.o Header.o htop.o ListItem.o LoadAverageMeter.o MemoryMeter.o Meter.o MetersPanel.o Object.o Panel.o BatteryMeter.o Process.o ProcessList.o RichString.o ScreenManager.o Settings.o SignalsPanel.o StringUtils.o SwapMeter.o TasksMeter.o UptimeMeter.o TraceScreen.o UsersTable.o Vector.o AvailableColumnsPanel.o AffinityPanel.o HostnameMeter.o OpenFilesScreen.o Affinity.o IncSet.o Action.o EnvScreen.o InfoScreen.o XAlloc.o linux/Platform.o linux/IOPriorityPanel.o linux/IOPriority.o linux/LinuxProcess.o linux/LinuxProcessList.o linux/LinuxCRT.o linux/Battery.o  -lncurses -ltinfo -lm  
/home/nrobert/opt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: linux/LinuxCRT.o: in function `CRT_handleSIGSEGV':
LinuxCRT.c:(.text+0x28): undefined reference to `backtrace'
/home/nrobert/opt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: LinuxCRT.c:(.text+0x48): undefined reference to `backtrace_symbols_fd'
collect2: error: ld returned 1 exit status
Makefile:820: recipe for target 'htop' failed
make[5]: *** [htop] Error 1
make[5]: Leaving directory '/home/nrobert/opt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/htop-2.2.0'
Makefile:603: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/nrobert/opt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/htop-2.2.0'
Makefile:55: recipe for target '/home/nrobert/opt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/htop-2.2.0/.built' failed
make[3]: *** [/home/nrobert/opt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/htop-2.2.0/.built] Error 2
```

nrobert13, 2020-04-16:
https://github.com/openwrt/packages/pull/11864

nrobert13, 2020-04-16:
https://github.com/openwrt/packages/pull/11865

hnyman, 2020-04-18:
> Building htop for fails with the following error. 

htop compiles just fine with musl.
Are you using some special config/options???

nrobert13, 2020-04-18:
```
 scripts/diffconfig.sh | egrep -v "(^#|^$|PACKAGE|BUSYBOX|DEFAULT|OPEN)" config_lede_w_tools 
CONFIG_DEVEL=y
CONFIG_BUILD_LOG=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_RSYNC_zlib=y
CONFIG_TVHEADEND_CWC_SUPPORT=y
CONFIG_TVHEADEND_LINUXDVB_SUPPORT=y
CONFIG_TOOLCHAINOPTS=y
```

hnyman, 2020-04-19:
Well, you grepped away all interesting package selections as well as the package disables with "#", which are part of the config...

Just show the diffconfig output.

nrobert13, 2020-04-19:
I thought the package selection would influence this... here's the diffconfig,
```
 scripts/diffconfig.sh                                                                                                                       [176/1934]
CONFIG_TARGET_mvebu=y                                                                                                                                                                                       
CONFIG_TARGET_mvebu_cortexa9=y                                                                                                                                                                              
CONFIG_TARGET_mvebu_cortexa9_DEVICE_linksys_wrt1900acs=y                                                                                                                                                    
CONFIG_DEVEL=y                                                                                                                                                                                              
CONFIG_TOOLCHAINOPTS=y                                                                                                                                                                                      
CONFIG_BUILD_LOG=y                                                                                                                                                                                          
CONFIG_GNUTLS_ALPN=y                                                                                                                                                                                        
CONFIG_GNUTLS_ANON=y                                                                                                                                                                                        
CONFIG_GNUTLS_DTLS_SRTP=y                                                                                                                                                                                   
CONFIG_GNUTLS_HEARTBEAT=y                                                                                                                                                                                   
CONFIG_GNUTLS_OCSP=y                                                                                                                                                                                        
CONFIG_GNUTLS_PSK=y                                                                                                                                                                                         
CONFIG_LIBCURL_COOKIES=y                                                                                                                                                                                    
CONFIG_LIBCURL_FILE=y                                                                                                                                                                                       
CONFIG_LIBCURL_FTP=y                                                                                                                                                                                        
CONFIG_LIBCURL_HTTP=y                                                                                                                                                                                       
CONFIG_LIBCURL_MBEDTLS=y                                                                                                                                                                                    
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_NPN=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENVPN_openssl_ENABLE_DEF_AUTH=y
CONFIG_OPENVPN_openssl_ENABLE_FRAGMENT=y
CONFIG_OPENVPN_openssl_ENABLE_LZ4=y
CONFIG_OPENVPN_openssl_ENABLE_LZO=y
CONFIG_OPENVPN_openssl_ENABLE_MULTIHOME=y
CONFIG_OPENVPN_openssl_ENABLE_PF=y
CONFIG_OPENVPN_openssl_ENABLE_PORT_SHARE=y
CONFIG_OPENVPN_openssl_ENABLE_SERVER=y
CONFIG_OPENVPN_openssl_ENABLE_SMALL=y
CONFIG_PACKAGE_NTFS-3G_HAS_PROBE=y
CONFIG_PACKAGE_ar=m
CONFIG_PACKAGE_bind-dig=y
CONFIG_PACKAGE_bind-libs=y
CONFIG_PACKAGE_binutils=m
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_ca-bundle=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_ddns-scripts=y
CONFIG_PACKAGE_diffutils=y
CONFIG_PACKAGE_e2fsprogs=y
CONFIG_PACKAGE_ebtables=y
CONFIG_PACKAGE_ebtables-utils=y
CONFIG_PACKAGE_expect=m                                                                                                                                                                           [122/1934]
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_gdb=m
CONFIG_PACKAGE_gdbserver=m
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_iftop=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_ipset=y
CONFIG_PACKAGE_ipset-dns=y
CONFIG_PACKAGE_kmod-br-netfilter=y
CONFIG_PACKAGE_kmod-crypto-aead=y
CONFIG_PACKAGE_kmod-crypto-crc32c=y
CONFIG_PACKAGE_kmod-crypto-des=y
CONFIG_PACKAGE_kmod-crypto-ecb=y
[... 136 lines cut ...]
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
[... cut ...]
CONFIG_PACKAGE_tvheadend=m
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_usbreset=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_uvcdynctrl=m
CONFIG_PACKAGE_wifischedule=y
CONFIG_PACKAGE_wireless-tools=y
CONFIG_PACKAGE_xupnpd=m
CONFIG_PACKAGE_zlib=y
CONFIG_RSYNC_zlib=y
CONFIG_TVHEADEND_CWC_SUPPORT=y
CONFIG_TVHEADEND_LINUXDVB_SUPPORT=y
# CONFIG_PACKAGE_python3-click-log-src is not set
# CONFIG_PACKAGE_python3-decorator-src is not set
# CONFIG_PACKAGE_python3-schema-src is not set

hnyman, 2020-04-19:
That is a massive set of packages.
Possibly the htop configure script finds a bit too much available functionality, and tries to compile also that backtrace support. Or something similar.  (If your htop vanilla compilation really fails, without you having tried to somehow force the backtrace to be active.)

Note that buildbot builds htop nicely for both master and 19.07, so it is possibly something in your config that triggers the breakage.

Your patches would introduce backtrace features and the libexecinfo library as mandatory part of htop for all users, and I am not sure if that is really appropriate. If it would be included, it should be optional, I think.

nrobert13, 2020-04-19:
Agree with you, but not sure how can we make it optional ...

micmac1, 2020-04-21:
I think there must be something special about your buildroot. LinuxCRT.c only uses backtrace in case HAVE_EXECINFO_H is set by configure script. That only happens when configure detects a usable execinfo.h, which shouldn't exist in your buildroot with musl.

Can you check where the header is located in your buildroot?

micmac1, 2020-04-21:
You can also look at config.log and search for "checking execinfo.h usability" if that helps. That should at least give you the include directories used.

micmac1, 2020-04-21:
Now I get it, you have libexecinfo because you've prepared pull request #11860 

htop doesn't have an off-switch for libexecinfo. You'd need to spin up a patch for that.

micmac1, 2020-04-21:
Or alternatively a patch to link to the external lib your pull request is providing.

nrobert13, 2020-04-21:
the cause effect is actually the other way around. I get this error during build, so I created the PR #11865 and #11864 which would fix it. now that you said that LinuxCRT is using backtrace only if configure detects execinfo.h, I realised that I introduced this libexecinfo some time back, because tvheadend build was failing due to the same error. So already having libexecinfo in my buildroot, htop also enables backtrace capability. Unfortunately I'm not proficient enough in C to patch htop...

micmac1, 2020-04-21:
Ah, thanks for clearing this up. This was confusion me quite a bit.

So I would suggest remove libexecinfo from your buildroot and be done with it, close this issue and the two pull requests of yours. I take it that you don't need htop linked against libexecinfo for debugging stuff.

I think introducing external libexecinfo like this could cause the same problem that you found with htop for other packages as well. More headache :)

nrobert13, 2020-04-21:
you are right, I don't need backtrace, but I cannot remove libexecinfo due to tvheadend. I need tvheadend, which doesn't build without it.

micmac1, 2020-04-21:
Understandable, but that's not an OpenWrt problem. You have your customized tvheadend package. That's why you need libexecinfo. Right?

And you can remove your libexecinfo before compiling htop. And when you want to compile your tvheadend again you can install libexecinfo again.

nrobert13, 2020-04-21:
true. guess what. Just tried to remove the libexecinfo from buildroot and remove it from tvheadend package as well, and it worked. thanks for helping me figure this out. closing this and the PR's.

closed 2020-04-21 by nrobert13