openwrt/packages #11873: minidlna 2019-12-09

er2de2, 2020-04-18:
Maintainer: @neheb
Environment: ramips openwrt 19.07

Applying ./patches/020-wrap_container_definitions_into_a_structure.patch using plaintext: 
patching file containers.c
Hunk #1 succeeded at 48 with fuzz 2 (offset 25 lines).
patching file containers.h
Hunk #1 succeeded at 24 with fuzz 2 (offset 8 lines).
patching file scanner.c
Hunk #1 FAILED at 529.
Hunk #2 FAILED at 576.
2 out of 2 hunks FAILED -- saving rejects to file scanner.c.rej
Patch failed!  Please fix ./patches/020-wrap_container_definitions_into_a_structure.patch!
Makefile:73: recipe for target '/home/roman/openwrt/build_dir/target-mipsel_24kc_musl/minidlna-2019-12-09-0763719f/.prepared_668c7af06309044395302eccbfc08b9b_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/roman/openwrt/build_dir/target-mipsel_24kc_musl/minidlna-2019-12-09-0763719f/.prepared_668c7af06309044395302eccbfc08b9b_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/roman/openwrt/feeds/packages/multimedia/minidlna'
time: package/feeds/packages/minidlna/compile#0.42#0.21#0.58
package/Makefile:111: recipe for target 'package/feeds/packages/minidlna/compile' failed
make[2]: *** [package/feeds/packages/minidlna/compile] Error 2

neheb, 2020-04-18:
Try now

er2de2, 2020-04-18:
Works, thank you

closed 2020-04-18 by er2de2