openwrt/packages #11879: Simple-adblock: TypeError result[0].trim is not a function

u8wdz7, 2020-04-18:
Maintainer: Stan Grishin <stangri@melmac.net>
Environment: ARMv7 Processor rev 0 (v7l), TP-Link Archer C2600, OpenWrt 19.07.2 r10947

Description: After installation the luci console shows the below error message at the Services->Adblock->Overview menu and nod loading anything:
TypeError
result[0].trim is not a function

[image]

dibdot, 2020-04-18:
You've installed adblock ... to fix the above error just update your local package repo (opkg update) and install the updated adblock & luci-app-adblock packages.

closed 2020-04-18 by dibdot

Lu5ck, 2020-04-20:
> You've installed adblock ... to fix the above error just update your local package repo (opkg update) and install the updated adblock & luci-app-adblock packages.

That doesn't fix it. I already install the latest adblock and luci shown in the list.

u8wdz7, 2020-04-22:
Restart solved the issue for me.

NatoBoram, 2020-05-04:
I've got the same error.

```yaml
OS: OpenWrt 19.07.2 armv7l
Host: Linksys WRT3200ACM
Kernel: 4.14.171
Uptime: 18 mins
Packages: 179 (opkg)
Shell: ash
Terminal: /dev/pts/0
CPU: Marvell Armada 380/385 (Device Tree) (2)
Memory: 181MiB / 499MiB
```

> Package adblock (4.0.4-4) installed in root is up to date.
> Package luci-app-adblock (git-20.105.64099-eeba108-1) installed in root is up to date.

jeff47, 2020-05-06:
I am seeing the same issue with LuCI openwrt-19.07 branch (git-20.125.82541-d320179) / OpenWrt 19.07.2 r10947-65030d81f3.  This was a fresh install so no prior adblock package to upgrade, brand new installs.

spicem, 2020-06-22:
same issue. reboot not helping

zombified, 2021-04-05:
FWIW I had this and then made sure to: 1) update package list through the web interface, 2) made sure adblock and the luci-app-adblock packages were updated, 3) upgraded all other packages that had updates available, and 4) rebooted the router. Worked fine after that.

xiaozhuai, 2021-09-15:
Restart `rpcd` solve this
```
service rpcd restart
```