openwrt/packages #11881: isc-dhcp-* : no rules to make target

moocan, 2020-04-18:
Maintainer: @pprindeville @neheb @yaheath @ja-pa @brianjmurrell @antonyantony 
Environment: ipq806x generic
Description: 

I'm trying desperately to build isc-dhcp-server-ipv6 using OpenWRT 19.07.2 build system.
due to a segmentation fault of dhcpd ipv4/ipv6.

After installing a new build system, updating all feeds, installing all feeds package, installing feeds package `isc-dhcp`, selecting` <M> isc-dhcp-server-ipv6` in menuconfig and compiling `zlib`

`make package/isc-dhcp/compile V=sc`, or even `clean`, `check` .... 
`make[1]: *** No rule to make target 'packages/isc-dhcp/compile'`

I can compile other package using the same build system.
Using a global make all packages are successfully built but no `isc-dhcp-*`.

Thank you in advance for your help.

Kind Regards

hnyman, 2020-04-19:
Works for me on 19.07 build system for ipq806x.
I selected just  `<M> isc-dhcp-server-ipv6` and compile works quite ok, and the .ipk package gets created.

```
perus@ub1910:/Openwrt/r7800-1907$ ./scripts/getver.sh 
r11007-55312cc202

perus@ub1910:/Openwrt/r7800-1907$ grep isc-dhcp .config
# CONFIG_PACKAGE_isc-dhcp-client-ipv4 is not set
# CONFIG_PACKAGE_isc-dhcp-client-ipv6 is not set
# CONFIG_PACKAGE_isc-dhcp-omshell-ipv4 is not set
# CONFIG_PACKAGE_isc-dhcp-omshell-ipv6 is not set
# CONFIG_PACKAGE_isc-dhcp-relay-ipv4 is not set
# CONFIG_PACKAGE_isc-dhcp-relay-ipv6 is not set
# CONFIG_PACKAGE_isc-dhcp-server-ipv4 is not set
CONFIG_PACKAGE_isc-dhcp-server-ipv6=m

perus@ub1910:/Openwrt/r7800-1907$ make package/isc-dhcp/compile
 make[1] package/isc-dhcp/compile
 make[2] -C package/libs/toolchain compile
 make[2] -C package/libs/zlib compile
 make[2] -C feeds/packages/net/isc-dhcp compile

perus@ub1910:/Openwrt/r7800-1907$ ls -l bin/packages/arm_cortex-a15_neon-vfpv4/packages/isc*
-rw-r--r-- 1 perus perus 1089072 huhti  19 20:06 bin/packages/arm_cortex-a15_neon-vfpv4/packages/isc-dhcp-server-ipv6_4.4.1-3_arm_cortex-a15_neon-vfpv4.ipk
```

closed 2020-04-19 by hnyman

moocan, 2020-04-20:
Hello,

Many thanks for your answer.
I have found yesterday how to successfully build `isc-dhcp` packages.
For me it works only if I select all `isc-dhcp` packages as module with menuconfig.
Then I successfully patched `isc-dhcp-server-* 4.4.1` with some ISC patch and built it and then updated and build to the last `4.4.2` release but I always have a `segmentation fault`

Impossible to get DDNS working.
I have no choice but to give up.

Kind Regards

pprindeville, 2020-04-21:
> Then I successfully patched isc-dhcp-server-* 4.4.1 with some ISC patch and built it and then updated and build to the last 4.4.2 release but I always have a segmentation fault

What would this patch look like/be called/have come from?