openwrt/packages #11900: meson issues with glib2

neheb, 2020-04-20:
Maintainer: @dhewg 
Environment: hosts lacking glib2

With the update to 2.64.2, several issues were uncovered. Most can be solved, but I'm stuck on these two.

 - HOST_LDFLAGS get applied to target builds.
 - Installed hostpkg libraries do not link properly.

Running ldd locally reveals that the hostpkg libraries link to the host's lib64 directory.

Running ldd on a debian VM lacking glib2 shows not found.

This causes problems with any package that depends on glib2/host.

I tried adding -Wl,--rpath-link=$(STAGING_DIR_HOSTPKG)/lib to HOST_LDFLAGS but it doesn't seem to do anything.

neheb, 2020-04-20:
From ninja -v:

```
gcc  -o gio/gdbus 'gio/6ae6c9e@@gdbus@exe/gdbus-tool.c.o' -L/home/mangix/devstuff/openwrt/staging_dir/host/lib -L/home/mangix/devstuff/openwrt/staging_dir/hostpkg/lib -L/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_uClibc/host/lib -Wl,--no-undefined -Wl,--start-group -liconv gio/libgio-2.0.so.0.6400.2 gmodule/libgmodule-2.0.so.0.6400.2 glib/libglib-2.0.so.0.6400.2 gobject/libgobject-2.0.so.0.6400.2 -Wl,--end-group -pthread '-Wl,-rpath,$ORIGIN/:$ORIGIN/../glib:$ORIGIN/../gobject:$ORIGIN/../gmodule:/home/mangix/devstuff/openwrt/staging_dir/hostpkg/lib' -Wl,-rpath-link,/home/mangix/devstuff/openwrt/build_dir/hostpkg/glib-2.64.2/openwrt-build/gio -Wl,-rpath-link,/home/mangix/devstuff/openwrt/build_dir/hostpkg/glib-2.64.2/openwrt-build/glib -Wl,-rpath-link,/home/mangix/devstuff/openwrt/build_dir/hostpkg/glib-2.64.2/openwrt-build/gobject -Wl,-rpath-link,/home/mangix/devstuff/openwrt/staging_dir/hostpkg/lib -Wl,-rpath-link,/home/mangix/devstuff/openwrt/build_dir/hostpkg/glib-2.64.2/openwrt-build/gmodule
```

I have no idea what's wrong there.  It's specifying hostpkg but linking against the host OS.

micmac1, 2020-04-20:
Looks like meson is still playing catch-up with cmake regarding rpath handling: [link](https://github.com/mesonbuild/meson/issues/314)

Buildroot colleagues patch meson for this: [link](https://gitlab.com/buildroot.org/buildroot/-/blob/master/package/meson/0001-Only-fix-RPATH-if-install_rpath-is-not-empty.patch)

Without patch:

```
sk@darth ~/tmp/openwrt/staging_dir/hostpkg $ ldd ./bin/glib-compile-schemas
	linux-vdso.so.1 (0x00007ffe53cbc000)
	libglib-2.0.so.0 => /usr/lib64/libglib-2.0.so.0 (0x00007f50444e4000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f5044315000)
	libpcre.so.1 => /lib64/libpcre.so.1 (0x00007f50442a2000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f504427f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f5044626000)
sk@darth ~/tmp/openwrt/staging_dir/hostpkg $
```

With patch:

```
sk@darth ~/tmp/openwrt/staging_dir/hostpkg $ ldd ./bin/glib-compile-schemas
	linux-vdso.so.1 (0x00007ffefc3a3000)
	libglib-2.0.so.0 => /home/sk/tmp/openwrt/staging_dir/hostpkg/lib/libglib-2.0.so.0 (0x00007f16cf2d7000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f16cf0f6000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f16cf0d3000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f16cf439000)
sk@darth ~/tmp/openwrt/staging_dir/hostpkg $
```

micmac1, 2020-04-21:
>     * HOST_LDFLAGS get applied to target builds.

Looks like that is happening because during target compile both cross-compile and host info is given to Meson. It then mixes things up.

One way to prevent that would be:

```
diff --git a/devel/meson/meson.mk b/devel/meson/meson.mk
index 16c6b1668..f669faeb5 100644
--- a/devel/meson/meson.mk
+++ b/devel/meson/meson.mk
@@ -95,11 +95,9 @@ define Host/Uninstall/Meson
 endef

 define Build/Configure/Meson
-       $(call Meson/CreateNativeFile,$(PKG_BUILD_DIR)/openwrt-native.txt)
        $(call Meson/CreateCrossFile,$(PKG_BUILD_DIR)/openwrt-cross.txt)
        $(call Meson, \
                --buildtype plain \
-               --native-file $(PKG_BUILD_DIR)/openwrt-native.txt \
                --cross-file $(PKG_BUILD_DIR)/openwrt-cross.txt \
                $(MESON_ARGS) \
                $(MESON_BUILD_DIR) \
```

I checked how buildroot does its setup. It doesn't provide a native file either. See [here](https://gitlab.com/buildroot.org/buildroot/-/blob/master/package/pkg-meson.mk).

dhewg, 2020-04-21:
There are projects which needs both (like mesa building native tools which generate source files), but maybe openwrt doesn't need this atm, I'm not sure.

@neheb You seem active and if you care enough I wouldn't mind you taking over as maintainer, I don't have much time recently to even look at issues...

neheb, 2020-04-22:
@micmac1 I will test those solutions and see what sticks.

neheb, 2020-04-25:
I've merged the linked PR. So this is fixed.

closed 2020-04-25 by neheb