openwrt/packages #11908: miniupnpd: cannot open STUN port for testing

Menion2k, 2020-04-21:
Maintainer: @neheb
Environment: (OpenWRT 19.07.2)

Description: miniupnpd cannot open STUN UDP port for STUN test

Hi all

I can replicate the issue descibed here:  https://forum.openwrt.org/t/miniupnpd-in-trunk-stun-setup/18688 (cannot make STUN without opening UDP port on firewall)

I opened a ticket on miniupnpd github but the dev explained that miniupnpd opens the UDP port for STUN testing on the firewall: https://github.com/miniupnp/miniupnp/issues/437

I looked at the actual miniupnpd code OpenWRT is using and the port opening is there, but it is clear it does not work, maybe due to the way the firewall is configured on OpenWRT?

Bye

ptpt52, 2020-04-26:
patch to work:
```
diff --git a/net/miniupnpd/files/miniupnpd.hotplug b/net/miniupnpd/files/miniupnpd.hotplug
index 29f9cdbd1..78bd2924a 100644
--- a/net/miniupnpd/files/miniupnpd.hotplug
+++ b/net/miniupnpd/files/miniupnpd.hotplug
@@ -1,6 +1,6 @@
 #!/bin/sh /etc/rc.common
 
-enabled miniupnpd || exit 0
+/etc/init.d/miniupnpd enabled || exit 0
 
 # If miniupnpd is not running:
 # - check on _any_ event (even updates may contribute to network_find_wan*)

```

Menion2k, 2020-04-28:
Do you know where the content of this file ends up on the firmware? I would like to fix it without recompiling the firmware

neheb, 2020-05-10:
This fix was pushed to the 19.07 branch. Update should be available tomorrow.

Neustradamus, 2021-08-31:
@neheb: I think it is solved now no?
- https://github.com/openwrt/packages/commit/b7c59c390f0b739403ae9099e1affe94d214b51d#diff-1978da76df9336b03f5c4519d38eed3f84ff696b013de10178ba9e95a9c25cd2

Neustradamus, 2021-08-31:
@neheb: I think it is solved now no?
- https://github.com/openwrt/packages/commit/b7c59c390f0b739403ae9099e1affe94d214b51

closed 2021-08-31 by neheb