openwrt/packages #11912: [19.07] Can not install Python 3 packages which requires compilation when using ccache

BKPepe, 2020-04-21:
**Maintainers:** @commodo , @jefferyto 
**Environment:** Turris OS* - Turris 1.x (powerpc8540), Turris Omnia (mvebu, cortexa9), Turris MOX (mvebu, cortexa53) . The latest OpenWrt 19.07. Shouldn't be a problem to reproduce it on vanilla OpenWrt. More details below. I haven't tested it on master, so far.
           *we are using .pyc files by default and we are using ccache_cc.
**Reported to me** by @Shulyaka and I was able to reproduce it.

Two issues:
1. If you are using only .pyc files, pip3 is looking for .py files. I think this can be solved very easily.
2. If you are using ccache_cc, it can not install packages unless you do symlink for gcc. Not good. [ccache](https://github.com/openwrt/openwrt/blob/master/tools/ccache/Makefile) exists as host package.

Steps to reproduce it:
1. In `make menuconfig`, in `Advanced configuration options (for developers)` select `Use ccache`.
2. Compile Python3
3. Install it on your router.

Description:
Install python3-pip.
```
root@turris:~# opkg install python3-pip
Installing python3-pip (19.2.3-1.17) to root...
Downloading https://repo.turris.cz/hbt/omnia/packages/packages/python3-pip_19.2.                                                                                                                                                             3-1_arm_cortex-a9_vfpv3.ipk
Installing python-pip-conf (0.1-1.0) to root...
Downloading https://repo.turris.cz/hbt/omnia/packages/packages/python-pip-conf_0                                                                                                                                                             .1-1_arm_cortex-a9_vfpv3.ipk
Configuring python-pip-conf.
Configuring python3-pip.
```
Files of python3-pip.
```
root@turris:~# opkg files python3-pip
Package python3-pip (19.2.3-1.17) is installed on root and has the following files:
/usr/lib/python3.7/site-packages/pip/_vendor/certifi/__main__.pyc
/usr/lib/python3.7/site-packages/pip/_vendor/webencodings/tests.pyc
/usr/lib/python3.7/site-packages/pip/_vendor/distlib/compat.pyc
/usr/lib/python3.7/site-packages/pip/_internal/distributions/source.pyc
/usr/lib/python3.7/site-packages/pip/_vendor/cachecontrol/adapter.pyc
/usr/lib/python3.7/site-packages/pip/_vendor/chardet/gb2312prober.pyc
/usr/lib/python3.7/site-packages/pip/_vendor/chardet/gb2312freq.pyc
/usr/lib/python3.7/site-packages/pip/_vendor/urllib3/exceptions.pyc
/usr/lib/python3.7/site-packages/pip/_vendor/chardet/langbulgarianmodel.pyc
and a lot more
```

Install any package which requires compilation (i.e. [PEP517](https://www.python.org/dev/peps/pep-0517/)). In this issue, I will mention multidict, which I packaged for OpenWrt myself, but yeah I can install it, but what about other packages.

```
root@turris:~# pip3 install multidict
  Downloading https://files.pythonhosted.org/packages/61/b4/475114b3f1671da634f89239e61038f8742d9ac13aa34b32a05bf8022d22/multidict-4.7.5.tar.gz (50kB)
     |████████████████████████████████| 51kB 1.1MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
ERROR: Command errored out with exit status 2: /usr/bin/python3 /usr/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpuo8pxpx3 Check the logs for full command output.
WARNING: You are using pip version 19.2.3, however version 20.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

And that does not work. Looking at log /tmp/pip-log.txt, I see
```
2020-04-21T19:53:53,020   Running command /usr/bin/python3 /usr/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpuo8pxpx3
2020-04-21T19:53:53,155   /usr/bin/python3: can't open file '/usr/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py': [Errno 2] No such file or directory
```

Ok. This I can solve by copying .pyc to .py.
`cp /usr/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.pyc /usr/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py`

I tried to install it once again, but this time, it ended up with:

```
2020-04-21T20:01:19,150   building 'multidict._multidict' extension
2020-04-21T20:01:19,151   creating build/temp.linux-armv7l-3.7
2020-04-21T20:01:19,151   creating build/temp.linux-armv7l-3.7/multidict
2020-04-21T20:01:19,152   ccache_cc -Wno-unused-result -Wsign-compare -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/ourserverpath/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/Python-3.7.7:Python-3.7.7 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/ourserverpath/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/Python-3.7.7:Python-3.7.7 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/ourserverpath/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/Python-3.7.7:Python-3.7.7 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/ourserverpath/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I/ourserverpath/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/include -I/ourserverpath/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/include -I/ourserverpath/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/include/fortify -I/ourserverpath/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/include -fPIC -I/usr/include/python3.7 -c multidict/_multidict.c -o build/temp.linux-armv7l-3.7/multidict/_multidict.o -O2 -std=c99 -Wall -Wsign-compare -Wconversion -fno-strict-aliasing -pedantic
2020-04-21T20:01:19,158   error: command 'ccache_cc' failed with exit status 1
2020-04-21T20:01:19,260   ERROR: Failed building wheel for multidict
2020-04-21T20:01:19,261   Running setup.py clean for multidict
```

____

The second issue was not exactly reproduced on Xiaomi Mi WiFi R3G, ramips, mt7621, OpenWrt 19.07 by @mmtj. When he tried to install package multidict, it leads to this error:

```
 root@OpenWrt:~# pip3 install multidict
Collecting multidict
  Downloading https://files.pythonhosted.org/packages/61/b4/475114b3f1671da634f89239e61038f8742d9ac13aa34b32a05bf8022d22/multidict-4.7.5.tar.gz (50kB)
     |████████████████████████████████| 51kB 771kB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
ERROR: Command errored out with exit status 2: /usr/bin/python3 /usr/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpw8xtahmg Check the logs for full command output.
WARNING: You are using pip version 19.2.3, however version 20.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
Looking at `/tmp/pip-log.txt` it failed because of:
`2020-04-21T10:12:54,760   /usr/bin/python3: can't open file '/usr/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py': [Errno 2] No such file or directory`

It was weird as the file is there, but he tried to remove it and install pip again and it was working, but it required to install `python3-dev` and `gcc` to install package `multidict` via `pip`.
```
root@OpenWrt:~# ls -l /usr/lib/python3.7/site-packages/pip/_vendor/pep517/
-rw-r--r--    1 root     root            84 Apr 21 10:16 __init__.py
drwxr-xr-x    2 root     root           968 Apr 21 10:17 __pycache__
-rw-r--r--    1 root     root          7792 Apr 21 10:16 _in_process.py
-rw-r--r--    1 root     root          3318 Apr 21 10:16 build.py
-rw-r--r--    1 root     root          5949 Apr 21 10:16 check.py
-rw-r--r--    1 root     root          4098 Apr 21 10:16 colorlog.py
-rw-r--r--    1 root     root           780 Apr 21 10:16 compat.py
-rw-r--r--    1 root     root          1129 Apr 21 10:16 dirtools.py
-rw-r--r--    1 root     root          6024 Apr 21 10:16 envbuild.py
-rw-r--r--    1 root     root          2463 Apr 21 10:16 meta.py
-rw-r--r--    1 root     root         10599 Apr 21 10:16 wrappers.py
```

commodo, 2020-04-22:
Compiling Python packages on the target is a bit uncommon.
So, we don't support it much. Well, I did not use to support it much.
And I am bit vague about it; I think `python3-dev` and `python-dev` were suggested by someone else via some PR, so I did not have much involvement directly into it.

One question is: how does this work on other distros? We can probably try to adapt it a bit with some input from them.
But given that it works with `python3-dev` & symlinking to `gcc`, it looks fine to me.
I think `python3-dev` being required on the target is mandatory to compile Python packages, but no idea if pip should always install it. Maybe document this?
Maybe there's a better way to provide the compiler to pip? Via OS env-vars or some other config? I'd be surprised they didn't support this already somehow.

jefferyto, 2020-04-23:
I don't think the ccache issue is limited to ccache... While testing #11942 (armvirt-64, qemu) I tried `pip3 install multidict` with python3 built by buildbots. pip wanted to run `aarch64-openwrt-linux-musl-gcc`. (I tried it with and without python3-dev installed, I got the same result.)

Basically the issue is, target Python remembers the host compiler program used to compile it, then wants to use the same program when compiling on target. It would be best if target Python can remember `gcc` instead.

I don't think other distros will have this issue since not many do cross-compilation for their packages. I think I'll have to study what Python is doing a bit more before knowing what is the best way to fix this.

jefferyto, 2020-04-23:
Oh the gcc package installs the compiler binary with the [full target name](https://github.com/openwrt/packages/blob/a4bb706918c58c7f8718e5de1de2e719eecabbd2/devel/gcc/Makefile#L190), so the default case (without ccache) works as expected :man_facepalming: 

The cheap/quick solution would be to use `TARGET_CC_NOCACHE` instead of `TARGET_CC` to compile target Python; compilation may be slower but the compiler name will be right.

commodo, 2020-04-24:
oohhh
makes sense;
maybe we need to adjust it like we do with the Python shebang; who knows?

jefferyto, 2020-04-24:
It looks like `ccache_cc` / `ccache_cxx` are recorded in two files (_sysconfigdata.py and config/Makefile), we can do a search-and-replace during the build.

jefferyto, 2020-05-10:
I think this issue is fixed :joy:

closed 2020-05-13 by neheb