openwrt/packages #11913: <transmission-daemon-openssl>: daemon does not start - user.err : jail: failed to load libpreload-seccomp.so

glucini, 2020-04-21:
Maintainer: Rosen Penev <rosenp@gmail.com>
Environment: (ath79, Archer C7 v2, SNAPSHOT r13024)

Description:
Tranmission daemon does not start.

user.err : jail: failed to load libpreload-seccomp.so
daemon.err transmission-daemon[10149]: jail: failed to load libpreload-seccomp.so
daemon.info procd: Instance transmission::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

After 6 tries it stops.

neheb, 2020-04-22:
ping @dangowrt 

I hate this seccomp support. Nothing but problems.

glucini, 2020-04-23:
Do you have any alternative? Remove?

glucini, 2020-04-24:
I have change the last part of /etc/init.d/transmission .. and now it works. However I am not fully sure about what I have done...

	procd_open_instance
	procd_set_param command "$PROG"
	procd_append_param command -f -g "$config_dir"
	procd_set_param user "$user"
	procd_set_param group "$group"
	procd_set_param stderr 1
	procd_set_param respawn

	[ -d "$web_home" ] && procd_set_param env TRANSMISSION_WEB_HOME="$web_home"

	procd_close_instance
}

start_service() {
	config_load 'transmission'
	config_foreach transmission 'transmission'
}

reload_service() {
	procd_send_signal "$PROG"
}

service_triggers() {
	procd_add_reload_trigger "transmission"
}

neheb, 2020-04-24:
looks about right. Simply removing procd_set_param seccomp should be all that's needed.

glucini, 2020-04-24:
Actually, a new error was appearing then, reg. ujail. I was removing a bit more..and now it works.
The error was:
daemon.err procd: unable to find /sbin/ujail: No such file or directory (-1)

Most probably it refers to a procd dependency that is not there.

I think that the init script should be rechecked.

neheb, 2020-04-24:
Works fine here.

It's related to the seccomp support.

There's a package: procd-ujail that you probably need to install based on your firmware. Actually probably procd-seccomp.

Are you using stock OpenWrt or a third party firmware?

glucini, 2020-04-24:
I am using a self-recompiled version of Openwrt (with few optimizations for mvebu target). 19.07.2.

Just checking.. I do not see it among the dependencies (take a look below), moreover, why we do need it if it works well without? (sorry for the silly question, just to know).
I am also checking the official release, I do not find them also there.

Symbol: PACKAGE_transmission-daemon-openssl [=m]                                                                                                        
PACKAGE_zlib [=m] && PACKAGE_librt [=y] && PACKAGE_librt [=y] && PACKAGE_libevent2 [=m] && PACKAGE_libminiupnpc [=m] && \                
PACKAGE_libc [=y] && PACKAGE_libssp [=n] && PACKAGE_libcurl [=m] && PACKAGE_libnatpmp [=m] && PACKAGE_libpthread [=y] && PACKAGE_libopenssl [=m] && \   
PACKAGE_libpthread [=y]

neheb, 2020-04-24:
It's not there. It shouldn't be. Something in your build is implicitly pulling it in. No idea what.

Anyway, just install procd-seccomp and you should be good. No changes needed.

glucini, 2020-04-24:
I am not changing any standard parameter regarding seccomp while compiling. I actually found procd-ujail and procd-seccomp which were originally excluded from my compile. However I am doing nothing to pull them in.
The only parameter could be (however I was not touching it vs. standard): 
Symbol: KERNEL_SECCOMP [=y]                                                                                                                             
Type  : boolean                                                                                                                                       
Prompt: Enable seccomp support                                                                                                                          
Location:                                                                                                                                           │  
Global build settings                                                                                                                          │  
(3)   -> Kernel build options  
I was also taking a look on forums.. it seems that I am not the only one having this issue.
In any case. Thanks! I am just willing to understand how it works.

neheb, 2020-04-24:
@dangowrt needs to fix this I think. As I said previously, I hate this seccomp support. I had nothing to do with it.

dangowrt, 2020-04-24:
So to understand the facts in between the rants, do I get right that the problem here is that despite 'require_jail' being unset, transmission fails to start on a system which has procd-ujail installed but doesn't have procd-seccomp installed?
I suggest that it could be solved by passing the 'required_jail' attribute down to ujail to indicate that failure to find libpreload-seccomp.so should be not treated as fatal?
Or just make procd-ujail depend on procd-seccomp if seccomp is enabled in kernel?

glucini, 2020-04-25:
My understanding of the situation is that both procd-seccomp and procd-ujail are not automatically pulled in a standard compile, while they are actually needed, at least to run the default init script of transmission daemon.
2 possible alternative solutions:
a) Modify the default init script of transmission-daemon (as actually I did) to avoid needing them. It works. Implications are unclear to me.
b) Make procd-ujail and procd-seccomp be automatically included while selecting a package that needs those. Still I have not tested if by installing them the original init scripts works or not.
I would actually opt for a), however I am not expert enough to derive conclusions here.
For sure, as it is, it does not work "out of the box" and requires a bit of reverse eng to understand what does not work.
Thanks

closed 2020-04-25 by jow-