openwrt/packages #11916: UQMI won’t connect with Private APN

suyuan168, 2020-04-21:
UQMI won’t connect with Private APN
Hey Folks,

I have a Problem with the QMI protocol.

I bought a private ip from my ISP and can't connect with the private APN, only the dynamic IP APN works.

With ModemManager the connection to my Private APN works as expected, so no fault on ISP side.

I googled a little bit and found out that I'm not the only one having problems to connect to Private APN's and everyone who has problems with it, has an APN with two "."(dots) in the APN.

Is that a bug?
How to troubleshoot this?
Please let me know which logs or output from CLI you need.

However I only tried connecting through LUCI because I can't figure out how to do it through CLI.

Thanks in common

JesseYang2020, 2020-05-27:
I have the same problem too!

feckert, 2020-05-27:
This is related https://github.com/openwrt/packages/issues/11916

feckert, 2024-02-12:
Quite old

closed 2024-02-12 by feckert