openwrt/packages #11917: MWAN3: Multi-wan routing configuration problem.

zdf, 2020-04-22:
Maintainer: @feckert 
Environment: OpenWrt SNAPSHOT, r12688-46d0ce19f1 mips GNU/Linux
Description: mwan3: Routing problem
background: wan + wan2 configuration.
wan2 (table 2) lacks incoming -> lan routing like the one in wan, causing devices unable to access the internet behind the router.

feckert, 2020-04-22:
I can't help you with the little information you give me. So far I don't have such problems in my setup. Have you looked at the [wiki](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3) I think this is a miss configuration.

zdf, 2020-04-22:
Hi, the router was always migrated from previous configuration. I didn't have this problem with the last stable.

Please have a look, table 2 lacks 192.168 route. Not completely sure if it is a bug or misconfiguration, but I never changed the configuration.

# ip ro show table 1
default via 14.104.168.1 dev pppoe-wan metric 10 
x.x.x.x dev pppoe-wan2 proto kernel scope link src x.x.x.x
x.x.x.x dev pppoe-wan proto kernel scope link src x.x.x.x
192.168.x.x/24 dev br-lan proto kernel scope link src 192.168.x.x
# ip ro show table 2
default via x.x.x.x dev pppoe-wan metric 10

zdf, 2020-04-22:
Hi, the router was always migrated from previous configuration. I didn't have this problem with the last stable.

Please have a look, table 2 lacks 192.168 route. Not completely sure if it is a bug or misconfiguration, but I never changed the configuration.

ip ro show table 1
default via x.x.x.x dev pppoe-wan metric 10
x.x.x.x dev pppoe-wan2 proto kernel scope link src x.x.x.x
x.x.x.x dev pppoe-wan proto kernel scope link src x.x.x.x
192.168.x.x/24 dev br-lan proto kernel scope link src 192.168.x.x

ip ro show table 2
default via x.x.x.x dev pppoe-wan metric 10

feckert, 2020-04-22:
* Does you have a process called mwan3rtmon?
* Is this pppoe interface on the same ethernet?
* Does there are mwan3track process for this inerfaces
* Do you have installed the ip-full?

zdf, 2020-04-22:
> Does you have a process called mwan3rtmon?
24136 root      1408 S    /bin/sh /usr/sbin/mwan3rtmon

> Is this pppoe interface on the same ethernet?
Physically yes, 2 macvlans.

> Does there are mwan3track process for this inerfaces
24005 root      1328 S    /bin/sh /usr/sbin/mwan3track wan pppoe-wan online xxxxxxxxx
24489 root      1328 S    /bin/sh /usr/sbin/mwan3track wan2 pppoe-wan2 online xxxxxxxx

> Do you have installed the ip-full?
No... I installed ip-full and reapplied mwan3, the aforementioned route I've added to table 2 disappeared: problem persists.

feckert, 2020-05-29:
I miss the following lines in `ip route table 2`
```
x.x.x.x dev pppoe-wan2 proto kernel scope link src x.x.x.x
x.x.x.x dev pppoe-wan proto kernel scope link src x.x.x.x
192.168.x.x/24 dev br-lan proto kernel scope link src 192.168.x.x
```

aaronjg, 2020-07-13:
Do you have this problem on base 19.07.3 and the latest `mwan3` from the openwrt/packages repo?

If the problem persists,  can you run `mwan3 restart` from the command line and post the output? It looks like the process may be crashing partway through.

zdf, 2020-07-26:
@aaronjg The snapshot now works fine if you don't keep the configurations.

CsHeng, 2020-07-30:
by checking the mwan3 source file:

https://github.com/openwrt/packages/blob/facf8ea299b185a82ae803d68e2b54a9599d8c14/net/mwan3/files/lib/mwan3/mwan3.sh

**i found `$IP4 route add table $tid $line` will only add the first line in `ip route list table main` with same subnet/proto kernel**

related mwan3 source code below
```bash
# return true(=0) if has any mwan3 interface enabled
# otherwise return false
mwan3_rtmon_ipv4()
{
	local tid=1
	local idx=0
	local ret=1
	local tbl=""
	mkdir -p /tmp/mwan3rtmon
	($IP4 route list table main  | grep -v "^default\|linkdown" | sort -n; echo empty fixup) >/tmp/mwan3rtmon/ipv4.main
	while uci get mwan3.@interface[$idx] >/dev/null 2>&1 ; do
		idx=$((idx+1))
		tid=$idx
		[ "$(uci get mwan3.@interface[$((idx-1))].family)" = "ipv4" ] && {
			tbl=$($IP4 route list table $tid 2>/dev/null)
			if echo "$tbl" | grep -q ^default; then
				(echo "$tbl"  | grep -v "^default\|linkdown" | sort -n; echo empty fixup) >/tmp/mwan3rtmon/ipv4.$tid
				cat /tmp/mwan3rtmon/ipv4.$tid | grep -v -x -F -f /tmp/mwan3rtmon/ipv4.main | while read line; do
					$IP4 route del table $tid $line
				done
				cat /tmp/mwan3rtmon/ipv4.main | grep -v -x -F -f /tmp/mwan3rtmon/ipv4.$tid | while read line; do
					$IP4 route add table $tid $line
				done
			fi
		}
		if [ "$(uci get mwan3.@interface[$((idx-1))].enabled)" = "1" ]; then
			ret=0
		fi
	done
	rm -f /tmp/mwan3rtmon/ipv4.*
	return $ret
}
```

so, assume that I have 2 pppoe line, and they share the same gateway, it will only add the first one to table 1/2

my real sample:

```bash
root@OpenWrt-GW:~# ip route list table main
default via 113.119.164.1 dev pppoe-adsl1 proto static metric 1
default via 113.119.164.1 dev pppoe-adsl2 proto static metric 2
113.119.164.1 dev pppoe-adsl1 proto kernel scope link src xx.xx.xx.xx
113.119.164.1 dev pppoe-adsl2 proto kernel scope link src xx.xx.xx.xx
192.168.0.0/21 dev br-lan proto kernel scope link src 192.168.2.101
192.168.255.0/24 dev bonding-bond0 proto kernel scope link src 192.168.255.1

root@OpenWrt-GW:~# ip route list table 1
default via 113.119.164.1 dev pppoe-adsl1 metric 1
113.119.164.1 dev pppoe-adsl1 proto kernel scope link src xx.xx.xx.xx
192.168.0.0/21 dev br-lan proto kernel scope link src 192.168.2.101
192.168.255.0/24 dev bonding-bond0 proto kernel scope link src 192.168.255.1

root@OpenWrt-GW:~# ip route list table 2
default via 113.119.164.1 dev pppoe-adsl2 metric 2
113.119.164.1 dev pppoe-adsl1 proto kernel scope link src xx.xx.xx.xx
192.168.0.0/21 dev br-lan proto kernel scope link src 192.168.2.101
192.168.255.0/24 dev bonding-bond0 proto kernel scope link src 192.168.255.1

```

table 1 missing pppoe-adsl2 route

and table 2 missing pppoe-adsl1 route

if I manually add the missing route `ip route add table 1 113.119.164.1 dev pppoe-adsl2 proto kernel scope link src xx.xx.xx.xx`, an error will occur: `RTNETLINK answers: File exists`

**but... should this be `iproute`'s feature or just by design (that we can't add route with same subnet which proto is kernel link) ?**

**will this situation cause any bug or we could just ignore it..**

(
actually I have bond0 ipaddress set to 192.168.2.1 earlier, and when I start mwan3, I f**ked up. (i have luci-proto-bonding to setup bond)
192.168.0.0/21 dev br-lan proto kernel scope link src 192.168.2.101
192.168.0.0/21 dev bonding-bond0 proto kernel scope link src 192.168.2.1
)

please help.  @feckert 

btw, thanks for writing such good scripts !!!

feckert, 2020-07-30:
Please try the newest mwan3 version in master or in openwrt-19.07(stable) branch.
There have been some improvements.

If you still have the problem please open a new issue.
This was already solved, so I will close them.

closed 2020-07-30 by feckert

aaronjg, 2020-07-31:
> but... should this be iproute's feature or just by design (that we can't add route with same subnet which proto is kernel link) ?

Yes, this is by design, you can't add two routes to the same destination with the same metric. How would the kernel know which one to use?

> will this situation cause any bug or we could just ignore it..
Potentially, yes.

Traffic to 113.119.164.1 that  mwan3 wants to route out of pppoe-adsl2 will actually go out of pppoe-adsl1 instead.
 
I am working on a somewhat extensive refactoring of mwan3 for some speed improvements, and I address this. Code should be ready in a week or two.