openwrt/packages #11920: python3-pip non-functional without extra tweaks

brada4, 2020-04-22:
# Issue template (remove lines from top till here)

Maintainer: @jefferyto (find it by checking history of the package Makefile)
snapshot today
19.07.2
ath79
x86_64

Description:

Following command fails:
```
python -m pip install --upgrade pip
```
It logs something about missing 517 files or so
Reading local help text turns out pep517 scripts are missing, forcing above command to legacy mode by parameter:
```
--no-use-pep517
```
yields working pip installation that does not complain in log about missing scripts.

The request is to update packaging lists of python3-pip to include pep517 library files, so that package is usable without excess tweaks again

jefferyto, 2020-04-22:
Are you trying to upgrade pip3 (using python3) or pip (with python 2)? `pip3 install --upgrade pip` worked for me with no errors.

brada4, 2020-04-22:
I configured PIP with `--no-binary :all:`
(/root/.config/pip/pip)
Python is python3 for certain, no trails of python 2.7

jefferyto, 2020-04-22:
Can you copy-and-paste or attach a file with the full error message (and pip log if there is one)?

brada4, 2020-04-22:
Tomorrow, like in 10h.

brada4, 2020-04-22:
[pip-log.txt](https://github.com/openwrt/packages/files/4519232/pip-log.txt)

So it is snapshot in a chroot on 19.07.2 on a Archer C7 v5 router. It supplies my internet pipes, so I would not jump on snapshot for that. Real target added for added sense of realism, same on x86_64 with less waiting.

What I am trying:
1) reinstall same pip without using binary wheels - fails
2) try harder with pep517 validation disabled - succeeds
3) now try 1st command again - succeeds

Conclusion: files named in output (same line in log between generic checksum validation and mirror crawling messages) 

(Preparation: I make chroot from build over git checkout, i.e tar it from build dir, then untar on a router usb drive, bind-mount /dev and /proc, then chroot with ash)

`root@OpenWrt:/# opkg update`
```
Downloading http://downloads.openwrt.org/snapshots/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/snapshots/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
```
`root@OpenWrt:/# opkg install python3-pip`
```
lots of packages
```
`root@OpenWrt:/# python3 -m pip --version`
```
pip 19.2.3 from /usr/lib/python3.8/site-packages/pip (python 3.8)
```
`root@OpenWrt:/# python3 -m pip install --no-binary :all:  --force-reinstall pip==19.2.3`
```
Collecting pip==19.2.3
  Downloading https://files.pythonhosted.org/packages/00/9e/4c83a0950d8bdec0b4ca72afd2f9cea92d08eb7c1a768363f2ea458d08b4/pip-19.2.3.tar.gz (1.4MB)
     |ASCII-ART-THAT-DO-NOT-PASTE| 1.4MB 1.0MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
ERROR: Command errored out with exit status 2: /usr/bin/python3 /usr/lib/python3.8/site-packages/pip/_vendor/pep517/_
in_process.py get_requires_for_build_wheel /tmp/tmpyhklgffw Check the logs for full command output.
WARNING: You are using pip version 19.2.3, however version 20.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
`root@OpenWrt:/# python3 -m pip install --no-binary :all: --no-use-pep517 --force-reinstall`
```
pip==19.2.3
Collecting pip==19.2.3
  Using cached https://files.pythonhosted.org/packages/00/9e/4c83a0950d8bdec0b4ca72afd2f9cea92d08eb7c1a768363f2ea458d08b4/pip-19.2.3.tar.gz
Installing collected packages: pip
  Found existing installation: pip 19.2.3
    Uninstalling pip-19.2.3:
      Successfully uninstalled pip-19.2.3
  Running setup.py install for pip ... done
Successfully installed pip-19.2.3
```
`root@OpenWrt:/# python3 -m pip install --no-binary :all:  --force-reinstall pip==19.2.3`
```
Collecting pip==19.2.3
  Using cached https://files.pythonhosted.org/packages/00/9e/4c83a0950d8bdec0b4ca72afd2f9cea92d08eb7c1a768363f2ea458d08b4/pip-19.2.3.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Skipping bdist_wheel for pip, due to binaries being disabled for it.
Installing collected packages: pip
  Found existing installation: pip 19.2.3
    Uninstalling pip-19.2.3:
      Successfully uninstalled pip-19.2.3
  Running setup.py install for pip ... done
Successfully installed pip-19.2.3
```
`root@OpenWrt:/# cat /etc/openwrt_*`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r13044-3773ae127a'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r13044-3773ae127a'
DISTRIB_TAINTS='no-all'
r13044-3773ae127a
```
`@root@OpenWrt:/# uname -a`
```
Linux OpenWrt 4.14.171 #0 Thu Feb 27 21:05:12 2020 mips GNU/Linux
```
`root@OpenWrt:/mnt/sda# cat /etc/openwrt_*`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.2'
DISTRIB_REVISION='r10947-65030d81f3'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.2 r10947-65030d81f3'
DISTRIB_TAINTS=''
r10947-65030d81f3
```

brada4, 2020-04-23:
I am trying to compile package locally instead of downloading a pre-packaged wheel from the internet.

brada4, 2020-04-23:
Somehow my pyc files are different (?always bigger?) than those distributed via opkg.

```
root@OpenWrt:/mnt/sda# ls -l py3/usr/lib/python3.8/site-packages/pip/_vendor/pep517/
-rw-r--r--    1 root     root           195 Apr 19 22:19 __init__.pyc
-rw-r--r--    1 root     root          5453 Apr 19 22:19 _in_process.pyc
-rw-r--r--    1 root     root          2709 Apr 19 22:19 build.pyc
-rw-r--r--    1 root     root          4678 Apr 19 22:19 check.pyc
-rw-r--r--    1 root     root          2865 Apr 19 22:19 colorlog.pyc
-rw-r--r--    1 root     root           917 Apr 19 22:19 compat.pyc
-rw-r--r--    1 root     root          4197 Apr 19 22:19 envbuild.pyc
-rw-r--r--    1 root     root          5457 Apr 19 22:19 wrappers.pyc
root@OpenWrt:/mnt/sda# ls -l opkg-py/usr/lib/python3.8/site-packages/pip/_vendor/pep517/
-rw-r--r--    1 root     root            84 Jul 30  2019 __init__.py
drwxr-xr-x    2 root     root          3488 Apr 22 21:53 __pycache__
-rw-r--r--    1 root     root          6061 Jul 30  2019 _in_process.py
-rw-r--r--    1 root     root          2699 Jul 30  2019 build.py
-rw-r--r--    1 root     root          5885 Jul 30  2019 check.py
-rw-r--r--    1 root     root          4098 Jul 30  2019 colorlog.py
-rw-r--r--    1 root     root           631 Jul 30  2019 compat.py
-rw-r--r--    1 root     root          5763 Jul 30  2019 envbuild.py
-rw-r--r--    1 root     root          5912 Jul 30  2019 wrappers.py
root@OpenWrt:/mnt/sda# ls -l py3/usr/lib/python3.8/site-packages/pip/_vendor/pep517/__pycache__/
-rw-r--r--    1 root     root           246 Apr 22 21:53 __init__.cpython-38.pyc
-rw-r--r--    1 root     root          5504 Apr 22 21:53 _in_process.cpython-38.pyc
-rw-r--r--    1 root     root          2760 Apr 22 21:53 build.cpython-38.pyc
-rw-r--r--    1 root     root          4729 Apr 22 21:53 check.cpython-38.pyc
-rw-r--r--    1 root     root          2916 Apr 22 21:53 colorlog.cpython-38.pyc
-rw-r--r--    1 root     root           968 Apr 22 21:53 compat.cpython-38.pyc
-rw-r--r--    1 root     root          4248 Apr 22 21:53 envbuild.cpython-38.pyc
-rw-r--r--    1 root     root          5508 Apr 22 21:53 wrappers.cpython-38.pyc
```

jefferyto, 2020-04-23:
I haven't replicated your error yet but I think your issue is the same pip issue described in #11912.

brada4, 2020-04-24:
I suspect the problem is that packaged pyc file is made in different version of python and gets rejected,

jefferyto, 2020-05-10:
Please re-test to see if your issue has been fixed.

brada4, 2020-05-10:
Thanks.
I confirm that problem with python3-pip disappears after package upgrade from 19.2.3-2 to 19.2.3-3

closed 2020-05-10 by brada4