openwrt/packages #11950: mwan3: Diagonistics page doesn't work with LuCI running through nginx

jamesmacwhite, 2020-04-24:
Maintainer: @feckert 
Environment: 19.07.2 Linksys WRT3200ACM

Description:

I run LuCI through nginx/nginx-ssl and I get the nginx 504 gateway timeout error page served when using the diagnostics page as below.

[image]

Not sure if this is a bug with LuCI or mwan3 though.

lucize, 2020-04-25:
for me with latest in chrome, all is working (ipv4)

jamesmacwhite, 2020-04-26:
@lucize Weird. Neither IPv4/IPv6 working for me, are you also running LuCI with nginx?

lucize, 2020-04-26:
it seems that executing only ping for tracking ip will not show you the results, but also using ping the default gateway (or the same command twice) will give you all the results

jamesmacwhite, 2020-04-26:
@lucize Indeed it does, clicking execute twice works. Potentially a bug then?

dl12345, 2020-04-29:
I use luci with nginx on 19.07.2 and pinging the tracking ip works just fine for me even first time around

jamesmacwhite, 2020-04-29:
@dl12345 Interesting. Not sure what could be different to cause that.

peter-stadler, 2020-05-03:
Does it help setting `threads = 1` in `/etc/uwsgi/vassals/luci-webui.ini` (cf. #10134)?

jamesmacwhite, 2020-05-04:
@peter-stadler Thanks for the suggestion. Setting threads to 1 and restarting nginx does seem to make the diagnostics page more reliable and behave normally.

feckert, 2020-05-04:
@peter-stadler thanks for the help. Could we add this into the Wiki?
I think this is not only a luci-app-mwan3 problem.

closed 2020-05-04 by feckert

jamesmacwhite, 2020-05-04:
@feckert My apologies. I wasn't entirely sure if this was a mwan3 issue. I do remember having issues with other LuCI areas, mainly the firmware sys upgrade process, so I should have looked into the nginx more closely.

peter-stadler, 2020-05-04:
I will make a PR until tomorrow …