openwrt/packages #11957: unbound: README.md and domain in parallel mode

AndreaBorgia-Abo, 2020-04-25:
Maintainer: Eric Luehrsen <ericluehrsen@gmail.com>
Environment: Fritzbox 4040, OpenWRT 19.07.2

Description: 
As the README.md is now, when configuring in parallel mode there is no suggestion to add a "domain" keyword, which is present for the serial mode. The startup script will generate a "search" line for resolv.conf using the default domain "lan". When the local domain name has been changed, this breaks shortname lookups on the router itself. Assuming there are no sideeffects, could this recommendation be added? Thanks.

EricLuehrsen, 2020-05-15:
In serial mode, it is important that Unbound block your local domain from external resolution. It is what dnsmasq should do anyway so if leaking, Unbound needs to block. In parallel mode, there can be (many) alternate configurations. dnsmasq locally is one authoritative-ish server, but easily other routers serving other subnets can be added as stub or forward zones. Someone may not want short names on the router. Example all routers are named "router" under a subdomain such as "router.subnet1.example.net." Because this affects only the local router and not clients (dnsmasq pushes DHCP search="subnet1.example.net."), it may be trivial to most administrators. Addressing this one use case opens the can of worms to related use cases. A simple README becomes cluttered. Complex topics might be best on OpenWrt wiki where they can be presented with a topology map inline.

AndreaBorgia-Abo, 2020-05-15:
Let me try another angle: take a default openwrt installation with a default "lan" local domain name. Now install Unbound in parallel mode: shortname lookups wil work. Why? The startup script silently adds a default "lan" domain.

Openly recommending a setup with a "domain" keyword for parallel makes this behaviour evident, instead of hiding it in the startup script. I am not asking for a change in the way unbound behaves, just making it more visible and consistent.

PR #12400 (2020-06-04): unbound: suggest matched domain option for dnsmasq link
Maintainer: me
Description: README only change. This should resolve #11957
closed 2020-06-04 by neheb