openwrt/packages #11960: less: clash between 'busybox/less' and 'less-wide'

BlueMax, 2020-04-26:
@Zokormazo @BKPepe

x86, master
Related to: https://github.com/openwrt/packages/commit/0ae161ecb6928f87103921237ce28a54cdfd6061

```
 * check_data_file_clashes: Package less-wide wants to install file openwrt/build_dir/target-x86_64_musl/root-x86/usr/bin/less
    But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package less-wide.
```

BKPepe, 2020-04-26:
For me, it seems you are using an old version of OpenWrt master. Because the same error was reported to me, I was able to reproduce it and fixed it.  Would you mind to share a full output? There's missing details like which file you were downloading. I'm interested in `PKG_RELEASE`. 

My output based on my changes:
```
root@turris:~# ls -la /bin | grep less
lrwxrwxrwx    1 root     root            12 Apr 26 11:05 less -> /bin/busybox
root@turris:~# opkg install less-wide
Installing less-wide (530-2.0) to root...
Downloading packages/less-wide_530-2_powerpc_8540.ipk
Configuring less-wide.
root@turris:~# ls -la /bin | grep less
lrwxrwxrwx    1 root     root            13 Apr 26 11:05 less -> /usr/bin/less
root@turris:~# opkg remove less-wide
Removing package less-wide from root...
root@turris:~# ls -la /bin | grep less
lrwxrwxrwx    1 root     root            12 Apr 26 11:05 less -> /bin/busybox
root@turris:~# opkg install less
Installing less (530-2.0) to root...
Downloading packages/less_530-2_powerpc_8540.ipk
Configuring less.
root@turris:~# ls -la /bin | grep less
lrwxrwxrwx    1 root     root            13 Apr 26 11:05 less -> /usr/bin/less
root@turris:~# opkg remove less
Removing package less from root...
root@turris:~# ls -la /bin | grep less
lrwxrwxrwx    1 root     root            12 Apr 26 11:05 less -> /bin/busybox
```

When I reverted my changes locally, I am able reproduce your issue.

Output and notice there's `PKG_RELEASE` 2→1 after `PKG_VERSION`:
```
root@turris:/tmp# ls -la | grep less
-rw-r--r--    1 root     root         66235 Apr 26 11:14 less-wide_530-1_powerpc_8540.ipk
-rw-r--r--    1 root     root         66203 Apr 26 11:14 less_530-1_powerpc_8540.ipk
root@turris:/tmp# opkg install less_530-1_powerpc_8540.ipk 
Installing less (530-1) to root...
Collected errors:
 * check_data_file_clashes: Package less wants to install file /bin/less
        But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package less.
root@turris:/tmp# opkg install less-wide_530-1_powerpc_8540.ipk 
Installing less-wide (530-1) to root...
Collected errors:
 * check_data_file_clashes: Package less-wide wants to install file /bin/less
        But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package less-wide.
```

BlueMax, 2020-04-26:
Sorry for the confusion.
I'm not talking about opkg installation but compiling from scratch. And yes, i'm on latest master/feeds (and 'make dirclean').

BKPepe, 2020-04-28:
In this case, you need to provide steps to reproduce your issue. I am not able to reproduce it on my end. I tried to compile it locally and even ran multiple full builds.

BlueMax, 2020-04-28:
No problem. Use this:
```
git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
./scripts/feeds update -a
./scripts/feeds install less-wide
./scripts/feeds install libpam libgnutls libopenldap libidn2 libssh2 liblzma libnetsnmp
make menuconfig
  * Target System (x86)
  * Subtarget (x86_64)
  * Target Profile (Generic x86/64)
  * Utilities --> [*] less-wide
make defconfig
make -j4
```

guidosarducci, 2020-05-13:
@BKPepe Hi Josef, could you please look again at this issue, your commit https://github.com/openwrt/packages/commit/0ae161ecb6928f87103921237ce28a54cdfd6061, and any related testing you may have done? There's definitely an issue with installing **less** after **busybox**.

I ran into this same problem yesterday multiple times using 19.07.2 on ar71xx and ipq40xx, whether using an ImageBuilder or freshly-installed live router system.

For example (note the less package version with your latest changes):
```
BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07.2, r10947-65030d81f3
 -----------------------------------------------------
root@EA6350v3:/tmp# opkg install less
Installing less (530-2) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/arm_cortex-a7_neon-vfpv4/packages/less_530-2_arm_cortex-a7_neon-vfpv4.ipk
Collected errors:
 * check_data_file_clashes: Package less wants to install file /usr/bin/less
        But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package less.
root@EA6350v3:/tmp#
```

zhanhb, 2020-05-16:
Here are two conflilcts package.
Conflict kmod with ubox.
Conflict logger with busybox.

BKPepe, 2020-05-16:
@guidosarducci Sure, I have it on my to-do list. Your output helped me and I think I know what's wrong. Thanks!

@zhanhb Please create a different issue. This is not related to this issue anyhow.

commit d8663931 (2020-06-10): less: update ALTERNATIVES

Avoid conflict with busybox
Closes #11960
closed 2020-06-10 by neheb