openwrt/packages #11961: ksmbd: Memory leak detected by kmemleak

jmarcet, 2020-04-26:
Maintainer: Andy Walsh <andy.walsh44+github@gmail.com>
Environment: master x86-64

Description:

While searching for a different issue, these were detected by kmemleak:

```
unreferenced object 0xffff8883fd413c40 (size 64):
  comm "kworker/6:0", pid 4324, jiffies 4350752981 (age 16144.250s)
  hex dump (first 32 bytes):
    00 00 00 00 04 00 00 00 00 f5 ce bb 83 88 ff ff  ................
    40 00 41 fd 83 88 ff ff a0 b2 57 e9 81 88 ff ff  @.A.......W.....
  backtrace:
    [<0000000077de2a30>] __kmalloc+0x121/0x270
    [<0000000062b9a96e>] ksmbd_alloc+0x1c/0x60 [ksmbd]
    [<00000000aa6d14b1>] ksmbd_tree_conn_connect+0x32/0x170 [ksmbd]
    [<0000000062439ba7>] smb2_tree_connect+0x63/0x1e0 [ksmbd]
    [<00000000a6a1ed85>] ksmbd_ipc_init+0x468/0x7b0 [ksmbd]
    [<0000000026f1a009>] process_one_work+0x18a/0x2e0
    [<000000005f9a8aec>] worker_thread+0x4b/0x3d0
    [<000000000c4b85f1>] kthread+0xf5/0x130
    [<00000000ce381d31>] ret_from_fork+0x35/0x40
unreferenced object 0xffff8883cbccee40 (size 64):
  comm "kworker/1:2", pid 1291, jiffies 4351216071 (age 14600.620s)
  hex dump (first 32 bytes):
    00 00 00 00 04 00 00 00 00 70 d0 e4 83 88 ff ff  .........p......
    40 fe cc cb 83 88 ff ff a0 0e 8a 8b 80 88 ff ff  @...............
  backtrace:
    [<0000000077de2a30>] __kmalloc+0x121/0x270
    [<0000000062b9a96e>] ksmbd_alloc+0x1c/0x60 [ksmbd]
    [<00000000aa6d14b1>] ksmbd_tree_conn_connect+0x32/0x170 [ksmbd]
    [<0000000062439ba7>] smb2_tree_connect+0x63/0x1e0 [ksmbd]                                                                                  
    [<00000000a6a1ed85>] ksmbd_ipc_init+0x468/0x7b0 [ksmbd]
    [<0000000026f1a009>] process_one_work+0x18a/0x2e0
    [<000000005f9a8aec>] worker_thread+0x4b/0x3d0
    [<000000000c4b85f1>] kthread+0xf5/0x130
    [<00000000ce381d31>] ret_from_fork+0x35/0x40
unreferenced object 0xffff8883cbccc440 (size 64):
  comm "kworker/1:2", pid 1291, jiffies 4351216078 (age 14600.597s)
  hex dump (first 32 bytes):
    01 00 00 00 05 00 00 00 80 a3 47 bc 83 88 ff ff  ..........G.....
    40 fe cc cb 83 88 ff ff 58 ee cc cb 83 88 ff ff  @.......X.......
  backtrace:
    [<0000000077de2a30>] __kmalloc+0x121/0x270                                                                                                 
    [<0000000062b9a96e>] ksmbd_alloc+0x1c/0x60 [ksmbd]
    [<00000000aa6d14b1>] ksmbd_tree_conn_connect+0x32/0x170 [ksmbd]
    [<0000000062439ba7>] smb2_tree_connect+0x63/0x1e0 [ksmbd]
    [<00000000a6a1ed85>] ksmbd_ipc_init+0x468/0x7b0 [ksmbd]
    [<0000000026f1a009>] process_one_work+0x18a/0x2e0
    [<000000005f9a8aec>] worker_thread+0x4b/0x3d0                                                                                              
    [<000000000c4b85f1>] kthread+0xf5/0x130
    [<00000000ce381d31>] ret_from_fork+0x35/0x40                                                                                               
unreferenced object 0xffff8883cbccc040 (size 64):
  comm "kworker/1:2", pid 1291, jiffies 4351216119 (age 14600.460s)                                                                            
  hex dump (first 32 bytes):
    02 00 00 00 04 00 00 00 00 f5 ce bb 83 88 ff ff  ................
    40 fe cc cb 83 88 ff ff 58 c4 cc cb 83 88 ff ff  @.......X.......
  backtrace:
    [<0000000077de2a30>] __kmalloc+0x121/0x270
    [<0000000062b9a96e>] ksmbd_alloc+0x1c/0x60 [ksmbd]
    [<00000000aa6d14b1>] ksmbd_tree_conn_connect+0x32/0x170 [ksmbd]
    [<0000000062439ba7>] smb2_tree_connect+0x63/0x1e0 [ksmbd]
    [<00000000a6a1ed85>] ksmbd_ipc_init+0x468/0x7b0 [ksmbd]
    [<0000000026f1a009>] process_one_work+0x18a/0x2e0
    [<000000005f9a8aec>] worker_thread+0x4b/0x3d0
    [<000000000c4b85f1>] kthread+0xf5/0x130
    [<00000000ce381d31>] ret_from_fork+0x35/0x40
```

hnyman, 2020-04-26:
cc @jmarcet 
Shouldn't you report that to ksmbd upstream ???

cc @Andy2244 
FYI

Andy2244, 2020-04-26:
yeah i'm just a maintainer, i don't fix code stuff.

Please open the issue here: https://github.com/cifsd-team/cifsd/issues

jmarcet, 2020-04-26:
Understood, I'll file it there.

closed 2020-04-26 by jmarcet