openwrt/packages #11962: ddns-scripts: unsecure http urls

bernhard-efler, 2020-04-26:
I have noticed that most ddns update urls are using unsecure http urls.

Before I go through all and check if an https version is available I want to know if there is any reason against it?
And will a pull requests be merged?

hnyman, 2020-04-26:
> if there is any reason against it?

The main reason is that by default OpenWrt does not have https support in order to keep the image size small for the old routers with really limited flash sizes.

For the same reason also the opkg package downloads from the router are http by default.

> And will a pull requests be merged?

PR proposing to change all/most DDNS providers to be https-only would not be received well.

bernhard-efler, 2020-04-26:
http only is not a good think because often there is some password/key required for ddns and that is transmitted unencrypted.
so the only good solution is to make https an config option which can be set by the user?

hnyman, 2020-04-26:
You can naturally edit the definition file on your router by yourself and change the sites that you use, to use https if the site supports that. 
It is just a normal text file.

bernhard-efler, 2020-05-03:
After installing libustream-openssl a secure option is displayed, so it is supported.
The only improvment I see is to display a notice to install SSL and do not hide the option complete.

closed 2020-05-03 by bernhard-efler