openwrt/packages #11965: mwan3: 2.8.3-2 /usr/sbin/mwan3track: line 139:  : not found

jamesmacwhite, 2020-04-26:
Maintainer: @feckert 
Environment: OpenWrt 19.07.2

Description:

I upgraded to the latest master version due to issues with ping and IPv6 interfaces, however I think this commit 6721587e8b491f1a539cdcc507b0f4811ff2b448 is causing `/usr/sbin/mwan3track: line 139:  : not found` errors to be spammed in terminal every few seconds (the check interval) when mwan3 is enabled. It doesn't happen on 2.8.1.

jamesmacwhite, 2020-04-27:
@brianjmurrell Just also pinging you, as it seems to be related to the recent IPv6 ping changes. Although, I understand the reason for the changes because ping with IPv6 interfaces isn't working well currently.

I can post my network interfaces/config if it helps.

feckert, 2020-04-27:
Maybe we have a problem with the tools (busybox vs. linux-tools)?
@jamesmacwhite what version of **ip** and **sed** habe you installed on your system?
I am using the following versions in the master branch
* sed from busybox (1.31.1)
* ip from ip-full (5.5.0)

jamesmacwhite, 2020-04-27:
Thanks.

My sed version is from BusyBox v1.30.1 and I do also have ip-full installed (5.0.0-2.1)

feckert, 2020-04-27:
Thanks, I think the PATH variable is not correct when calling the script?

jamesmacwhite, 2020-04-27:
It's not potentially a sh vs bash thing is it? Is use of `:-` safe with sh?

brianjmurrell, 2020-04-27:
```
# /bin/bash
# unset b
# a=${b:-notset}
# echo $a
notset
# exit
exit
# /bin/sh

BusyBox v1.30.1 () built-in shell (ash)

# unset b
# a=${b:-notset}
# echo $a
notset
```

Seems to work fine with `/bin/sh`

brianjmurrell, 2020-04-27:
@jamesmacwhite what does `/bin/sh -c 'unset b; a=${b:-notset}; echo $a'` report on your system?

jamesmacwhite, 2020-04-27:
No worries, just a thought, I'm struggling to understand why not found gets thrown, when the command works fine outside of mwan3track.

I get `notset` reported when running the above.

brianjmurrell, 2020-04-27:
If you put:

```bash
set -x
exec 2>/tmp/mw3track.debug
```

after line 4 of `/usr/sbin/mwan3track` that will log the _xtrace_ to `/tmp/mw3track.debug` so that you can see exactly what's causing that to be printed.

jamesmacwhite, 2020-04-27:
@brianjmurrell Thanks!

This is what's being reported.

```
+ echo 2001:4860:4860::8888
+ grep -q :
+ ip -6 addr ls dev 6in4-wan6
+ sed -ne 's/ *inet6 \([^ \/]*\).* scope global.*/\1/p'
+ ' ' 'ADDR=2001:470:xxxx:xx::2'
/usr/sbin/mwan3track: line 142:  : not found
```

jamesmacwhite, 2020-04-27:
I think I've found the issue. It's weird though. There is a weird character on the line that's being referenced on the /usr/bin/mwan3track script.

[image]

I've confirmed it's not something I've caused locally, as I extracted the contents from the ipk from recent snapshot. Something to with encoding?

jamesmacwhite, 2020-04-27:
It's present in the source code too. I didn't notice it before, but that's the reason why it's blowing up!

https://github.com/openwrt/packages/blob/6721587e8b491f1a539cdcc507b0f4811ff2b448/net/mwan3/files/usr/sbin/mwan3track#L139

closed 2020-04-27 by feckert

jamesmacwhite, 2020-04-27:
@feckert Thank you!

feckert, 2020-04-27:
@jamesmacwhite Thanks for the report

commit 3c34b151 (2020-04-27): mwan3: fix whitespace issue

fixes #11965
commit 0d795d20 (2020-04-28): mwan3: fix whitespace issue

fixes #11965

(cherry picked from commit 3c34b151f649d243126240298d4f51c7e2dd7939)