openwrt/packages #11969: irqbalance: linking against libiconv-stub only is erroneous

dl12345, 2020-04-27:
The CONFIGURE_VARS for irqbalance in the Makefile assume that the build has the package libiconv selected and not the package libiconv-full selected. 

If the libiconv-full package is selected then irqbalance will be unable to find a libiconv.a in the staging area as the libiconv-stub folder will be missing. The following patch will fix this.

This problem report applies to both master and openwrt-19.07

```
diff --git a/utils/irqbalance/Makefile b/utils/irqbalance/Makefile
index d0cc583..713448f 100644
--- a/utils/irqbalance/Makefile
+++ b/utils/irqbalance/Makefile
@@ -52,9 +52,15 @@ CONFIGURE_ARGS+= \
        --without-irqbalance-ui \
        --enable-static=glib2
 
+ifeq ($(CONFIG_PACKAGE_libiconv-full),y)
+       LIB_ICONV += "libiconv-full"
+else
+       LIB_ICONV += "libiconv-stub"
+endif
+
 CONFIGURE_VARS += \
        GLIB2_LIBS="$(STAGING_DIR)/usr/lib/libglib-2.0.a \
-               $(STAGING_DIR)/usr/lib/libiconv-stub/lib/libiconv.a"
+               $(STAGING_DIR)/usr/lib/$(LIB_ICONV)/lib/libiconv.a"
 
 define Package/irqbalance/install
        $(INSTALL_DIR) $(1)/usr/sbin```

dl12345, 2020-04-27:
or perhaps that should more correctly be

`ifneq ($(CONFIG_PACKAGE_libiconv-full),)`

dl12345, 2020-04-27:
The following fixed it correctly for me

```
iff --git a/utils/irqbalance/Makefile b/utils/irqbalance/Makefile
index d0cc583..ffa5412 100644
--- a/utils/irqbalance/Makefile
+++ b/utils/irqbalance/Makefile
@@ -28,8 +28,13 @@ include $(INCLUDE_DIR)/nls.mk
 
 include $(INCLUDE_DIR)/package.mk
 
+ifneq ($(CONFIG_PACKAGE_libiconv-full),)
+PKG_BUILD_DEPENDS:=glib2 gettext-full 
+else
+PKG_BUILD_DEPENDS:=glib2 gettext
+endif
+
 define Package/irqbalance
-  PKG_BUILD_DEPENDS:=glib2
   SECTION:=utils
   CATEGORY:=Utilities
   DEPENDS:=+libintl-full +libpthread
@@ -43,8 +48,6 @@ define Package/irqbalance/description
  increase performance.
 endef
 
 CONFIGURE_ARGS+= \
        --disable-numa \
        --with-libcap_ng=no \
@@ -54,7 +57,9 @@ CONFIGURE_ARGS+= \
 
 CONFIGURE_VARS += \
        GLIB2_LIBS="$(STAGING_DIR)/usr/lib/libglib-2.0.a \
-               $(STAGING_DIR)/usr/lib/libiconv-stub/lib/libiconv.a"
+               $(ICONV_PREFIX)/lib/libiconv.a \
+               -L$(STAGING_DIR_ROOT)/lib/libpthread -lpthread \
+               -L$(INTL_PREFIX)/lib -lintl"
 
 define Package/irqbalance/install
        $(INSTALL_DIR) $(1)/usr/sbin
```

Note that pthread cannot be linked statically while libc is linked dynamically. If trying to do so, then the following errors result

```
libc/root-x86/lib/libpthread.a(pthread_create.o): in function `__pthread_create_2_1':
pthread_create.c:(.text+0x167a): undefined reference to `_dl_stack_flags'
/opt/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/8.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: pthread_create.c:(.text+0x1f2c): undefined reference to `_dl_stack_flags'
/opt/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/8.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: /opt/openwrt/staging_dir/target-x86_64_glibc/root-x86/lib/libpthread.a(elision-lock.o): in function `_dl_tunable_set_elision_enable':
elision-lock.c:(.text+0xe2): undefined reference to `_dl_x86_cpu_features'
```

So this would not work, even if we wanted to link  statically with libpthread / libintl

```
CONFIGURE_VARS += \
    GLIB2_LIBS="$(STAGING_DIR)/usr/lib/libglib-2.0.a \
        $(ICONV_PREFIX)/lib/libiconv.a \
        $(STAGING_DIR_ROOT)/lib/libpthread.a \
        $(INTL_PREFIX)/lib/libintl.a"
```

dl12345, 2020-04-27:
Actually, looking at nls.mk it has all the necessary definitions added to TARGET_LDFLAGS already, so the following suffices

```
CONFIGURE_VARS += \
    GLIB2_LIBS="$(STAGING_DIR)/usr/lib/libglib-2.0.a \
        $(TARGET_LDFLAGS) \
        -liconv -lintl -lpthread" 
```

dl12345, 2020-04-27:
Apologies for the multiple revisions, but the solution looked so kludgy that I wan't happy with it.

I think the following is cleaner

```
diff --git a/utils/irqbalance/Makefile b/utils/irqbalance/Makefile
index 88172af..c2b5a20 100644
--- a/utils/irqbalance/Makefile
+++ b/utils/irqbalance/Makefile
@@ -23,15 +23,16 @@ PKG_REMOVE_FILES:=autogen.sh
 
 PKG_BUILD_PARALLEL:=1
 
-# -liconv due to glib2, to be revisited later
 include $(INCLUDE_DIR)/nls.mk
 
 include $(INCLUDE_DIR)/package.mk
 
+PKG_BUILD_DEPENDS += glib2
+
 define Package/irqbalance
-  PKG_BUILD_DEPENDS:=glib2
   SECTION:=utils
   CATEGORY:=Utilities
+  DEPENDS:=$(ICONV_DEPENDS) $(INTL_DEPENDS)
   TITLE:=IRQ usage balancing for multi-core systems
   URL:=https://github.com/Irqbalance/irqbalance
 endef
@@ -42,6 +43,14 @@ define Package/irqbalance/description
  increase performance.
 endef
 
+ifeq ($(CONFIG_BUILD_NLS),y)
+TARGET_LDFLAGS += -lintl -liconv
+endif
+
+ifeq ($(CONFIG_USE_GLIBC),y)
+TARGET_LDFLAGS += -lpthread
+endif
+
 CONFIGURE_ARGS+= \
        --disable-numa \
        --with-libcap_ng=no \
@@ -51,7 +60,7 @@ CONFIGURE_ARGS+= \
 
 CONFIGURE_VARS += \
        GLIB2_LIBS="$(STAGING_DIR)/usr/lib/libglib-2.0.a \
-               $(STAGING_DIR)/usr/lib/libiconv-stub/lib/libiconv.a"
+               $(TARGET_LDFLAGS)"
 
 define Package/irqbalance/install
        $(INSTALL_DIR) $(1)/usr/sbin
```

neheb, 2020-04-27:
Whoa. PKG_BUILD_DEPENDS is in the wrong place. No idea how this slipped through.

closed 2020-04-28 by dl12345