openwrt/packages #11971: miniupnpd log spamming

axet, 2020-04-27:
Hello! miniupnpd spamming logs and here is no way to make it quite. Example:

```
Mon Apr 27 12:33:20 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 0.0.0.0 closed unexpectedly
Mon Apr 27 12:33:20 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 0.0.0.0 closed unexpectedly
Mon Apr 27 12:33:20 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 192.168.54.178 closed unexpectedly
Mon Apr 27 12:33:20 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 192.168.54.178 closed unexpectedly
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 12:33:30 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:01:56 2020 daemon.notice miniupnpd[1503]: remove port mapping 9487 TCP because it has expired
Mon Apr 27 13:01:56 2020 daemon.notice miniupnpd[1503]: remove port mapping 22143 TCP because it has expired
Mon Apr 27 13:02:28 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 0.0.0.0 closed unexpectedly
Mon Apr 27 13:02:28 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 0.0.0.0 closed unexpectedly
Mon Apr 27 13:02:28 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 192.168.54.222 closed unexpectedly
Mon Apr 27 13:02:28 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 192.168.54.222 closed unexpectedly
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:02:38 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed:290:4cff:fe14:4329' to ip address
Mon Apr 27 13:03:30 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 0.0.0.0 closed unexpectedly
Mon Apr 27 13:03:30 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 0.0.0.0 closed unexpectedly
Mon Apr 27 13:03:30 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 192.168.54.178 closed unexpectedly
Mon Apr 27 13:03:30 2020 daemon.warn miniupnpd[1503]: HTTP Connection from 192.168.54.178 closed unexpectedly
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:03:40 2020 daemon.err miniupnpd[1503]: Failed to convert hostname '2001:470:28:6ed::cc3' to ip address
Mon Apr 27 13:14:05 2020 authpriv.info dropbear[1953]: Child connection from 192.168.54.178:55848
```

Is it possible to add option to hide some errors?

neheb, 2020-04-28:
There’s a log_output setting in /etc/config/upnp

axet, 2020-04-28:
mine set to "log_output 0"

neheb, 2020-04-28:
Strange. What version of OpenWrt is this?

axet, 2020-04-28:
OpenWrt 19.07.1, r10911-c155900f66

I build it manually.

neheb, 2020-04-28:
What does ps | grep miniupnp say?

axet, 2020-04-28:
```
 1503 root      2160 S    /usr/sbin/miniupnpd -f /var/etc/miniupnpd.conf
```

_/var/etc/miniupnpd.conf_
```
ext_ifname=eth0
listening_ip=br-lan
port=5000
enable_natpmp=yes
enable_upnp=yes
secure_mode=yes
pcp_allow_thirdparty=no
system_uptime=yes
force_igd_desc_v1=no
lease_file=/var/upnp.leases
bitrate_down=8388608
bitrate_up=4194304
uuid=3ffe42eb-dd4b-4984-9dea-979f0b3ade55
allow 1024-65535 0.0.0.0/0 1024-65535 #Allow high ports
deny 0-65535 0.0.0.0/0 0-65535 #Default deny
```

neheb, 2020-04-28:
Very strange.

neheb, 2020-05-19:
Does https://github.com/openwrt/packages/pull/12174 help?

axet, 2020-05-20:
I install it, lets wait a day.

axet, 2020-05-20:
Yep, no more miniupnpd spam

axet, 2020-05-20:
Its the same

```
Wed May 20 19:27:12 2020 daemon.warn miniupnpd[1915]: HTTP Connection from 192.168.54.100 closed unexpectedly
Wed May 20 19:27:12 2020 daemon.warn miniupnpd[1915]: HTTP Connection from 192.168.54.100 closed unexpectedly
Wed May 20 19:27:12 2020 daemon.warn miniupnpd[1915]: HTTP Connection from 0.0.0.0 closed unexpectedly
Wed May 20 19:27:12 2020 daemon.warn miniupnpd[1915]: HTTP Connection from 0.0.0.0 closed unexpectedly
```

I forget to `chmod +x /etc/init.d/miniupnpd` so it sit quite for whole day :) Now it spamming again

lnpz1, 2020-06-23:
According to https://github.com/miniupnp/miniupnp/issues/408#issuecomment-554783888, this is a client bug that IPv6 address is used in an IPv4 request. Best solution is fix the client, another option if you don't need UPnP for IPv6 (i.e. no IPv6 firewall, no IPv6 NAT) is disable `miniupnpd IPv6`, but you'd need to compile `miniupnpd` yourself because the feature is in a newer version than the one in OpenWrt repository https://github.com/miniupnp/miniupnp/issues/408#issuecomment-630734145.

lnpz1, 2020-06-30:
New miniupnpd version has started to appear in OpenWrt repository, ideally `ipv6_disable=yes/no` [option](https://framagit.org/miniupnp/miniupnp/-/blob/master/miniupnpd/miniupnpd.conf#L62) needs to be parsed in https://github.com/openwrt/packages/blob/master/net/miniupnpd/files/miniupnpd.init to avoid the need to use `upnpd.config.config_file` with a custom `miniupnpd.conf`

axet, 2020-07-01:
Ok. I got new miniupnpd daemon. It's better then before, but I still got this:

```
Wed Jul  1 13:51:51 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul  1 13:51:51 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul  1 14:02:43 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.222 closed unexpectedly
Wed Jul  1 14:02:43 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.222 closed unexpectedly
Wed Jul  1 14:22:01 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul  1 14:22:01 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul  1 14:32:53 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.222 closed unexpectedly
Wed Jul  1 14:32:53 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.222 closed unexpectedly
Wed Jul  1 14:52:12 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul  1 14:52:12 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul  1 15:03:03 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.222 closed unexpectedly
Wed Jul  1 15:03:03 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.222 closed unexpectedly
Wed Jul  1 15:22:22 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul  1 15:22:22 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul  1 15:33:13 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.222 closed unexpectedly
Wed Jul  1 15:33:13 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.222 closed unexpectedly
Wed Jul  1 15:52:32 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul  1 15:52:32 2020 daemon.warn miniupnpd[1244]: HTTP Connection from 192.168.54.179 closed unexpectedly
```

neheb, 2020-07-08:
Does https://github.com/openwrt/packages/pull/12675 fix this?

axet, 2020-07-08:
It is MUTCH bettter, I can read some logs beside miniupnpd. But it is still spamming as I said in https://github.com/openwrt/packages/issues/11971#issuecomment-652404870

lnpz1, 2020-07-29:
Many clients don't comply with IGDv2. Try the following assuming you only need IGDv1
```
uci set upnpd.config.igdv1=1
uci set upnpd.config.ipv6_disable=1
```
or change `/var/etc/miniupnpd.conf` directly to include 2 lines below
```
force_igd_desc_v1=yes
ipv6_disable=yes
```

lnpz1, 2020-07-29:
> miniupnpd 2.1.20191006 heads/v19.07.3-f3f38f40da May 16 2020

That's the old version, it doesn't support `ipv6_disable` . You need `2.1.20200510` to use `ipv6_disable`, (`2.1.20200510-3` or later to use `ipv6_disable` with `uci`)

axet, 2020-07-29:
Ok. First hour gives me this:

miniupnpd 2.1.20200510 HEAD-f3f38f40da May 16 2020

```
Wed Jul 29 20:29:34 2020 daemon.warn miniupnpd[1783]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul 29 20:29:34 2020 daemon.warn miniupnpd[1783]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul 29 20:59:44 2020 daemon.warn miniupnpd[1783]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul 29 20:59:44 2020 daemon.warn miniupnpd[1783]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul 29 21:29:54 2020 daemon.warn miniupnpd[1783]: HTTP Connection from 192.168.54.179 closed unexpectedly
Wed Jul 29 21:29:54 2020 daemon.warn miniupnpd[1783]: HTTP Connection from 192.168.54.179 closed unexpectedly

```

axet, 2020-08-01:
After two days, I can say it seems less spam but still:

```
Sat Aug  1 09:51:26 2020 daemon.notice miniupnpd[1715]: remove port mapping 21101 TCP because it has expired
Sat Aug  1 09:51:26 2020 daemon.notice miniupnpd[1715]: remove port mapping 22143 TCP because it has expired
Sat Aug  1 09:54:50 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 09:54:50 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 10:17:00 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 10:17:02 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 10:21:02 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 10:21:02 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 10:25:00 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 10:25:00 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 10:31:14 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 10:31:14 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 10:55:10 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 10:55:10 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 11:25:20 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 11:25:20 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 11:31:25 2020 daemon.notice miniupnpd[1715]: remove port mapping 22143 TCP because it has expired
Sat Aug  1 11:31:25 2020 daemon.notice miniupnpd[1715]: remove port mapping 21101 TCP because it has expired
Sat Aug  1 11:33:16 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 11:33:16 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 11:55:30 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 11:55:30 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 12:06:56 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 12:06:56 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 12:25:40 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 12:25:40 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 12:42:19 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 12:42:19 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 12:48:20 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 12:48:20 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
Sat Aug  1 12:55:50 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 12:55:50 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly
Sat Aug  1 13:13:15 2020 authpriv.info dropbear[2016]: Child connection from 192.168.54.179:59742
```

root@OpenWrt:~# miniupnpd --version
miniupnpd 2.1.20200510 HEAD-f3f38f40da May 16 2020
using netfilter(iptables) backend

lnpz2, 2020-09-20:
> Sat Aug  1 12:48:20 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.241 closed unexpectedly
> Sat Aug  1 12:55:50 2020 daemon.warn miniupnpd[1715]: HTTP Connection from 192.168.54.179 closed unexpectedly

What UPnP clients are running on these hosts? Maybe they're too buggy?

@neheb The latest version (release 3) #12675 doesn't seem to be available in the repository which still has release 2 according to the file name https://downloads.openwrt.org/releases/packages-19.07/x86_64/packages/

neheb, 2020-09-21:
Backported.

axet, 2020-12-31:
I rebuild openwrt-19.07-5 and it stops spamming. 'logread' now shows logins from past few days and nothing is spammed. Did I failed to rebuild it properly last time?

Neustradamus, 2021-08-31:
@ all: What is the status of this ticket?

tiagogaspar8, 2022-08-25:
Hi @axet, is this still present in the latest versions of miniupnpd?

axet, 2022-08-25:
You got to be kidding me. It even worse. Now it spamming like most important service ever existed. Non stop:

```
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: nft_send_rule(0x77e64610, 8, 2) send_batch failed -4
Thu Aug 25 21:35:16 2022 daemon.notice miniupnpd[2817]: remove port mapping 49922 TCP because it has expired
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: send_batch: mnl_cb_run returned -1
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: nft_send_rule(0x77e64610, 8, 2) send_batch failed -4
Thu Aug 25 21:35:16 2022 daemon.notice miniupnpd[2817]: remove port mapping 45418 TCP because it has expired
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: send_batch: mnl_cb_run returned -1
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: nft_send_rule(0x77e64610, 8, 2) send_batch failed -4
Thu Aug 25 21:35:16 2022 daemon.notice miniupnpd[2817]: remove port mapping 49923 TCP because it has expired
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: send_batch: mnl_cb_run returned -1
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: nft_send_rule(0x77e64610, 8, 2) send_batch failed -4
Thu Aug 25 21:35:16 2022 daemon.notice miniupnpd[2817]: remove port mapping 45419 TCP because it has expired
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: send_batch: mnl_cb_run returned -1
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: nft_send_rule(0x77e64610, 8, 2) send_batch failed -4
Thu Aug 25 21:35:16 2022 daemon.notice miniupnpd[2817]: remove port mapping 49924 TCP because it has expired
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: send_batch: mnl_cb_run returned -1
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: nft_send_rule(0x77e64610, 8, 2) send_batch failed -4
Thu Aug 25 21:35:16 2022 daemon.notice miniupnpd[2817]: remove port mapping 48836 TCP because it has expired
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: send_batch: mnl_cb_run returned -1
Thu Aug 25 21:35:16 2022 daemon.err miniupnpd[2817]: nft_send_rule(0x77e64610, 8, 2) send_batch failed -4
Thu Aug 25 21:35:16 2022 daemon.notice miniupnpd[2817]: remove port mapping 45420 TCP because it has expired
```

tiagogaspar8, 2022-08-25:
Is this on firewall4?

danielfdickinson, 2025-12-15:
Two of these three should be closed as duplicates: https://github.com/openwrt/packages/issues/21685, https://github.com/openwrt/packages/issues/11971 (this issue), and https://github.com/openwrt/packages/issues/17258