openwrt/packages #11973: procps-ng: bad PKG_BUILD_DEPENDS line

dl12345, 2020-04-27:
procps-ng has a build dependency on gettext and libiconv 

```
PKG_BUILD_DEPENDS:=gettext libiconv
```

The package Makefile fails to check if the build is actually using gettext-full and libiconv-full instead, in which case the line should read

```
PKG_BUILD_DEPENDS:=gettext-full libiconv-full
```
This error is present in both master and openwrt-19.07

neheb, 2020-04-27:
Sort of. This package needs nls.mk and that line removed.

dl12345, 2020-04-28:
@neheb @hnyman 

Tt turns out that this PKG_BUILD_DEPENDS line is not actually necessary and neither is nls.mk, since procps-ng does not use libiconv or gettext. 

In fact, removing the PKG_BUILDS_DEPEND line and adding nls.mk causes the build to error out when compiling with libiconv-full (unable to find a bunch of iconv symbols).

This line was originally added back in 2016 in [commit aafbee](https://github.com/openwrt/packages/commit/aafbeea12b10e98b9ea9a05c996730988a938cf5#diff-37db63196fd9d4cad61f606748108fa9) due to an error compiing on malta.

I've now tested compilation of procps-ng (with all options enabled) without the PKG_BUILD_DEPENDS line on x86_64_glibc with full NLS / libiconv-full and libint-full, on mips24_musl in default configuration without NLS and on malta/mipsel_24kc_musl (the previously offending platform) with both. I've tested runtime functionality on x86_64.

It seems that the problem that originally caused the addition of this PKG_BUILD_DEPENDS line no longer exists and that it can be safely removed.

Given that this line would cause libiconv-stub to be built and potentially erroneously linked by other  buggy Makefiles (such as was the case with irqbalance), it strikes me that it would be sensible to fix it so that it does not hide these buggy Makefiles in the future. 

There is of course the risk that other, un-tested platforms may still require the PKG_BUILD_DEPENDS line, but I think it unlikely given that the reported platform was malta and I've tested it against Malta.

So, to create a PR or not?

neheb, 2020-04-29:
Merged.

closed 2020-04-29 by neheb